espressif/esp-hosted #92: INIT event is not received from ESP32

gediz, 2022-01-02:
Hi. First of all, I'd like to thank you for all your efforts. This repo is pretty well documented and I am amazed.

I'm trying to make esp-hosted work using official `ESP-WROVER-KIT` (v3) board and `Raspberry Pi 3 Model B V1.2` to no avail.

On ESP32 I've tried both released binary files and self compiled up-to-date master branch of esp-hosted repository. Also using latest Raspberry Pi OS (`2021-10-30-raspios-bullseye-armhf-lite`) with `Linux 5.10.63` on the RPi.

Paired the devices using female to female jumper wires. After reading previously submitted issues I've cutted them half to make them even shorter. Removed and re-wired all of them after every single failure and even triple checked them. But I still do not see any output after `ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]` on dmesg.

Collected logs:

<details>
  <summary>Raspberry Pi - dmesg</summary> 

  ```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1488 SMP Thu Nov 18 16:14:44 GMT 2021
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x1ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 2133 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=2577b6ec-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 681812K/970752K available (10240K kernel code, 1312K rwdata, 2948K rodata, 1024K init, 863K bss, 26796K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32054 entries in 63 pages
[    0.000000] ftrace: allocated 63 pages with 6 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3ac/0x580 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000027] Switching to timer-based delay loop, resolution 52ns
[    0.000314] Console: colour dummy device 80x30
[    0.001132] printk: console [tty1] enabled
[    0.001205] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001264] pid_max: default: 32768 minimum: 301
[    0.001483] LSM: Security Framework initializing
[    0.001744] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001794] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003285] cgroup: Disabling memory control group subsystem
[    0.003573] CPU: Testing write buffer coherency: ok
[    0.004098] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005362] Setting up static identity map for 0x100000 - 0x10003c
[    0.005571] rcu: Hierarchical SRCU implementation.
[    0.006521] smp: Bringing up secondary CPUs ...
[... 552 lines cut ...]
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:f6:42:8c brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.176/24 brd 192.168.100.255 scope global dynamic noprefixroute eth0
       valid_lft 691023sec preferred_lft 604623sec
    inet6 fe80::c4ec:c7bc:4f39:9586/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:a3:17:d9 brd ff:ff:ff:ff:ff:ff
4: ethsta0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
5: ethap0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
```
</details>

This is the wiring between the devices. I could not see any problem even after checking all of the connections using continuity mode of my multimeter but maybe you can tell what I'm doing wrong.

[image]

Looking forward for your inputs to try. Thanks in advance.

MangeshMalusare, 2022-01-03:
Hi @gediz ,
Thanks for the detailed report and your kind words.
W.r.t ESP32 SPI, there are some changes in hardware pin connections in Release 0.3 and latest code base. Those are not compatible with each other.

Release 0.3 will not work if connections made are based on latest documentation. Like wise, latest code will not work if connections made are based on release 0.3 documentation.

Can you please help share your pin connections? We will verify if those are appropriate for the code base you are trying to use. 

Also please share /boot/config.txt on Raspberry-Pi and sdkconfig (./esp/esp_driver/network_adapter/sdkconfig) for a case where you locally built esp-hosted firmware.

Thanks,
Mangesh

gediz, 2022-01-03:
Hello again @MangeshMalusare,
Your comment made everything clear for me. I've applied necessary changes and it works right now, thanks to you. There's a sample speed test result, which is more than enough for my use case:

```
Speedtest by Ookla

     Server: Edgeuno - Istanbul (id = 36189)
        ISP: TurkNet
    Latency:    12.51 ms   (0.72 ms jitter)
   Download:     5.54 Mbps (data used: 5.1 MB )                               
     Upload:     0.64 Mbps (data used: 740.2 kB )                               
Packet Loss:     0.7%
```

I'd like to write down what I've tried, I hope it'd be helpful for those who experience the same problem.

1. Flashed [v0.3 release binary for SPI](https://github.com/espressif/esp-hosted/releases/download/release%2Fv0.3/esp_hosted_firmware_esp32_spi_v0.3.zip) to the ESP32.
2. Updated my cabling according to the [older setup guide published for v0.3](https://github.com/espressif/esp-hosted/blob/82d0e3ebc53fee01b1f293fe4ca447b0c51bc2c1/docs/Linux_based_host/SPI_setup.md#111-esp32-setup).
2. Checked out [release/v0.3](https://github.com/espressif/esp-hosted/tree/release/v0.3) tag on the RPi.
3. Tried to compile it but received an error complaining about the function signature of `ndo_tx_timeout`.
4. Browsed previous commits and found out that this problem is fixed with [ac8bf21](https://github.com/espressif/esp-hosted/commit/ac8bf2124052e6bfa8154dce8808be15ebf43ab8). Checked out related commit, successfully compiled it and finally `Received INIT event from ESP32 peripheral` message is seen on `dmesg` output of Raspberry Pi.
5. Then tested Python scripts mentioned in [Getting Started](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/Getting_started.md) page.

closed 2022-01-03 by gediz