espressif/esp-hosted #93: esp_sdio: probe of mmc1:0001:1 failed with error -110

gediz, 2022-01-02:
Hi. I've tried to make it work using SDIO after failing with SPI (#92) but it became another failure.

Using official `ESP-WROVER-KIT` (v3) board and `Raspberry Pi 3 Model B V1.2`.

On ESP32 I've tried both released binary files and self compiled up-to-date master branch of esp-hosted repository. Also using latest Raspberry Pi OS (`2021-10-30-raspios-bullseye-armhf-lite`) with `Linux 5.10.63` on the RPi.

Connected the devices using female to female jumper wires. After reading previously submitted issues I've cutted them half to make them even shorter. Removed and re-wired all of them after every failure and checked them again.

But even though I can see ESP32 as a peripheral:
```
sudo cat /sys/devices/platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1\:0001/mmc1\:0001\:1/uevent 
SDIO_CLASS=00
SDIO_ID=6666:2222
SDIO_REVISION=0.0
MODALIAS=sdio:c00v6666d2222
```

There is a problem during the probe of `mmc1`.
`esp_sdio: probe of mmc1:0001:2 failed with error -22`

I have collceted some logs with the hope of making the analyze easier:

<details>
  <summary>Raspberry Pi - dmesg</summary> 

  ```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1488 SMP Thu Nov 18 16:14:44 GMT 2021
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x1ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 2133 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=2577b6ec-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 681812K/970752K available (10240K kernel code, 1312K rwdata, 2948K rodata, 1024K init, 863K bss, 26796K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32054 entries in 63 pages
[    0.000000] ftrace: allocated 63 pages with 6 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3ac/0x580 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000311] Console: colour dummy device 80x30
[... 467 lines cut ...]
  <summary>Raspberry Pi - ifconfig</summary> 

```
ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.112  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::c4ec:c7bc:4f39:9586  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:f6:42:8c  txqueuelen 1000  (Ethernet)
        RX packets 1241  bytes 81798 (79.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 597  bytes 122871 (119.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 14  bytes 1842 (1.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 14  bytes 1842 (1.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
</details>

Thanks in advance.

mantriyogesh, 2022-02-18:
Hello @gediz ,

Apologies that we could not reply in due time.
Were you able to resolve the issue?

From your logs I could see that SDIO is detected at host, but probing is not successful.
This typically happens if wires are loose or longer.
Have you tried with smaller wires like 6-7cm?

gediz, 2022-02-19:
Hello @mantriyogesh, thanks for the reply.
I've used shorter cables and even tried it on a perfboard with better wiring to no avail.

mantriyogesh, 2022-02-21:
I could see the sdcard is detected by Linux but is not getting probed by the ESP32 driver.
You can ignore -110 and -22 error as if now.

To get you going, some alternatives,
1. Can you try with release/v4.0
2. May be some issue with ESP32, is it possible to try with another ESP32?
3. Alternatively, Is it possible to use SPI as transport?

mantriyogesh, 2022-02-23:
Some questions/comments:

1. Could you please post your SDIO setup's /boot/config.txt here?
2. How are you loading esp32 driver? Is it manual or as daemon/service?
3. overall -110 shows signal integrity issue, can you please verify the connections and cables?

mantriyogesh, 2022-02-23:
Also please have a look at https://github.com/espressif/esp-hosted/issues/26#issuecomment-819344162 if you have correct driver strength ?