espressif/esp-hosted #94: Compilation error on RPI 64bit - raspios_lite_arm64

ublsoftware, 2022-01-06:
Hi,
`uname -a
Linux raspberrypi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
`

with sudo ./rpi_init.sh sdio i have this error:

`/home/pi/esp-hosted/host/linux/host_driver/esp32/esp_serial.c: At top level:
/home/pi/esp-hosted/host/linux/host_driver/esp32/esp_serial.c:179:10: error: initialization of ‘ssize_t (*)(struct file *, char *, size_t,  loff_t *)’ {aka ‘long int (*)(struct file *, char *, long unsigned int,  long long int *)’} from incompatible pointer type ‘int (*)(struct file *, char *, size_t,  loff_t *)’ {aka ‘int (*)(struct file *, char *, long unsigned int,  long long int *)’} [-Werror=incompatible-pointer-types]
  .read = esp_serial_read,
          ^~~~~~~~~~~~~~~
/home/pi/esp-hosted/host/linux/host_driver/esp32/esp_serial.c:179:10: note: (near initialization for ‘esp_serial_fops.read’)
/home/pi/esp-hosted/host/linux/host_driver/esp32/esp_serial.c:180:11: error: initialization of ‘ssize_t (*)(struct file *, const char *, size_t,  loff_t *)’ {aka ‘long int (*)(struct file *, const char *, long unsigned int,  long long int *)’} from incompatible pointer type ‘int (*)(struct file *, const char *, size_t,  loff_t *)’ {aka ‘int (*)(struct file *, const char *, long unsigned int,  long long int *)’} [-Werror=incompatible-pointer-types]
  .write = esp_serial_write,
           ^~~~~~~~~~~~~~~~
/home/pi/esp-hosted/host/linux/host_driver/esp32/esp_serial.c:180:11: note: (near initialization for ‘esp_serial_fops.write’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:280: /home/pi/esp-hosted/host/linux/host_driver/esp32/esp_serial.o] Error 1
`

is there anyone who can help me?
Thank you
Robert

ajita02, 2022-01-06:
Hi,

We have tested hosted solution till `5.10.17-v7+`. We will support this version ASAP in master and we will update you.

Thanks for patience,
Ajita

ublsoftware, 2022-01-06:
Hi, 
to correct the problem I changed:
from
`static int esp_serial_read...`
to
`static ssize_t esp_serial_read...`

and

from
`static int esp_serial_write`
to
`static ssize_t esp_serial_write`

Thank you
Roberto

ublsoftware, 2022-01-06:
Hi, 
the card appears to be connected but cannot create the Wlan interface
thank you

[ 1360.154725] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[ 1360.162215] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[ 1360.165475] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[ 1360.167861] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[ 1360.167965] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[ 1360.168067] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[ 1360.169961] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[ 1360.170065] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[ 1360.170167] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[ 1360.170374] mmc1: new SDIO card at address 0001
[ 1360.170826] esp_probe: ESP network device detected
[ 1360.173379] esp_sdio: probe of mmc1:0001:2 failed with error -22

ajita02, 2022-01-07:
Hi @ublsoftware,

WLAN interface will get created after calling `station_connect.py` or `softap_config.py` scripts in python or similar c implementation.

Thanks,
Ajita

ublsoftware, 2022-01-07:
Hi @ajita02 ,
thanks for the answer, I still have a communication problem via sdio:

with:
pi@raspberrypi:~/esp-hosted/host/linux/host_control/python_support $ sudo python3 ap_scan_list.py
esp_hosted_driver_read, Unable to read data before 30 seconds timeout
transport_pserial_data_handler:154 Failed to read RX data
wifi_ap_scan_list:943 Failed to process rx data
Failed to get AP scan list

i see in esp32 log:
data ready
I (960116) slave_commands: Station mode set in scan handler
I (961716) slave_commands: Total APs scanned = 7
I (961716) slave_commands: Details of AP no 0
I (961716) slave_commands: SSID                 ***
I (961716) slave_commands: RSSI                 -31
I (961716) slave_commands: Channel              6
I (961726) slave_commands: BSSID                ***
I (961726) slave_commands: Auth mode            3

could it be a sdio connection problem, i am using an esp-wrover-kit card?
thank you
Rob.

ajita02, 2022-01-07:
Hi @ublsoftware ,

Referring to RPi log, It seems that its a connection issue. Use around 6 cm wires for better connectivity.

Thanks,
Ajita

ajita02, 2022-01-25:
> Hi, to correct the problem I changed: from `static int esp_serial_read...` to `static ssize_t esp_serial_read...`
> 
> and
> 
> from `static int esp_serial_write` to `static ssize_t esp_serial_write`
> 
> Thank you Roberto

Hi @ublsoftware,
Your suggestion is added in current master. 

Thanks,
Ajita