espressif/esp-hosted #97: Create esp32c3 binary for flash in esp32-c3-mini

er-rahul-km, 2022-01-25:
I want to compile sorce code for esp32-c3 for linux hosted solution. I am following link
https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/SPI_setup.md
I got stuck in step 2.2.2 Source Compilation
I am not able to understand the source code cloning procedure.
Please help me to know source code repository url. Can you please provide me the url of source code repository.
Please also provide steps to compile the source code.

mantriyogesh, 2022-01-25:
hello @er-rahul-km 
This is basic step, to clone github repo.
Please visit root of repo,
https://github.com/espressif/esp-hosted
when you click code, you would be shown multiple ways to clone and the URL to be cloned.
[image]

mantriyogesh, 2022-01-25:
Please follow complete documentation for [2.1 Linux based host](https://github.com/espressif/esp-hosted#21-getting-started-with-linux-host). When you progress ahead, building steps and how to load kernel driver, verify if driver loaded would be explained.