espressif/esp-hosted #100: slow response from esp32

shchoi1, 2022-02-09:
Hi,
i have several boards with the same esp32 module(wroom-32d) installed.
after testing the connection for a long time, there are boards that are fine in the beginning but slow down the communication speed.
and at that time, if i send a simple command to esp32, for example wifi_get_mac, timeout error happens in "esp_hosted_driver_read" function.
in normal state, if i execute wifi_get_mac, below log is displayed almost at the same time from the esp32 uart channel
>>>>>>>>>>>>>>>>>>>>
data ready
I (2538630) slave_commands: Get station mac address
<<<<<<<<<<<<<<<<<<<<

but when wifi is slow, if i execute same function, same log is displayed after about 30 seconds.

what do i do to solve wifi slow issue?
could you give me any advice?
thanks.

mantriyogesh, 2022-02-09:
Hello @shchoi1 

I did not clearly understand slow wifi.
To communicate with wifi, ESP32 doen't need internet. So unless there is huge interference or wifi outage or some wifi router configuration such that it may sleep, I believe the wifi is not issue.

It would be helpful to understand if you could give some information:
1. git commit you are using or release details at both side, ESP32 and host
2. transport used, SPI/SDIO?
3. all possible logs you have

In case you are using Linux,
To address to your problem, some suggestions,
- Is the wires loose?
- Is there anything suspicious log in `dmesg` or `/var/log/messages` ?
In case unsure, in new terminal, run, `$ tail -F /var/log/messages` and reproduce the issue. This will give current session host log.
- you can enable the ESP32 and host side serial logs which would help in understanding issue.
  - For ESP32,
    - Please build the esp code for that transport using documentation
    - Before building, Please enable logging using `idf.py menuconfig` -> `Example Configuration` -> Enable `Debug Serial driver data path` -> `idf.py flash monitor`
  - For Linux,
    - uncomment https://github.com/espressif/esp-hosted/blob/master/host/linux/host_driver/esp32/main.c#L406
    - uncomment  https://github.com/espressif/esp-hosted/blob/master/host/linux/host_driver/esp32/esp_serial.c#L126
    - reload driver using `rpi_init.sh` as per documentation

  - This way it would be easier to understand if control msg request (wifi_get_mac request) being reached at ESP32 and response (wifi_get_mac response) being reached at host.

shchoi1, 2022-02-10:
Hi @mantriyogesh,

thank you for your quick & detailed reply.
let me tell you some more information

- platform: rk3399, linux
- esp32 driver: https://github.com/espressif/esp-hosted, release/v0.3 
- interface: sdio

- slow wifi
  board(with esp32)  ----------------- AP ------------------ pc
  i try to connect with the board from my PC using ssh or scp(tera-term, putty, winscp, etc..), but response is very slow.
  AP and PC has no problem. internet speed of PC is normal. 
  and in this case, response from control channel(uart) also slow.
  there was nothing strange in print_hex_dump result. 

- when i saw your comment, it seems user can modify and build esp32 S/W and download it. 
  can you let me know where i can get esp32 S/W and document?
  i have only esp32 linux host driver.

mantriyogesh, 2022-02-18:
Also as you are using release/0.3, Can you please verify your wire connections with [release/0.3 documentation](https://github.com/espressif/esp-hosted/blob/d30942e7d863a78c1ea2d13c0f964529ff542fef) -> [ESP32 SPI connections for release/0.3](https://github.com/espressif/esp-hosted/blob/d30942e7d863a78c1ea2d13c0f964529ff542fef/docs/Linux_based_host/SPI_setup.md#111-esp32-setup)

Alternatively, you can also try [release/0.4](https://github.com/espressif/esp-hosted/releases/tag/release%2Fv0.4)

mantriyogesh, 2022-03-28:
@shchoi1 
Hello @shchoi1 

Were you able to get the software working?
If yes, can you please close the issue?

closed 2022-03-28 by shchoi1