espressif/esp-hosted #101: esp32-c3 not responding

er-rahul-km, 2022-02-11:
Hi,
I am stuck at esp32-c3-mini integration in host mode with my hardware.
I am running esp32-c3 in spi mode for both wifi and bluetooth.
As per document link given below
https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/porting_guide.md

 My host has debian os.
We have modified our dtb and driver code matching with our hardware.
I can load driver esp32_spi.ko at host successfully.
I am giving my dmesg logs here..
 [  144.947874] esp32_spi: loading out-of-tree module taints kernel.
[  144.953483] esp_reset, ESP32: Resetpin of Host is 108
[  144.955374] esp_reset, ESP32: Triggering ESP reset.
[  144.986323] ESP32 peripheral is registered to SPI bus [2],chip select [0], SPI Clock [10]
 
We have also verified spi signals at esp32 using DSO(Digital storage Oscilloscope) and seems it is valid when driver is loaded. I am able see clocks also at DSO.
 
I ran ifconfig -a at host.
I received following messages
ethap0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 00:00:00:00:00:00  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
 
ethsta0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 00:00:00:00:00:00  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
 
As per the documentation i am expecting to receive

[   65.591541] esp32: Triggering ESP reset.
[   65.593385] ESP32 device is registered to SPI bus [0],chip select [0]
[   66.201597] Received INIT event from esp32
[   66.201613] ESP32 capabilities: 0x78
[   66.619381] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   66.619388] Bluetooth: BNEP filters: protocol multicast
[   66.619404] Bluetooth: BNEP socket layer initialized

But i am not getting these logs. My esp32 is not working.
 
Following logs given below i got from esp32-c3 using uart interface via minicom.
I (501) NETWORK_ADAPTER: ESP Bluetooth MAC addr: a0-76-4e-6c-9e-46      
                                                                       
I (501) SPI_DRIVER: Using SPI interface                                
I (511) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0
I (521) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0
I (1531) pp: pp rom version: 9387209                                    
I (1531) net80211: net80211 rom version: 9387209                        
I (1541) NETWORK_ADAPTER: Initial set up done
 
Please confirm whether there would be any extra logs when driver would be loaded at host.
 
Please provide me solution to solve issue for esp32-c3 host mode integration.

mantriyogesh, 2022-02-14:
Hello @er-rahul-km,

Thanks a lot for exact description of the problem.

For your logs, I could clearly infer,
The first transaction is not happening, that means SPI instance which is configured in device tree, is running correctly.

Some suggestion based on our experience:
1.  Before going into deep debugging, it would be really good to verify pins and wires are correctly connected
2. Please check expected SPI instance is correctly configured in device tree
3. The chip select also is expected to be correctly set.
4. In stead of jumping directly to spi disable,  Could you please verify using some **userspace drivers** like spidev, if spi is getting correctly configured? This way we will ensure the SPI peripheral is correctly instantiated.
5. the device would be listed as say /dev/spidev0 or something.
6. Now you can disable the spi so that kernel driver would be able to able to access SPI instance exposed in step 5.

er-rahul-km, 2022-02-17:
Dear MantriYogesh,
I did checked my connection well.
I checked SPI instance, It is correct. 
I checked CS that is correctly set.
Spi is correctly configured.
Dievice is /dev/spidev0 does exist after driver load.

Now when i do dmesg on host, It is showing that data ready pin and handshake pin is working.
But application is not receiving event 1 and event 2 .
This is the reason in my case esp32-c3 is not working.
my kernel version is 4.14
Please provide solution to this bug.

Rainbow-Cai, 2022-02-17:
Dear @mantriyogesh  ， Could you please provide a solution for this problem?

mantriyogesh, 2022-02-17:
@er-rahul-km 
1. Please send us the logs when you are using userspace driver like spidev?
2. ls -lrt /dev/spi*
3. what was the SPI instance configured in DT
4. Whats is chip select used

mantriyogesh, 2022-02-17:
Could you please provide the dmesg log for the last run?
We will be in better position to understand the issue..

Overall, to run hosted driver, transport has to be setup first.
While porting for Linux using [porting guide](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/porting_guide.md), steps to keep in mind:
- Device Tree
  - Modify & load Device tree settings for getting expected SPI instance.
  - Make a note of which /dev/spidevX.Y is created newly for this instance.
- Verify SPI is working
  - Use any user space driver like `spidev` to verify SPI on /dev/spidevX.Y
  - Configure host as SPI master and ESP32 (all supported variants like S2, C3) as SPI slave.
  - For ESP32 SPI slave driver, you can use **ESP32 IDF SPI slave examples**
- Disable spidev
  - Hosted project is not yet in picture
  - Now we are sure that SPI is working fine at userspace
  - Disable the spidev driver, for the interested SPI instance
  - User can choose to add disabled as overlay or Device Tree configuration
- Load hosted project
  - If any of prior steps failing, loading driver will not work.
  - Flash ESP32 and host using hosted project
  - Observe dmesg log for initial log as stated in documentation

er-rahul-km, 2022-02-18:
Hi,
device tree node
&ecspi3 {
        fsl,spi-num-chipselects = <1>;
        pinctrl-names = "default";
        pinctrl-0 = <&pinctrl_ecspi3 &pinctrl_ecspi3_cs>;
        cs-gpios = <&gpio4 11 0>; 
        status = "okay";

	esp32: spi@0 {
		reg = <0>;
		compatible = "spidev";
		spi-max-frequency = <30000000>;
	};
	
};

&esp32{
	status = "disabled";
};

       pinctrl_ecspi3: ecspi3grp {					/* ESP32-C3 SPI */
                fsl,pins = <
			MX7D_PAD_I2C1_SCL__ECSPI3_MISO		0x3B 	/* 0xf */ 
			MX7D_PAD_I2C1_SDA__ECSPI3_MOSI		0x3B 	/* 0xf */ 
			MX7D_PAD_I2C2_SCL__ECSPI3_SCLK		0x3B 	/* 0xf */ 
                >;
        };

        pinctrl_ecspi3_cs: ecspi3_cs_grp {
                fsl,pins = <
			MX7D_PAD_I2C2_SDA__ECSPI3_SS0		0x34	  
                >;
        };

dmesg logs for esp32 driver
[    4.962491] systemd-journald[115]: Received request to flush runtime journal from PID 1
[    5.937665] evbug: Connected device: input0 (30370000.snvs:snvs-powerkey at snvs-pwrkey/input0)
[    5.988393] 8021q: 802.1Q VLAN Support v1.8
[    6.101769] Generic PHY 30be0000.ethernet-1:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=30be0000.ethernet-1:00, irq=POLL)
[    6.117859] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.203820] Bluetooth: HCI UART driver ver 2.3
[    6.206979] Bluetooth: HCI UART protocol H4 registered
[    6.231371] Bluetooth: HCI UART protocol LL registered
[    6.235483] Bluetooth: HCI UART protocol Intel registered
[    6.325900] imx-sdma 30bd0000.sdma: no iram assigned, using external mem
[    6.373762] spi_imx 30830000.ecspi: probed
[    6.380979] spi_imx 30840000.ecspi: probed
[    6.398415] imx-sdma 30bd0000.sdma: loaded firmware 4.5
[    6.455179] Bluetooth: hci0: change remote baud rate command in firmware
[    7.535362] wl18xx_driver wl18xx.0.auto: Falling back to user helper
[    7.785295] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.789328] Bluetooth: BNEP filters: protocol multicast
[    7.794042] Bluetooth: BNEP socket layer initialized
[    8.010001] wlcore: wl18xx HW: 183x or 180x, PG 2.2 (ROM 0x11)
[    8.023238] wlcore: loaded
[    8.243392] fec 30be0000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[    8.258181] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    8.859078] wlcore: PHY firmware version: Rev 8.2.0.0.242
[    8.974734] wlcore: firmware booted (Rev 8.9.0.0.78)
[    8.988185] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   31.121292] random: crng init done
[   31.123410] random: 7 urandom warning(s) missed due to ratelimiting
[  155.812223] esp32_spi: loading out-of-tree module taints kernel.
[  155.817886] esp_reset, ESP32: Resetpin of Host is 77
[  155.818407] esp_reset, ESP32: Triggering ESP reset.
[  155.819174] ESP32 peripheral is registered to SPI bus [2],chip select [0], SPI Clock [10]

If i compare these logs with rpi.
Here i did not receive event 1 and event 2
[... 60 lines cut ...]
I (411) NETWORK_ADAPTER: Supported features are:                                   
I (421) NETWORK_ADAPTER: - WLAN over SPI                                           
I (421) ESP_BT: - BT/BLE                                                           
I (431) ESP_BT:    - HCI Over SPI                                                  
I (431) ESP_BT:    - BLE only                                                      
I (441) BTDM_INIT: BT controller compile version [33175c8]                         
I (441) coexist: coexist rom version 9387209                                       
I (441) phy_init: phy_version 907,3369105-dirty,Dec  3 2021,14:55:12               
I (491) system_api: Base MAC address is not set                                    
I (491) system_api: read default base MAC address from EFUSE                       
I (491) BTDM_INIT: Bluetooth MAC: 84:f7:03:60:b8:e6                                
                                                                                   
I (501) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 84-f7- 3-60-b8-e6                 
                                                                                   
I (501) SPI_DRIVER: Using SPI interface                                            
I (511) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 
I (521) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 
I (1531) pp: pp rom version: 9387209                                               
I (1531) net80211: net80211 rom version: 9387209                                   
I (1541) NETWORK_ADAPTER: Initial set up done                                 // stuck here

Checked before sending above logs are give below:
I am using chip select 0.
i am using esp32 at SPI3 of imx7 
After disabling spi in device tree /dev/spidev is not listed

er-rahul-km, 2022-02-18:
@mantriyogesh

mantriyogesh, 2022-02-18:
Hello @er-rahul-km 

To summarise from your logs,
1. Handshake and data ready are mapped to correct GPIO and working.
2. ESP log is correct
3. chip select is 0, configured correctly.
4. We can not confirm the device tree just by looking at it, as it very specific to the board/SoC. However the initial analysis of device tree looks fine. \
One reference device Tree we we used for other board/SoC looks like [this](https://github.com/espressif/esp-hosted/issues/44#issuecomment-903220729)

Please proceed with following steps:
- Remove device tree part where `spidev` is disabled and verify if step 2, *i.e.*  `Verify SPI is working` from [above](https://github.com/espressif/esp-hosted/issues/101#issuecomment-1043053188) comment
- Get /dev/spidevX.Y working with userspace driver (like spidev) and ESP32 SPI slave example
- If `Verify SPI is working` doesn't work, you have to revisit step 1, *i.e.* `Device Tree` for any inconsistencies

In short, Unless SPI instance is not working fine, esp32 kernel driver is not testable

er-rahul-km, 2022-02-18:
Hi @mantriyogesh,
I tested host and ESP32 with all modes of SPI mode 1, 2, 3. I did not got positive result.
I tested  ESP32 SPI slave example with user-space spidev application.
i took application from link given below.
https://github.com/azorg/spi
Application is able to send data to ESP32. I verified data through serial cable.
Now please suggest what to do next.

mantriyogesh, 2022-02-18:
Both, **RX** and **TX** cases need to be verified with user space driver
Can you please give output of:
```sh
ls -lrt /dev/spi*
```
These tests are done for SPI 3 with chip select 0 .. correct?

er-rahul-km, 2022-02-21:
When no driver is loaded
# ls -lrt /dev/spi*
ls: cannot access '/dev/spi*': No such file or directory

after running script i got this
crw-rw-rw- 1 root root 221, 0 Feb 21 05:30 /dev/spi0

I treid with the esps0 also but it does not work with this also
crw-rw-rw- 1 root root 221, 0 Feb 21 05:30 /dev/esps0

mantriyogesh, 2022-02-21:
Hello @er-rahul-km, 
It doesn't make sense at this point to try `treid with the esps0 also but it does not work with this also`, the driver has to be loaded first.

As I already stated, repeating stepwise procedure,

- Device Tree
  - Modify & load Device tree settings for getting expected SPI instance.
  - Make a note of which /dev/spidevX.Y is created newly for this instance.
- **Verify SPI is working**
  - Use any user space driver like `spidev` to verify SPI on /dev/spidevX.Y
  - Configure host as SPI master and ESP32 (all supported variants like S2, C3) as SPI slave.
  - For ESP32 SPI slave driver, you can use **ESP32 IDF SPI slave examples**
- Disable spidev
  - Hosted project is not yet in picture
  - Now we are sure that SPI is working fine at userspace
  - Disable the spidev driver, for the interested SPI instance
  - User can choose to add disabled as overlay or Device Tree configuration
  - /dev/spidevX.Y will not list after this point.
- Load hosted project
  - If any of prior steps failing, loading driver will not work.
  - X and Y which above to be configured in host spi code (refer porting guide above)
  - Flash ESP32 and load built host kernel module using hosted project
  - Observe dmesg log for initial log as stated in documentation
  
Unless /dev/spidevX.Y is created as output of device tree, and is tested with userspace driver like spidev (TX,RX), you should not proceed ahead to disable the spidev driver.
  
To add further, This kind of porting is **prerequisite** for ESP-Hosted project and is out of scope for ES-Hosted.
But as you are stuck, we are trying to help to get you working software.
Getting the platform up, is critical, without which, loading driver  or using serial driver like `/dev/esps0` does not make any sense.
 
 /dev/esps0 as described in documentation, is serial interface, over esp hosted kernel module. So unless first transaction for SPI is not done, serial driver `/dev/esps0` (which works on top of transport like SPI),

Could you please confirm if /dev/spidevX.Y is tested using spidev in RX and TX so that you can transfer some logical data to and fro host and ESP32?
Which device are you using spidev on?

er-rahul-km, 2022-02-21:
hi @mantriyogesh ,

Thank you for your support and cooperation.

Sorry for the confusion arised.

As you suggested now we have followed first 2 steps only.

"Device Tree
Modify & load Device tree settings for getting expected SPI instance.
Make a note of which /dev/spidevX.Y is created newly for this instance.
Verify SPI is working
Use any user space driver like spidev to verify SPI on /dev/spidevX.Y
Configure host as SPI master and ESP32 (all supported variants like S2, C3) as SPI slave.
For ESP32 SPI slave driver, you can use ESP32 IDF SPI slave examples"

After following above steps following is output
step 1 : Device Tree
spidev is enable in dtb
ls -lrt /dev/spi*
crw------- 1 root root 153, 0 Feb 21 07:44 /dev/spidev1.0
crw------- 1 root root 153, 1 Feb 21 07:44 /dev/spidev2.0 // This device is being used for test

step 2 : Verify SPI is working
We have used base code User space application for spidev test at host is based on https://github.com/azorg/spi and we have customized it for our changes and code is placed at url https://gist.github.com/er-rahul-km/a6538eb46973fa9f67af0ab49bde77d5

Logs of spi test are given below:
On host side logs:

root@cl-debian:~/Downloads/esp-hosted/spi-master# ./spi_test
SPI: open device='/dev/spidev2.0' mode=0 bits=8 lsb=0 max_speed=30000000 [Hz]
>>> spi_init() return 0
write buffer: ABCDEFGHIJKLMNOPQRSTUVWXYZABCD // sent to esp32 from host
read buffer: �his is the receiver, sendin�� // Received from esp32 at host

Logs on ESp32C3 application running at esp32 built from esp-idf/esp-idf-v4.4/examples/peripherals/spi_slave/receiver
Esp32 uart console logs given below:

Received: ABCDEFGHIJKLMNOPQRSTUVWXYZAB���������������������������������������.

Our observation for Tx (Host-->Esp32):
Seems like last 2bytes are always missing.
Data transmitted from host is 30 bytes, but at esp32 I am getting 2 byts lesser than transmitted.
I changed this size also, but data recieved at esp32 remains two bytes lesser than transmitted.

Our observation for Rx(Esp32-->Host) :
As per source code for esp32 given in example esp-idf/esp-idf-v4.4/examples/peripherals/spi_slave/receiver/main/app_main.c
sprintf(sendbuf, "This is the receiver, sending data for transmission number %04d.", n); //line no 148
Received string at host is not exactly as per code. First byte is corrupt. I received only 28 bytes.

I am not using Handshake signal for now at host side user space application used for spidev.

Can you please help?

er-rahul-km, 2022-02-21:
Hi @mantriyogesh,
Thanks for you reply.
We have checked with mode 1 and mode 3.
My first byte corruption at receive is solved now.
Regarding missing of last 2 bytes i noticed that this issue occurs only when packet size is more that 28 bytes. 
I tested with less than 28 bytes, the packets are received correctly at both ends.
I am not sure what is the cause of the issue.
Please help..

mantriyogesh, 2022-02-21:
Hello @er-rahul-km 

We cannot comment on your application as such as it is out of scope.

But it is worth to check size of read/write buffers, number of bytes passed in [spi_exchange](https://gist.github.com/er-rahul-km/a6538eb46973fa9f67af0ab49bde77d5#file-spi_test-c-L32 ) etc.

er-rahul-km, 2022-02-21:
Hi @mantriyogesh 
We have tested same user space application on raspberrypi (host) with esp32c3 (slave) and it is working fine.
No issues are there.
So, it doesn't seems application issue.
We need your support to make it working on our i.mx7d board.

mantriyogesh, 2022-02-21:
It is worthwhile to check SPI documentation for the expected board.
As the application works on Raspberry Pi, there has to some difference with respect to i.mx7d Linux SPI driver or with respect to SPI timings.

Can you lower frequency to say 1M and try?

Once you have working  back to back SPI transactions for TX and RX at user space, proceed with next set of steps

er-rahul-km, 2022-02-22:
hi @mantriyogesh,

I tested my host device in loopback mode.
I did not get any errors during transmission and reception.
I am assured that my spi is working fine.
Esp32-c3 problem is still same.

mantriyogesh, 2022-02-22:
Did you try frequency like 1M ?

mantriyogesh, 2022-02-22:
Also can you please get logic analyser output?
That will help to get clear picture.

er-rahul-km, 2022-02-23:
hi @mantriyogesh,
Thanks for your support.
With your support we are able to communicate with esp32 successfully.

I ran following command 
esp-hosted/host/linux/host_control/c_support/test.out scan
I am able to get list of available APs.
After that i started basic testing for Bluetooth and followed following link 
https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/Getting_started.md 
Test 1: I tested for gatt server setting in section 2.1.1. It worked. 
Test 2: I tested for gatt client  in section 2.1.2 . At step 8 I executed scan on, But i cannot see bluetooth device list on host. I followed step from 1 to 7 precisely.  
Can you please help why it is not listing available bluetooth device.

Thanks

mantriyogesh, 2022-02-23:
Hello @er-rahul-km,

Good to know that the ESP-Hosted is loaded fine.
Can you please let us know the issue you fixed, this might help someone later point of time.

With respect to `Test 2`, Can you please provide bluetoothctl  & btmon logs?

mantriyogesh, 2022-03-02:
@er-rahul-km Any updates further?
Were you be able to get it working?

er-rahul-km, 2022-03-03:
Hi @mantriyogesh,
I am sorry for late reply, I was traveling.
I am grateful to you for your support.

My most of the issues are resolved but still i am facing issues.
I was able to run ESP32 with my host.
ESP32 is running at lower clock speed 1Mhz.
But i am facing one problem.
When i load driver at host i did not get event.
To make it work I need to unload driver and reload it again. 
When i load driver 2nd time it works. 
Please provide the solution to this issue.

mantriyogesh, 2022-03-03:
Hello @er-rahul-km 

To understand the current state, please answer,
- Is the connect to AP in station mode working
- Is the RX and TX ping working

For your scenario of driver loaded second time, can you please
- Check `dmesg` or `/var/log/messages` log for any obvious problems
- Check if reset pin resetting the ESP **every** time? Every driver load should **reset** the ESP32.
  - In case of ESP32-C3 release binaries for flashing, `idf.py build` and `idf.py monitor` in `esp/esp_driver/network_adapter/` directory
  - If you are using code directly, `idf.py build flash monitor` in `esp/esp_driver/network_adapter/` directory
- Verify if the interrupts are working fine for `data ready` and `handshake` pin
  - If data ready interrupt missed, the host will not trigger SPI transaction
  - Adding logs in host interrupt handlers and ESP side might help to localise issue
- Worst case, Logic analyzer might help to look what wrongdoing happening if all above fails

mantriyogesh, 2022-03-28:
Hello @er-rahul-km 

Any further updates?
Were you able to get the software working?

mantriyogesh, 2022-06-23:
Hello @er-rahul-km 

Can you please help closing issue in case you have working solution?

mantriyogesh, 2022-07-13:
@er-rahul-km 
were you able to get the solution working?

KingMichaelCoy, 2022-09-07:
I also have a problem. My ArduinoCode code uploads successfully of a ble controller sketch but Bluetooth does not show on laptop or cellphone as midi device? What do I do?

mantriyogesh, 2022-09-08:
@KingMichaelCoy 
Can you please raise a separate issue for this?

er-rahul-km, 2022-10-11:
Dear mantra,
The software was working.
But there is existence of bug.
I need to load driver 2 times.
Then it does work.
Can you suggest what could be to problem.

mantriyogesh, 2022-10-11:
@er-rahul-km ,

Can you please test over master at both the ESP & Host?
There are quite an umber of fixes that had gone in since your last commit.

There should not be need to load the driver two times. Can you please let me know exactly, what works? It will help us to understand the context.

Also, can you please share the logs? It will be easier to understand the command and associated log, if you run the `tail -F /var/log/kern.log &` in the same terminal and load load the driver & connect with AP (or something else?)