espressif/esp-hosted #105: Integration With wpa_supplicant

anoop18pant, 2022-03-02:
hi,

We have successfully tested ESP32C3 module in hosted mode on Raspberry pi 3.

we want to use wpa_supplicant to connect to WIFI access point not with the python script provided "station_connect.py".

We tried using wpa_supplicant command "wpa_supplicant -B -Dwext -c /etc/wpa_supplicant/wpa_supplicant.conf -i ethsta0" but it failed.

Can you please help us.

Thanks

mantriyogesh, 2022-03-02:
Hello @anoop18pant ,

Thanks for your question.

`wpa_supplicant` and `802.11`, is not in the scope for ESP-Hosted.
ESP-Hosted exposes `802.3` interface and control path makes use of 802.3 interface to get the WiFi up.

anoop18pant, 2022-03-02:
hi @mantriyogesh ,

Thank you for you response.

Any specific reason not to support wpa_supplicant and 802.11 for ESP hosted?

Thanks

mantriyogesh, 2022-03-02:
For connecting to Wi-Fi AP, Security protocols supported are listed [here](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/network/esp_wifi.html#_CPPv416wifi_auth_mode_t)

When script `station_connect` is run at host, it will trigger connection from ESP at [cmd_set_ap_config_handler](https://github.com/espressif/esp-hosted/blob/master/esp/esp_driver/network_adapter/main/slave_commands.c#L477) using IDF functionality [esp_wifi_set_config](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_set_config16wifi_interface_tP13wifi_config_t)

- Regarding your question, how can we choose specific auth mode before triggering request?
  - As we rely on ESP IDF Wi-Fi API [esp_wifi_set_config](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_set_config16wifi_interface_tP13wifi_config_t), At present we do not have such choice as we underlying IDF API doesn't  support/need it.

mantriyogesh, 2022-03-03:
Basically ESP will try to connect to highest possible auth mode that AP supports.

In case the filtering is must, you can find what AP's auth mode using [scan](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/python_support/ap_scan_list.py) and filter at user input level. But all that work I feel is non necessary.

If user with given SSID and password not able to connect to AP, will simply report failure.

Please do let us know for any more inputs/questions.

mantriyogesh, 2022-03-04:
As I can see, you have created https://github.com/espressif/esp-hosted/issues/107 separate,
Can we close this or that issue?

anoop18pant, 2022-03-04:
yes please

mantriyogesh, 2022-03-04:
We cannot close from our side, Please help close..

mantriyogesh, 2022-07-26:
Hello @anoop18pant ,

This is to inform you update regarding the original issue you had raised.
With the latest release, cfg802.11 support is added. With that, one can directly configure using wpa_supplicant.