espressif/esp-hosted #106: No Mac address in Ifconfig

anoop18pant, 2022-03-02:
hi,

We have successfully tested ESP32C3 module in hosted mode on Raspberry pi 3.

we used "station_connect.py" script to connect to WIFI access point and connection was sucessfull.

After that we tried ifconfig ethsta0 but it doesnt show mac address of ESP32 WIFI. 

Can you please help us.

Thanks

mantriyogesh, 2022-03-02:
Hello @anoop18pant 

Please check `host/linux/host_control/python_support/test_util.py` -> [test_station_mode_get_mac_addr()](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/python_support/test_api.py#L76) for station mode mac address.

Please note,
- Only subset of commands are exposed in [convenience scripts](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/Getting_started.md#11-controlconfigure-wi-fi-interface)
- For all possible functionalities, please do check [test.py](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/python_support/test.py)
- C based handling is also available, in case needed

In documentation these details are already present and explained much better way.

anoop18pant, 2022-03-02:
'hi @mantriyogesh 

Thank you for your support.

The problem is we can not see mac address in ifconfig -a if esp32 is not connected to wifi access point.

thanks

mantriyogesh, 2022-03-02:
Are you using sudo?
For example, station_connect script [needs sudo](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/python_support/station_connect.py#L62) to set mac address

Can you please share logs if still issue persist -
1.  dmesg
2. Terminal where commands executed
3. ifconfig -a

anoop18pant, 2022-03-03:
@mantriyogesh 

Hi Yogesh, 

One point want o make clear that we are not running station_connect.py.
We have loaded the driver and run ifconfig -a . We can see ethsta0 in the list but mac adress is 00:00:00:00:00

So we want to know why its not displaying mac without connect.
Thanks

mantriyogesh, 2022-03-03:
That is expected behaviour.
We are creating network interface [here](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_driver/esp32/main.c#L675) after loading driver.
and
removing them [here](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_driver/esp32/main.c#L704)

After just loading driver, As they are logical network interfaces created, there will be no mac address.
We will assign them mac address which is ESP mac address for that type of interface.
For `ethsta0`, station mode ESP mac address will be assigned.
For `esthap0`, softap mode ESP mac address will be assigned.

We assign mac address after the connection/network path is established for station/softap and is implemented as part of station_connect.py or test.py or test.out

mantriyogesh, 2022-03-28:
Hello @anoop18pant 

Would like to listen from you.
Any further update or can you please close the issues?

mantriyogesh, 2022-05-09:
Hello @anoop18pant 
Can you please close the issue?

mantriyogesh, 2022-09-16:
@anoop18pant Can you please close the pending issues, where you have found out solutions?
It will be great help for us.

anoop18pant, 2022-09-16:
not implemented

closed 2022-09-16 by anoop18pant

mantriyogesh, 2022-09-16:
Is there anything, that is not clear as part of https://github.com/espressif/esp-hosted/issues/106#issuecomment-1057722307?
It would be helpful, if you can let me know what your expectation? Just 'not implemented' doesn't help much.

mantriyogesh, 2022-09-16:
Just to make it clear through code,
https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/control_lib/src/ctrl_core.c#L1675, is where the ioctl used to set mac address.

When we use latest code from First Generation branch, _i.e._ https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host, the code flow to connect to Wi-Fi:

1. Look & feel : See gif demo for test.py [here](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/common/python_demo.md#modes-supported). This is python demo app. I am not going for 'C' demo, as question is specific to python.
2. `connect_ap` with arguments like SSID password is called from test.py 
3. [connect_ap() from cmds.py ](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/linux/host_control/python_support/py_parse/cmds.py#L106) -> [process_connect_ap() from process.py](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/linux/host_control/python_support/py_parse/process.py#L119) is triggered when you use `connect_ap` from test.py. 
4. [test_sync_station_mode_connect() from commands_lib.py](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/linux/host_control/python_support/commands_lib.py#L774)
5. [wifi_connect_ap() from commands_map_py_to_c.py](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/linux/host_control/python_support/commands_map_py_to_c.py#L58)
6. [wifi_connect_ap() from host/control_lib/src/ctrl_api.c](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/control_lib/src/ctrl_api.c#L99) This will send station connect request to ESP and wait for response
7. Response received in [ctrl_app_resp_callback()](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/linux/host_control/python_support/commands_lib.py#L366)
8. Response parsed at [process_resp_connect_ap()](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/linux/host_control/python_support/commands_lib.py#L105)
9. In this function, many networking related things done as virtual interface need to be created etc. 
for example, 
a. interface_down
b. set_hw_addr
c. interface_up
10. @anoop18pant please have a look at `set_hw_addr ()`, https://github.com/espressif/esp-hosted/blob/f209c91ca010c0899b37dfbb978b8c1dba1d57c4/host/linux/host_control/python_support/commands_lib.py#L134-L142 which might help you to understand how the MAC address is being set to the newly virtual interface created for station.
11. This further translates to [set_hw_addr() in host/control_lib/src/ctrl_core.c](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/control_lib/src/ctrl_core.c#L1645), where ioctl is used to set the mac address using `ret = ioctl(sockfd, SIOCSIFHWADDR, &req)`
https://github.com/espressif/esp-hosted/blob/f209c91ca010c0899b37dfbb978b8c1dba1d57c4/host/control_lib/src/ctrl_core.c#L1666-L1680

I hope now it is clear how it is implemented