espressif/esp-hosted #107: ESP32C3 hosted WPA Supplicant support

anoop18pant, 2022-03-04:
hi ,

We want to 802.11 compatible driver we understand that current implementation is 802.3 based. We have following queries:

1. Do ESP has a plan to provide 802.11 and wpa supplicant support.
2. If ESP dont have such plan and we want to implement it by our self is there any limitation on ESP32 side?
3. in case its possible to do it can you provide us an idea how much effort will it take?

Looking for your response.

Thanks

MangeshMalusare, 2022-03-04:
Hi @anoop18pant,
Thank you for your interest. As mentioned in project description, providing 802.11 interface is not in the scope. 
In future, we may create a separate project that will provides 802.11 interface. But it can not be promised at present.
ESP32 firmware side support is also needed to accomplish this. This is not present at this moment.

MangeshMalusare, 2022-07-26:
Hello @anoop18pant,
We have added 802.11 interface support in latest release. Kindly try it and let us know if you face any issues.

Thanks,
Mangesh