espressif/esp-hosted #108: Log or command to check current BLE connection type

anoop18pant, 2022-03-07:
hi,

I am using ESP32C3 in hosted mode. I am connecting to different ble devices 4.1,4.2 and 5.0 so how can check or make sure i am connecting on same profile like ble5.0 is connecting to esp32c3 5.0.

is there a way  to make it confirm by some logs or commands?

mantriyogesh, 2022-03-08:
This is more of user side 'how-to' question of 'Bluetooth', and not related to ESP-Hosted working as such.

One way I can think of, to achieve this could be using,
```
$ sudo btmon 
```
but this is upto user how achieve this on top of Bluetooth.

anoop18pant, 2022-03-09:
hi Yogesh,
i tried with btmon but can not get that information thats why i asked if on esp side there is some command or log thorugh which i can get this info.

Can you please help in that

mantriyogesh, 2022-03-09:
There is no processing on Bluetooth messages from Esp-Hosted.
ESP-Hosted just stands as transport mediaum, where Bluetooth traffic is transparently sent as HCI traffic to ESP Bluetooth Controller.
Overall HCI functionality is upto Bluetooth stack.

Overall, A ble connection is ble connection ..there is no difference in 4.0 ..4.1 ..4.2 ..5.0 etc
What differs is functionality.
For e.g. `LE Enhanced Connection` is not in 4.0, as because this feature was not present in that version
btmon can show you these messages.

Re-iterating myself, this is just suggestion, how to handle this is upto you.
Probably you might check better Bluetooth tools which operate over HCI, which may suit your use case.

mantriyogesh, 2022-05-09:
Hello @anoop18pant 
Can you please close the issue as nothing is to be done at ESP-Hosted side?