espressif/esp-hosted #109: hciconfig -a

anoop18pant, 2022-03-09:
I am using ESP32C3 hosted solution.

when i am running hciconfig -a command to see hci version it not showing .Can you please let me know how to get hciversion.

the result of hciconfig -a

hci0:   Type: Primary  Bus: SPI
        BD Address: 84:F7:03:5F:C7:F6  ACL MTU: 251:12  SCO MTU: 0:0
        UP RUNNING
        RX bytes:367120 acl:512 sco:0 events:17617 errors:0
        TX bytes:10225 acl:0 sco:0 commands:0 errors:0
        Features: 0x00 0x00 0x00 0x00 0x60 0x00 0x00 0x00
        Packet type: DM1 DH1 HV1
        Link policy:
        Link mode: SLAVE ACCEPT
Can't read local name on hci0: Input/output error (5)

mantriyogesh, 2022-03-09:
Better place to check this would be manual page of hciconfig.
ESP-Hosted provides transport (transparent medium) like SPI/ SDIO, just to get HCI messages reach to ESP chipset.

Also it would be better to let us know your related queries in same bug, in stead of just creating number of bugs.

anoop18pant, 2022-03-09:
hi Yogesh,

hciconfig is giving all information for builtin ble chip connected on UART. The issue is occurring only with ESP32C3 chip

mantriyogesh, 2022-03-09:
Can you please run same command on built-in bluetooth * **on same host** * and get us the output?
Basically, Need to understand what difference you get with or without `ESP-Hosted` loaded

anoop18pant, 2022-03-09:
hi,
 Please see below its same till Link mode after that ESP doesnt provide other details.:

Builtin bluetooth on same host 
hci0: Type: Primary Bus: UART
BD Address: ------------ ACL MTU: 1021:6 SCO MTU: 180:4
UP RUNNING
RX bytes:5090 acl:0 sco:0 events:630 errors:0
TX bytes:124845 acl:0 sco:0 commands:631 errors:0
Features: 0xff 0xfe 0x2d 0xfe 0xdb 0xff 0x7b 0x87
Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
Link policy: RSWITCH HOLD SNIFF
Link mode: SLAVE ACCEPT
Name: 'anoophost'
Class: 0x000000
Service Classes: Unspecified
Device Class: Miscellaneous,
HCI Version: 4.1 (0x7) Revision: 0x0
LMP Version: 4.1 (0x7) Subversion: 0xac03
Manufacturer: ABC....

ESP32 on same host

hci0: Type: Primary Bus: SPI
BD Address: 84:F7:03:5F:C7:F6 ACL MTU: 251:12 SCO MTU: 0:0
UP RUNNING
RX bytes:367120 acl:512 sco:0 events:17617 errors:0
TX bytes:10225 acl:0 sco:0 commands:0 errors:0
Features: 0x00 0x00 0x00 0x00 0x60 0x00 0x00 0x00
Packet type: DM1 DH1 HV1
Link policy:
Link mode: SLAVE ACCEPT
Can't read local name on hci0: Input/output error (5)

anoop18pant, 2022-03-11:
hi Yogesh,

Any progress on this issue?

mantriyogesh, 2022-03-11:
We will check this issue.

However, this local name setting, *i.e.* `Can't read local name on hci1: Input/output error (5)` is related to management side of bluez.
Your bluetooth functionality should be working anyway.

anoop18pant, 2022-03-11:
yes thats working but we need to get hci version

anoop18pant, 2022-03-15:
hi yogesh any progress on this.

mantriyogesh, 2022-03-25:
Hi @anoop18pant 

Read_Local_Name cmd only support for the BREDR controller. 
As we know esp32c3 and esp32s3 doesn't support BREDR , only BLE, Read_Local_Name is not used in BLE or c3/s3 chip set.

Please check the spec here -
[image]

You can see Raspberry Pi own controller is BR EDR type. When BLE only controller used like ESP32-C3, you should see this error, as it is as per spec.

Nevertheless, the all the BLE supported features should work normally in ESP32-C3.

mantriyogesh, 2022-05-09:
Hello @anoop18pant 
Can you please close this issue?

mantriyogesh, 2022-05-09:
Yes. ESP32-C3 doesn't support, as it is not supported in the specification of BLE controller, as pointed above.