espressif/esp-hosted #110: Some questions about the "ESP-Hosted" SDK

Rainbow-Cai, 2022-03-11:
Hi Espressif，

We are porting the esp-hosted solution to a Legato system. We are able to build the kernel module but still have some details to get around.
 
1. During the build and modprobe processes some similar warnings appear, is this due to the kernel being version 3.18.44?
2. Is there a way to disable BT for old kernel versions?
3. Can you point out where in the code or build process do we change the pins used?

**The Log as follow：**

[64/384] Build kernel driver module
make: Entering directory '/home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi'
make -C /home/carlos/fx30_workspace/leaf-data/SWI-FX30-CATM/wp77-toolchain/sysroots/armv7a-neon-poky-linux-gnueabi/usr/src/kernel M=/home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi modules
make[1]: Entering directory '/home/carlos/.leaf/wp77-toolchain_SWI9X06Y_02.18.05.00-linux64/sysroots/armv7a-neon-poky-linux-gnueabi/usr/src/kernel'
  CC [M]  /home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/main.o
  CC [M]  /home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp_bt.o
  CC [M]  /home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp_rb.o
  CC [M]  /home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp_serial.o
  CC [M]  /home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp_spi.o
  LD [M]  /home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp32_spi.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: "hci_register_dev" [/home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp32_spi.ko] undefined!
WARNING: "hci_recv_frame" [/home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp32_spi.ko] undefined!
WARNING: "hci_unregister_dev" [/home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp32_spi.ko] undefined!
WARNING: "bt_err" [/home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp32_spi.ko] undefined!
WARNING: "hci_alloc_dev" [/home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp32_spi.ko] undefined!
WARNING: "hci_free_dev" [/home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp32_spi.ko] undefined!
  CC      /home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp32_spi.mod.o
  LD [M]  /home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi/esp32_spi.ko
make[1]: Leaving directory '/home/carlos/.leaf/wp77-toolchain_SWI9X06Y_02.18.05.00-linux64/sysroots/armv7a-neon-poky-linux-gnueabi/usr/src/kernel'
make: Leaving directory '/home/carlos/fx30_workspace/_build_mySystem/wp77xx/modules/esp32_spi'

root@fx30s:~# logread -f | grep "esp32"
Jan  7 23:43:11 fx30s user.info Legato:  INFO | supervisor[22226]/supervisor T=main | kernelModules.c ReleaseModulesMemory() 840 | Released memory of module 'esp32_spi.ko'
Jan  7 23:43:14 fx30s user.info Legato:  INFO | supervisor[22522]/supervisor T=main | kernelModules.c ModuleGetParams() 355 | Module esp32_spi.ko uses no parameters.
Jan  7 23:43:14 fx30s user.info Legato:  INFO | supervisor[22522]/supervisor T=main | kernelModules.c ExecuteCommand() 190 | Execute '/sbin/insmod /legato/systems/current/modules/esp32_spi.ko'
Jan  7 23:43:14 fx30s user.err Legato: =ERR= | supervisor[22522]/supervisor T=main | kernelModules.c installModules() 771 | Error in installing module esp32_spi.ko
Jan  7 23:43:14 fx30s user.warn kernel: [171791.136516] esp32_spi: Unknown symbol hci_free_dev (err 0)
Jan  7 23:43:14 fx30s user.warn kernel: [171791.136650] esp32_spi: Unknown symbol hci_alloc_dev (err 0)
Jan  7 23:43:14 fx30s user.warn kernel: [171791.136689] esp32_spi: Unknown symbol bt_err (err 0)
Jan  7 23:43:14 fx30s user.warn kernel: [171791.136702] esp32_spi: Unknown symbol hci_unregister_dev (err 0)
Jan  7 23:43:14 fx30s user.warn kernel: [171791.136718] esp32_spi: Unknown symbol hci_recv_frame (err 0)
Jan  7 23:43:14 fx30s user.warn kernel: [171791.136742] esp32_spi: Unknown symbol hci_register_dev (err 0)
^C
root@fx30s:~# uname -a
Linux fx30s 3.18.44 #2 PREEMPT Tue Jan 29 11:21:07 PST 2019 armv7l GNU/Linux

Rainbow-Cai, 2022-03-11:
Hi @[mantriyogesh](https://github.com/mantriyogesh)  , Could you provide some help ？

mantriyogesh, 2022-03-11:
Hello @Rainbow-Cai ,

Thank you for trying ESP-Hosted solution.

These errors are actually because of old kernel.
Be disabling bluetooth, you can compile the kernel.
Give me some time, I had old code patch to do this.

mantriyogesh, 2022-03-11:
Hello Carlos,

Can you please provided output of 
```
$ uname -a
```

darthrookie, 2022-03-11:
> Hello Carlos,
> 
> Can you please provided output of
> 
> ```
> $ uname -a
> ```

Sure

`Linux fx30s 3.18.44 #2 PREEMPT Tue Jan 29 11:21:07 PST 2019 armv7l GNU/Linux`

mantriyogesh, 2022-03-13:
> Can you point out where in the code or build process do we change the pins used?

Overall to get SPI working, stepwise explanation is provided in documentation.
But it is worth to mention bug, https://github.com/espressif/esp-hosted/issues/101#issuecomment-1046513672 where we tried to address issues user faced for getting SPI up for platform.

While getting platform up, Focus should be to check [porting guide](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/porting_guide.md). Here pins like Reset pin, handshake and data ready pinconfiguration is mentioned.

> During the build and modprobe processes some similar warnings appear, is this due to the kernel being version 3.18.44?
> Is there a way to disable BT for old kernel versions?
Ideally there should not be issues while building, as I could see, in linux kernel v3.18.44, function like [hci_register_dev()](
https://elixir.bootlin.com/linux/v3.18.44/source/include/net/bluetooth/hci_core.h#L852) is available.

Alternative solutions:
1. Try to ***load bluetooth module***, as it will allow these functions available
2. If bluetooth is not at all required, all symbols related to bluetooth and bt_init etc should be simply be commented and then try loading the module. In this case, the kernel driver will support WiFi related functionalities.

Ncerzzk, 2022-03-13:
hi， I met the same problem, and just commented the code related with bluetooth. For wifi, it works normally now.
may be you can refer the changes: https://github.com/Ncerzzk/esp-hosted/commit/34405135844630febeed82a74927007dbef29a59

darthrookie, 2022-03-15:
Thanks @Ncerzzk and @mantriyogesh. I got a little bit further by commenting out BT. I also updated to the lastest image of my target just in case.

```
root@fx30s:~# uname -a
Linux fx30s 3.18.140 #1 PREEMPT Fri May 14 11:13:23 UTC 2021 armv7l GNU/Linux
```

Now the log shows

```
Jan  9 04:21:02 fx30s user.info Legato:  INFO | sdir[8004]/framework T=main | user.c user_Init() 445 | /etc is writable
Jan  9 04:21:03 fx30s user.info Legato:  INFO | supervisor[7988]/supervisor T=main | kernelModules.c ModuleGetParams() 575 | Module esp32_spi.ko uses no parameters.
Jan  9 04:21:03 fx30s user.info Legato:  INFO | supervisor[7988]/supervisor T=main | kernelModules.c ExecuteCommand() 244 | Execute '/sbin/modinfo -F depends /legato/systems/current/modules/esp32_spi.ko '
Jan  9 04:21:03 fx30s user.info Legato:  INFO | supervisor[7988]/supervisor T=main | kernelModules.c ExecuteCommand() 244 | Execute '/sbin/modinfo -F depends /legato/systems/current/modules/example.ko '
Jan  9 04:21:03 fx30s user.info Legato:  INFO | supervisor[7988]/supervisor T=main | kernelModules.c ModuleGetParams() 575 | Module spisvc.ko uses no parameters.
Jan  9 04:21:03 fx30s user.info Legato:  INFO | supervisor[7988]/supervisor T=main | kernelModules.c ExecuteCommand() 244 | Execute '/sbin/modinfo -F depends /legato/systems/current/modules/spisvc.ko '
Jan  9 04:21:03 fx30s user.info Legato:  INFO | supervisor[7988]/supervisor T=main | kernelModules.c ExecuteCommand() 244 | Execute '/sbin/insmod /legato/systems/current/modules/esp32_spi.ko '
Jan  9 04:21:03 fx30s user.err kernel: [ 3745.746175] Failed to obtain SPI master handle
```

Perhaps SPI module is not being loaded?. If I load it manually with `insmod /mnt/legato/system/modules/spisvc.ko` then I can see `/dev/spidev1.0`. Is that enough?

mantriyogesh, 2022-03-15:
I see that spi disabler is not in place yet.
Overall, step wise procedure mentioned in https://github.com/espressif/esp-hosted/issues/101#issuecomment-1046513672 would be much of help.

darthrookie, 2022-03-16:
Hi @mantriyogesh, since I can't use the spidev_disabler.dts directly on my platform I need to look at how it is configured then run the tests mentioned in #101.

This is the log so far after disabling spisvc.ko

```
Jan  9 22:47:17 fx30s user.info Legato:  INFO | supervisor[5900]/supervisor T=main | kernelModules.c ModuleGetParams() 575 | Module esp32_spi.ko uses no parameters.
Jan  9 22:47:17 fx30s user.info Legato:  INFO | supervisor[5900]/supervisor T=main | kernelModules.c ExecuteCommand() 244 | Execute '/sbin/modinfo -F depends /legato/systems/current/modules/esp32_spi.ko '
Jan  9 22:47:17 fx30s user.info Legato:  INFO | supervisor[5900]/supervisor T=main | kernelModules.c ExecuteCommand() 244 | Execute '/sbin/modinfo -F depends /legato/systems/current/modules/example.ko '
Jan  9 22:47:17 fx30s user.info Legato:  INFO | supervisor[5900]/supervisor T=main | kernelModules.c ExecuteCommand() 244 | Execute '/sbin/insmod /legato/systems/current/modules/esp32_spi.ko '
Jan  9 22:47:17 fx30s user.err kernel: [ 2147.343489] Failed to obtain SPI master handle
```

After that the target resets because of the failed module load.

mantriyogesh, 2022-03-16:
This is part of getting the platform up so that ESP-Hosted can be into the picture.
Did you check -
1. If the SPI is correctly working in user driver like spidev?
2. If yes, can you please show device tree how you are disabling spidev? 
  2.1 If spidev disabled for say `/dev/spidev1.0`, then this device will no longer be claimed by spidev driver and hence, you will not appear in `ls /dev/spi*` result.
  2.2 With help of porting guide, check the changes expected in code in order to change SPI instance used and chip select.
  
  It might take little time to get the platform up. But this is important, without which ESP-Hosted will not be able to load.

mantriyogesh, 2022-03-18:
For device tree related changes, you can also refer to [sample DT for SPI disabler](https://github.com/espressif/esp-hosted/issues/44#issuecomment-903220729)

mantriyogesh, 2022-03-28:
Hello @Rainbow-Cai 
Any further updates wrt issue?

darthrookie, 2022-03-28:
> Hello @Rainbow-Cai Any further updates wrt issue?

I had to attend other priority issues with the HW and target platform. But as soon as I can test SPI and other recommendations I'll get back here. Thanks.

darthrookie, 2022-04-04:
> Hello @Rainbow-Cai Any further updates wrt issue?

We have found an issue with the SPI interface, which we have now solved. The kernel module now compiles and loads. Thanks for your help.

mantriyogesh, 2022-04-04:
Hello @darthrookie 
Can you please explain the issue, so that we can try to fix if it is in hosted repo.
Even otherwise, it may help somebody watching this thread later.

mantriyogesh, 2022-06-23:
Hello @darthrookie 
Can you please help closing the issue?

darthrookie, 2022-06-30:
> Hello @darthrookie Can you please explain the issue, so that we can try to fix if it is in hosted repo. Even otherwise, it may help somebody watching this thread later.

It was related to how Legato needs different kernel modules and the order those are enabled to be able to use the SPI interface as in an RPi.

> Hello @darthrookie Can you please help closing the issue?

I consider that the issue has been solved, but it was @Rainbow-Cai who opened it.

mantriyogesh, 2022-07-01:
Thank you @darthrookie for the heads up.

@Rainbow-Cai, Could you please help in closing the issue if it is solved up to your expectations?

meera1310, 2022-09-23:
Hi, 
      I am facing this errors. Can you please tell me why

**ERROR: modpost: "hci_recv_frame" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!
| ERROR: modpost: "hci_unregister_dev" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!
| ERROR: modpost: "bt_err" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!
| ERROR: modpost: "hci_alloc_dev" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!
| ERROR: modpost: "hci_free_dev" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!**

mantriyogesh, 2022-09-23:
@meera1310 
I see your issue is already active on https://github.com/espressif/esp-hosted/issues/137#issuecomment-1256155979.

It is better if you can raise new issue and tag related issue in that.

closed 2023-11-22 by mantriyogesh