espressif/esp-hosted #111: Stm32mp1 ESP32C3 SPI Bluetooth not working, but WiFi works fine.

codebug8, 2022-03-11:
This is insmod log.
```
# insmod esp32_spi.ko resetpin=15
[   57.067806] esp32_spi: loading out-of-tree module taints kernel.
[   57.073185] esp32_spi: module verification failed: signature and/or required key missing - tainting kernel
[   57.095254] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
# [   58.627420]
[   58.627420] Received INIT event from ESP32 peripheral
[   58.627433] EVENT: 2
[   58.634685] EVENT: 1
[   58.636840] ESP Reconfigure SPI CLK to 30 MHz
[   58.859027] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [30]
[   59.078245] EVENT: 0
[   59.078964] ESP peripheral capabilities: 0x68
[   59.083296] ESP Reset BT
[   59.298778] ESP  capabilities: 0x68
```
This use bluz5 - 5.54 log.

```
# hciconfig
hci0:   Type: Primary  Bus: SPI
        BD Address: 84:F7:03:13:79:86  ACL MTU: 251:12  SCO MTU: 0:0
        UP RUNNING
        RX bytes:471 acl:0 sco:0 events:80 errors:0
        TX bytes:182 acl:0 sco:0 commands:0 errors:0
# hcitool lescan
Set scan parameters failed: Input/output error
# hcitool lescan
Set scan parameters failed: Input/output error
```
WiFi scans SSID log information normally.
```
# ./stress 100 scan

Iteration 1:
==== test_get_available_wifi =>
[  376.426570] esp_serial_rx: 3df10924: 01 07 00 63 6f 6e 74 72 6f 6c 02 99 02 08 13 ea  ...control......
[  376.434482] esp_serial_rx: 803d2734: 01 93 05 08 0e 12 2f 0a 0b 50 72 6f 67 72 61 6d  ....../..Program
[  376.443739] esp_serial_rx: 4ce05f42: 6d 65 72 73 10 07 18 d2 ff ff ff ff ff ff ff ff  mers............
[  376.453029] esp_serial_rx: 6661b887: 01 22 11 39 34 3a 64 39 3a 62 33 3a 62 37 3a 63  .".94:d9:b3:b7:c
[  376.462313] esp_serial_rx: 5f1172c1: 39 3a 30 61 28 04 12 29 0a 05 48 46 54 4b 4a 10  9:0a(..)..HFTKJ.
[  376.471561] esp_serial_rx: 44b9a951: 01 18 c6 ff ff ff ff ff ff ff ff 01 22 11 61 38  ............".a8
[  376.480848] esp_serial_rx: d257ca31: 3a 30 32 3a 64 62 3a 63 30 3a 33 65 3a 38 65 28  :02:db:c0:3e:8e(
[  376.490113] esp_serial_rx: 4d80c866: 04 12 24 0a 00 10 01 18 c6 ff ff ff ff ff ff ff  ..$.............
[  376.499409] esp_serial_rx: 78f4101c: ff 01 22 11 61 61 3a 30 32 3a 64 62 3a 62 30 3a  ..".aa:02:db:b0:
[  376.508669] esp_serial_rx: 45f627b9: 33 65 3a 38 65 28 04 12 31 0a 0d 43 68 69 6e 61  3e:8e(..1..China
[  376.517915] esp_serial_rx: 84fca287: 4e 65 74 2d 6b 52 41 48 10 01 18 c1 ff ff ff ff  Net-kRAH........
[  376.527230] esp_serial_rx: b22eae29: ff ff ff ff 01 22 11 66 30 3a 39 32 3a 62 34 3a  .....".f0:92:b4:
[  376.536515] esp_serial_rx: 8311ca5f: 61 36 3a 30 33 3a 39 31 28 04 12 31 0a 0d 43 68  a6:03:91(..1..Ch
[  376.545786] esp_serial_rx: b29b2d7e: 69 6e 61 4e 65 74 2d 48 53 78 61 10 01 18 c1 ff  inaNet-HSxa.....
[  376.555073] esp_serial_rx: 886b49a2: ff ff ff ff ff ff ff 01 22 11 35 63 3a 61 34 3a  ........".5c:a4:
[  376.564435] esp_serial_rx: 390e676c: 61 34 3a 30 61 3a 35 34 3a 30 64 28 04 12 31 0a  a4:0a:54:0d(..1.
[  376.573637] esp_serial_rx: 27f58431: 0d 43 68 69 6e 61 4e 65 74 2d 73 71 4a 72 10 04  .ChinaNet-sqJr..
[  376.582897] esp_serial_rx: 2ddd45ec: 18 b5 ff ff ff ff ff ff ff ff 01 22 11 63 38 3a  ...........".c8:
[  376.592187] esp_serial_rx: 2a32525f: 35 30 3a 65 39 3a 62 62 3a 66 35 3a 30 61 28 04  50:e9:bb:f5:0a(.
[  376.601475] esp_serial_rx: d17aa89b: 12 32 0a 0e 33 36 30 57 69 46 69 2d 42 31 32 43  .2..360WiFi-B12C
[  376.610769] esp_serial_rx: 22d50a96: 39 38 10 0b 18 b4 ff ff ff ff ff ff ff ff 01 22  98............."
[  376.620027] esp_serial_rx: 0bb2689c: 11 32 63 3a 36 31 3a 30 34 3a 62 31 3a 32 63 3a  .2c:61:04:b1:2c:
[  376.629319] esp_serial_rx: 5e35ce49: 39 38 28 04 12 32 0a 0e 33 36 30 57 69 46 69 2d  98(..2..360WiFi-
[  376.638576] esp_serial_rx: d60361e0: 35 43 46 45 43 38 10 01 18 b3 ff ff ff ff ff ff  5CFEC8..........
[  376.647821] esp_serial_rx: 00eea286: ff ff 01 22 11 63 38 3a 35 62 3a 61 30 3a 35 63  ...".c8:5b:a0:5c
[  376.657140] esp_serial_rx: bb169aeb: 3a 66 65 3a 63 38 28 04 12 24 0a 00 10 06 18 b1  :fe:c8(..$......
[  376.666416] esp_serial_rx: 965437ca: ff ff ff ff ff ff ff ff 01 22 11 64 61 3a 33 35  .........".da:35
[  376.675696] esp_serial_rx: a09ef3de: 3a 33 38 3a 37 30 3a 32 33 3a 38 65 28 00 12 36  :38:70:23:8e(..6
[  376.684978] esp_serial_rx: 8af340c0: 0a 12 e5 a4 a9 e5 88 9b e7 a7 91 e6 8a 80 e5 85  ................
[  376.694254] esp_serial_rx: c6a395c6: ac e7 94 a8 10 06 18 ad ff ff ff ff ff ff ff ff  ................
[  376.703529] esp_serial_rx: 1bb796ca: 01 22 11 64 34 3a 33 35 3a 33 38 3a 37 30 3a 32  .".d4:35:38:70:2
[  376.712900] esp_serial_rx: 22a408ca: 33 3a 38 65 28 03 12 3a 0a 16 20 e5 a4 a9 e5 88  3:8e(..:.. .....
[  376.722100] esp_serial_rx: 5e47bee1: 9b e7 a7 91 e6 8a 80 e6 8b bc e5 a4 9a e5 a4 9a  ................
[  376.731366] esp_serial_rx: 87725e1a: 10 06 18 ac ff ff ff ff ff ff ff ff 01 22 11 64  .............".d
[  376.740659] esp_serial_rx: a522f9b0: 65 3a 33 35 3a 33 38 3a 37 30 3a 32 33 3a 38 65  e:35:38:70:23:8e
[  376.749936] esp_serial_rx: 7bd1b834: 28 03 12 24 0a 00 10 02 18 ab ff ff ff ff ff ff  (..$............
[  376.759204] esp_serial_rx: 59a0c133: ff ff 01 22 11 35 32 3a 30 36 3a 61 37 3a 37 35  ...".52:06:a7:75
[  376.768477] esp_serial_rx: 53df5b54: 3a 34 38 3a 32 65 28 04 12 24 0a 00 10 07 18 ab  :48:2e(..$......
[  376.777719] esp_serial_rx: 6dcb48f4: ff ff ff ff ff ff ff ff 01 22 11 34 34 3a 34 37  .........".44:47
[  376.787032] esp_serial_rx: 85a7c07a: 3a 63 63 3a 66 32 3a 64 30 3a 32 31 28 03 12 24  :cc:f2:d0:21(..$
[  376.796311] esp_serial_rx: 632ccb7e: 0a 00 10 01 18 aa ff ff ff ff ff ff ff ff 01 22  ..............."
[  376.805590] esp_serial_rx: d89f832a: 11 61 34 3a 62 36 3a 31 65 3a 31 35 3a 64 66 3a  .a4:b6:1e:15:df:
[  376.814867] esp_serial_rx: a1784130: 38 64 28 03 18 00                                8d(...
Number of available APs is 14
0 th AP's ssid "Programmers" bssid "94:d9:b3:b7:c9:0a" rssi "-46" channel "7" authentication mode "4"
1 th AP's ssid "HFTKJ" bssid "a8:02:db:c0:3e:8e" rssi "-58" channel "1" authentication mode "4"
2 th AP's ssid "" bssid "aa:02:db:b0:3e:8e" rssi "-58" channel "1" authentication mode "4"
3 th AP's ssid "ChinaNet-kRAH" bssid "f0:92:b4:a6:03:91" rssi "-63" channel "1" authentication mode "4"
4 th AP's ssid "ChinaNet-HSxa" bssid "5c:a4:a4:0a:54:0d" rssi "-63" channel "1" authentication mode "4"
5 th AP's ssid "ChinaNet-sqJr" bssid "c8:50:e9:bb:f5:0a" rssi "-75" channel "4" authentication mode "4"
6 th AP's ssid "360WiFi-B12C98" bssid "2c:61:04:b1:2c:98" rssi "-76" channel "11" authentication mode "4"
7 th AP's ssid "360WiFi-5CFEC8" bssid "c8:5b:a0:5c:fe:c8" rssi "-77" channel "1" authentication mode "4"
8 th AP's ssid "" bssid "da:35:38:70:23:8e" rssi "-79" channel "6" authentication mode "0"
```
I don't know exactly where the problem is, and I can't find the reason through strace appending. If anyone encounters a similar problem, please let me know the solution, thank you!

codebug8, 2022-03-11:
I re-referenced the operation steps of the host bluetooth and found the reason. I need to execute bluetoothd -d & in the background first.
```
# /usr/libexec/bluetooth/bluetoothd -d &
# hciconfig -a
hci0:   Type: Primary  Bus: SPI
        BD Address: 84:F7:03:13:79:86  ACL MTU: 251:12  SCO MTU: 0:0
        DOWN
        RX bytes:478 acl:0 sco:0 events:84 errors:0
        TX bytes:202 acl:0 sco:0 commands:0 errors:0
        Features: 0x00 0x00 0x00 0x00 0x60 0x00 0x00 0x00
        Packet type: DM1 DH1 HV1
        Link policy:
        Link mode: SLAVE ACCEPT

[bluetooth]# scan on
Discovery started
[CHG] Controller 84:F7:03:13:79:86 Discovering: yes
[NEW] Device 73:7E:83:9A:92:1B 73-7E-83-9A-92-1B
[NEW] Device 60:40:FB:50:C2:A9 60-40-FB-50-C2-A9
[NEW] Device 78:7E:F0:83:53:CA 78-7E-F0-83-53-CA
[NEW] Device 46:6C:CC:34:73:01 46-6C-CC-34-73-01
[NEW] Device 6C:E3:99:9B:EC:B4 6C-E3-99-9B-EC-B4
[CHG] Device 46:6C:CC:34:73:01 RSSI: -80
[CHG] Device 73:7E:83:9A:92:1B RSSI: -85
[CHG] Device 73:7E:83:9A:92:1B RSSI: -75
[CHG] Device 60:40:FB:50:C2:A9 RSSI: -73
[CHG] Device 73:7E:83:9A:92:1B RSSI: -66

```

mantriyogesh, 2022-03-11:
Hello Yuxuan,

Thanks for detailed log and appreciate your efforts to debug.

I think already you are using `bluetoothctl` tool, which is good move. As `hcitool` has been slowly [deprecated](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1870837), please refrain using it.

If you still face issue, It is worth removing and reinstalling the bluez related pakages using [troubleshoot](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/Troubleshoot.md) -> [4. Bluetooth does not work](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/Troubleshoot.md#4-bluetooth-does-not-work)

Please help closing bug, if you are not facing anymore issues.
Thanks!

codebug8, 2022-03-11:
thank you for your reply.

closed 2022-03-11 by codebug8