espressif/esp-hosted #113: ESP32-WROOM-32E Custom Mac Address is not considered

ublsoftware, 2022-03-21:
Hi,
I set the mac address with the instruction:
espefuse.py --port com7 burn_custom_mac 70:B3:D5:DD:4F:FD
`with 
espefuse.py summary --port COM7 --baud 115200

Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):                        BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                                 Voltage reference calibration                      = 1142 R/W (0b00110)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                           Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = True R/W (0b1)
XPD_SDIO_REG (BLOCK0):                             If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = True R/W (0b1)
XPD_SDIO_TIEH (BLOCK0):                            If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 3.3V R/W (0b1)
CLK8M_FREQ (BLOCK0):                               8MHz clock freq override                           = 56 R/W (0x38)
SPI_PAD_CONFIG_CLK (BLOCK0):                       Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):                         Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):                         Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):                        Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):                       Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):                        Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                                   Efuse write disable mask                           = 0 R/W (0x0000)
RD_DIS (BLOCK0):                                   Efuse read disable mask                            = 0 R/W (0x0)
CODING_SCHEME (BLOCK0):                            Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                               Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                                      Factory MAC Address
   = 8c:4b:14:91:b2:a8 (CRC 0x45 OK) R/W
MAC_CRC (BLOCK0):                                  CRC8 for factory MAC address                       = 69 R/W (0x45)
CHIP_VER_REV1 (BLOCK0):                            Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                            Silicon Revision 2                                 = True R/W (0b1)
CHIP_VERSION (BLOCK0):                             Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                             Chip package identifier                            = 1 R/W (0b001)
MAC_VERSION (BLOCK3):                              Version of the MAC field                           = Custom MAC in BLOCK3 R/W (0x01)
CUSTOM_MAC (BLOCK3):                               Custom MAC
   = 70:b3:d5:dd:4f:fd (CRC 0x1a OK) R/W
CUSTOM_MAC_CRC (BLOCK3):                           CRC of custom MAC                                  = 26 R/W (0x1a)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 0 R/W (0b0000000)
UART_DOWNLOAD_DIS (BLOCK0):                        Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):                       Flash encryption config (key tweak bits)           = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE (BLOCK0):                    Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                               Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                               Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                             Disable JTAG                                       = False R/W (0b0)
DISABLE_DL_ENCRYPT (BLOCK0):                       Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):                       Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):                         Disable flash cache in UART bootloader             = False R/W (0b0)
BLOCK1 (BLOCK1):                                   Flash encryption key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK2 (BLOCK2):                                   Secure boot key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK3 (BLOCK3):                                   Variable Block 3
   = 1a 70 b3 d5 dd 4f fd 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) set to 3.3V by efuse.`

------------------------------------------------------------------------------------------------
on module ESP32-WROOM-32E:

 (29) boot: ESP-IDF v4.4-351-g121ddb87e5 2nd stage bootloader
I (29) boot: compile time 13:38:12
I (29) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  3 factory          factory app      00 00 00010000 00100000
I (100) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (108) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (115) boot: End of partition table
I (120) boot: Defaulting to factory image
I (124) boot_comm: chip revision: 3, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1d354h (119636) map
I (183) esp_image: segment 1: paddr=0002d37c vaddr=3ffbdb60 size=02c9ch ( 11420) load
I (188) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=90a68h (592488) map
I (403) esp_image: segment 3: paddr=000c0a90 vaddr=3ffc07fc size=028d0h ( 10448) load
I (408) esp_image: segment 4: paddr=000c3368 vaddr=40080000 size=1d444h (119876) load
I (458) esp_image: segment 5: paddr=000e07b4 vaddr=50000000 size=00010h (    16) load
I (473) boot: Loaded app from partition at offset 0x10000
I (473) boot: Disabling RNG early entropy source...
I (485) cpu_start: Pro cpu up.
I (485) cpu_start: Starting app cpu, entry point is 0x40081280
I (0) cpu_start: App cpu up.
I (501) cpu_start: Pro cpu start user code
I (501) cpu_start: cpu freq: 240000000
I (501) cpu_start: Application information:
I (506) cpu_start: Project name:     network_adapter
I (511) cpu_start: App version:      release/v0.4
I (517) cpu_start: Compile time:     Mar 15 2022 13:38:05
I (523) cpu_start: ELF file SHA256:  7fc3c228d44b8ecb...
I (529) cpu_start: ESP-IDF:          v4.4-351-g121ddb87e5
I (535) heap_init: Initializing. RAM available for dynamic allocation:
I (542) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (548) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (554) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (560) heap_init: At 3FFD2910 len 0000D6F0 (53 KiB): DRAM
I (567) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (573) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (579) heap_init: At 4009D444 len 00002BBC (10 KiB): IRAM
I (587) spi_flash: detected chip: generic
I (590) spi_flash: flash io: dio
I (595) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (609) NETWORK_ADAPTER: *********************************************************************
I (615) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.4
I (623) NETWORK_ADAPTER:                 Transport used :: SDIO
I (631) NETWORK_ADAPTER: *********************************************************************
I (637) NETWORK_ADAPTER: Supported features are:
I (643) NETWORK_ADAPTER: - WLAN over SDIO
I (645) ESP_BT: - BT/BLE
I (649) ESP_BT:    - HCI Over SDIO
I (651) ESP_BT:    - BT/BLE dual mode
I (663) BTDM_INIT: BT controller compile version [5688ed5]
I (663) system_api: Base MAC address is not set
I (665) system_api: read default base MAC address from EFUSE
I (671) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (993) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 8c-4b-14-91-b2-aa

mantriyogesh, 2022-03-22:
Hello @ublsoftware 

Overall there are multiple MAC addresses ESP32 like for Sta, softap and bluetooth, etc check [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html#_CPPv414esp_mac_type_t)
Further to get the MAC addresses info, use [esp_read_mac()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html#_CPPv412esp_read_macP7uint8_t14esp_mac_type_t) API
Please `do not compare` your fused custom MAC address with `Bluetooth MAC addr`.

Anyway, in case if it is useful, you can set the MAC address of ESP-Hosted specific interface like `ethsta0` or `ethap0`.
Please use [wifi_set_mac()](https://github.com/espressif/esp-hosted/blob/master/docs/c_api.md#22-int-wifi_set_mac-int-mode-char-mac) to set MAC address of the interface once up.
Please have a look at [wifi_set_mac for station mode](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/c_support/test_api.c#L110). With this API, you may avoid flashing custom MAC.

mantriyogesh, 2022-03-28:
Hello @ublsoftware 

Would like to listen from you! Any further updates?