espressif/esp-hosted #114: WIFI Disconnnetion status not available and its not reconnecting automatically while using esp host control

vijay-yaga, 2022-03-28:
Hi we are working on the esp hosted driver we have succesfully connected using esp hostcontrol c file test.c
but we are unable to know the status of WIFI disconnection and we are unable to connect to same ssid and password when wifi is available 
 kindly can you help me rresolve this issue

thank you
Regards
vijay

mantriyogesh, 2022-03-29:
Hello @vijay-yaga 

We have developed the control path events for such scenarios.
It will shortly be pushed out to github, I would suggest you to wait for another day or so.

What are these events?
Upon some interesting condition like disconnection in station mode happens, a event message will be sent to host.
Upon receiving such event, host can do expected things (open for user what to do) like re-connection etc.

mantriyogesh, 2022-03-29:
Hello @vijay-yaga,

The GitHub page is updated for above change.

Please visit https://github.com/espressif/esp-hosted and use master branch to test the changes.
As there are multiple optimizations and code changes, I would suggest going through [control path documentation](https://github.com/espressif/esp-hosted#3-control-path) to get to know info like how the events are expected to be subscribed etc.

vijay-yaga, 2022-03-31:
Hi @mantriyogesh 
we ported the master hosrted driver and c support file to our board
before with release v4.0 its all working fine but now with new master branch i am unable to connect the wifi to my ssid
we have release 4 firmware in esp32c3 module
error liog are shown as below:

$ insmod ../../bsp/esp32_spi_new_event.ko
esp32_spi: loading out-of-tree module taints kernel.
ESP32 peripheral is registered to SPI bus [2],chip select [0], SPI Clock [10]
 ./test.out sta_connect
Enable heartbeat with duration 2
Received INIT event from ESP32 peripheral0

EVENT: 2
EVENT: 1
ESP Reconfigure SPI CLK to 30 MHz
ESP32 peripheral is registered to SPI bus [2],chip select [0], SPI Clock [30]
EVENT: 0
ESP peripheral capabilities: 0x68
Control response timed out after
Received INIT event from ESP32 peripheral 30 sec
Respons
e not received
EVENT: 2
EVENT: 1
ESP Reconfigure SPI CLK to 30 MHz
ESP32 peripheral is registered to SPI bus [2],chip select [0], SPI Clock [30]
EVENT: 0
ESP peripheral capabilities: 0x68

Requested operation complete
Sleeping for some time just to showcase heartbeat
Illegal instruction

kindly give me the solution

mantriyogesh, 2022-03-31:
Have you used same commit at both places, at esp and linux?

Also, please try to mimic rpi_init.sh to load the driver.
After loading new module, please rebuild the application. And keep at least 3 sec gap after loading driver to run the app.

mantriyogesh, 2022-03-31:
Still not working despite above trials?
please get us the detailed log..

vijay-yaga, 2022-04-01:
Hi @mantriyogesh 
thanks for your quick response
i am attching the esp32 module error log its giving protocomm_pserial_common_handler failed
so the module is not responding to driver
please find the error log attachement 
and we have compile host driver and host control from master branch 
but we have loaded the firware version 4.0 for esp32
[esp32_release4_log.txt](https://github.com/espressif/esp-hosted/files/8395413/esp32_release4_log.txt)

mantriyogesh, 2022-04-01:
Okay.
The problem is that for code changes done as part of control path optimizations are not compatible with old ESP32 firmwares. There are changes in protobuf messages.

In short, you need all to point same commit, ESP32 and host.
Considering this, you have two options to proceed,
- 1. Use release 0.4 at both, ESP and host (without events for which bug was raised) and wait for next release.
- 2. Use the master at both, ESP and host
  - In this case you have to flash the ESP32 instead of using release, which should be fairly simple, using documentation.

mantriyogesh, 2022-04-04:
@vijay-yaga 
Any updates?

vijay-yaga, 2022-04-05:
Hi @mantriyogesh 

we have succesfully compiled and loaded the master C3 firmware and we also compiled master branch hosted driver and host controller c support test.out
we have loaded the module and reset the esp32 module and after we have run the below commands 
  $ mknod /dev/esps0 c 221 0
  $ chmod 666 /dev/esps0
after that i have executed the test.out with arg sta_connect
and its not connecting to wifi i am attaching the error log belo
[esp32_hosted_error_log.txt](https://github.com/espressif/esp-hosted/files/8417282/esp32_hosted_error_log.txt)
w

mantriyogesh, 2022-04-06:
Hello @vijay-yaga,

Thank you for the all logs.

The ESP log says it is connected to sta, 'yaga'.
Overall observations:
1. First SPI transaction is complete ('EVENT' in the log). This is Rx path wrt host.
2. Multiple control path commands worked (Tx,Rx), but sta_connect had not received the response.
3. I see 'Control response timed out after 30 sec', which is a first problem.
4. 'Illegal instruction' in the log is second problem.
5. 'test.out' is expected to be run as superuser, which I hope you are doing.

Can you please let us know -
1. If there is some delay after loading module and running `$ sudo ./test.out <some_command>`
2. If commenting line at https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/c_support/test_utils.c#L717 and rebuild, rerun of test.out helps?
3. Is this Raspberry Pi?
4. Output of `$ uname -a` command at host
5. Can you please attach the `$ git diff` at both the sides, ESP and Linux?

vijay-yaga, 2022-04-06:
Hi @mantriyogesh 

1. 1.If there is some delay after loading module and running $ sudo ./test.out <some_command>

A.we are manually loading the module and after that we are executing the test.out(1 min max)
2.If commenting line at https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/c_support/test_utils.c#L717 and rebuild, rerun of test.out helps?
A.we have commented the 717 line but still same issue occures
3.Is this Raspberry Pi?
A.No we are working on our own custom board based on NUC980
4.Output of $ uname -a command at host
A.Linux iot 4.4.289 #95 PREEMPT Sat Mar 26 11:20:29 IST 2022 armv5tejl GNU/Linux

1. 

1. 3.

mantriyogesh, 2022-04-06:
Okay. Understood.
From the log,
```
 ./test.out sta_connect
Enable heartbeat with duration 2
Received INIT event from ESP32 peripheral0

EVENT: 2
EVENT: 1
ESP Reconfigure SPI CLK to 30 MHz
ESP32 peripheral is registered to SPI bus [2],chip select [0], SPI Clock [30]
EVENT: 0
ESP peripheral capabilities: 0x68
Control response timed out after 30 sec
Response not received
1970-01-01 00:04:01 >  App EVENT: ESP INIT
Requested operation complete
Sleeping for some time just to showcase heartbeat
Illegal instruction
```

- Something which bothers me is that why `Received INIT event from ESP32 peripheral0` is received after running ./test.out sta_connect. \
This event would come only after getting ESP **reset** or **crash**.

- Coming to `Illegal instruction`,  I am really unsure what is this error. Is this NULL pointer access or mem alignment issue or something else? Any prior experience about this in your platform would be really helpful.

- Overall, I will add a debug logs and provide a patch to get some additional log capture to trace the issue.
Can you please send us the your code change over Hosted (also base commit) at both ESP and host side? I will try to add debug prints over that, to understand the issue.

mantriyogesh, 2022-04-07:
Can you please lower the frequency just to try if the event is getting received after above points?
https://github.com/espressif/esp-hosted/blob/master/host/linux/host_driver/esp32/spi/esp_spi.c#L352 ->

Change
```
esp_board.max_speed_hz = spi_clk_mhz * NUMBER_1M;
```
to 
```
esp_board.max_speed_hz = 1 * NUMBER_1M;
```

mantriyogesh, 2022-04-07:
- 1.  Source of `Illegal instruction` has to be checked.
- 2. Why ESP got reset/ rebooted after running `sudo ./test sta_connect` → Make sure there is good time gap of > 3sec in between insmod and `sudo ./test.out sta_connect`
- 3. Suggestion: It will be better to create one such script for your linux which mimics operations in `rpi_init.sh`.
- 4. Over hosted commit 787f474e13753c541bd0ebc2e1a5eeffe7c30bd4 (*i.e.* master),
I had added some debug prints in 
[github_vijay_114_debug_7apr.patch.gz](https://github.com/espressif/esp-hosted/files/8439807/github_vijay_114_debug_7apr.patch.gz)
- 5. Please reboot, reload the module, rebuild the test.out with expected ssid, pwd and get the output.
<details>
<summary><b><u>Success log</u></b></summary>
<p>
pi@raspberrypi:~/hosted/host/linux/host_control/c_support $ vim ctrl_config.h
pi@raspberrypi:~/hosted/host/linux/host_control/c_support $
pi@raspberrypi:~/hosted/host/linux/host_control/c_support $ make clean
rm -f *.out *.o
pi@raspberrypi:~/hosted/host/linux/host_control/c_support $ make
gcc -Wall -I/home/pi/hosted/host/linux/host_control/c_support/../../../../common/protobuf-c -I/home/pi/hosted/host/linux/host_control/c_support/../../../../common/include -I/home/pi/hosted/host/linux/host_control/c_support/../../../control_lib/include -I/home/pi/hosted/host/linux/host_control/c_support/../../../control_lib/src/include -I/home/pi/hosted/host/linux/host_control/c_support/../../../virtual_serial_if/include -I/home/pi/hosted/host/linux/host_control/c_support/../../../components/include -I/home/pi/hosted/host/linux/host_control/c_support/../../port/include -I. /home/pi/hosted/host/linux/host_control/c_support/../../../../common/protobuf-c/protobuf-c/protobuf-c.c /home/pi/hosted/host/linux/host_control/c_support/../../../../common/esp_hosted_config.pb-c.c /home/pi/hosted/host/linux/host_control/c_support/../../../control_lib/src/ctrl_core.c /home/pi/hosted/host/linux/host_control/c_support/../../../control_lib/src/ctrl_api.c /home/pi/hosted/host/linux/host_control/c_support/../../../virtual_serial_if/src/serial_if.c /home/pi/hosted/host/linux/host_control/c_support/../../../components/src/esp_queue.c /home/pi/hosted/host/linux/host_control/c_support/../../port/src/platform_wrapper.c ./test_utils.c -lpthread -lrt test.c -o test.out
pi@raspberrypi:~/hosted/host/linux/host_control/c_support $ sudo ./test.out sta_connect
set_event_callback:1055 event: 301
set_event_callback:1055 event: 302
set_event_callback:1055 event: 303
set_event_callback:1055 event: 304
ctrl_app_send_req:1157 Sending req 121
Enable heartbeat with duration 20
CB available for 302
config_heartbeat:158 sync proc for req 121
process_ctrl_rx_msg:657 msg type: 0x2 msg id: 0xdd
ctrl_app_parse_resp:294 msgid 221
Success for ctrl resp msg id 221
process_ctrl_rx_msg:746 sync proc for 221
ctrl_rx_ind:616
ctrl_rx_thread:846 processed ctrl rx msg
get_response:923 data: 0x764005b8
ctrl_app_resp_callback:452 processing resp 221
Heartbeat operation successful
ctrl_app_send_req:1157 Sending req 107
wifi_connect_ap:104 sync proc for req 107
process_ctrl_rx_msg:657 msg type: 0x2 msg id: 0xcf
ctrl_app_parse_resp:294 msgid 207
ctrl_app_parse_resp:422 Success
Success for ctrl resp msg id 207
process_ctrl_rx_msg:746 sync proc for 207
ctrl_rx_ind:616
ctrl_rx_thread:846 processed ctrl rx msg
get_response:923 data: 0x764007c0

Requested operation complete
Sleeping for some time just to showcase heartbeat

process_ctrl_rx_msg:657 msg type: 0x3 msg id: 0x12e
CB available for 302
process_ctrl_rx_msg:673: event cb available
EVENT: Heartbeat
ctrl_app_event_callback:105 processing event 302
2022-04-07 05:59:04 >  App EVENT: Heartbeat event [0]
ctrl_rx_thread:846 processed ctrl rx msg
process_ctrl_rx_msg:657 msg type: 0x3 msg id: 0x12e
CB available for 302
process_ctrl_rx_msg:673: event cb available
EVENT: Heartbeat
ctrl_app_event_callback:105 processing event 302
2022-04-07 05:59:29 >  App EVENT: Heartbeat event [1]
ctrl_rx_thread:846 processed ctrl rx msg
ctrl_app_send_req:1157 Sending req 121
Disable Heartbeat
config_heartbeat:158 sync proc for req 121
process_ctrl_rx_msg:657 msg type: 0x2 msg id: 0xdd
ctrl_app_parse_resp:294 msgid 221
Success for ctrl resp msg id 221
process_ctrl_rx_msg:746 sync proc for 221
ctrl_rx_ind:616
ctrl_rx_thread:846 processed ctrl rx msg
get_response:923 data: 0x76400a78
ctrl_app_resp_callback:452 processing resp 221
Heartbeat operation successful
reset_event_callback:1063 event: 301
set_event_callback:1055 event: 301
reset_event_callback:1063 event: 302
set_event_callback:1055 event: 302
reset_event_callback:1063 event: 303
set_event_callback:1055 event: 303
reset_event_callback:1063 event: 304
set_event_callback:1055 event: 304
Exiting..pi@raspberrypi:~/hosted/host/linux/host_control/c_support $
</p>
</details>

vijay-yaga, 2022-04-07:
Hi @mantriyogesh 

after making changes in using patch file 

./test.out sta_connect
set_event_callback:1055 event: 301
set_event_ca
Received INIT event from ESP32 peripheralllback:1055 even
t: 302
set_evenEVENT: 2
t_callback:1055 EVENT: 1
event: 303
set_ESP Reconfigure SPI CLK to 30 MHz
event_callback:1055 event: 304
ctrl_app_send_req:1157 Sending req 121
Enable heartbeat with duration 20
CB available for 302
config_heartbeat:158 sync proc for req 121
ESP32 peripheral is registered to SPI bus [2],chip select [0], SPI Clock [30]
EVENT: 0
ESP peripheral capabilities: 0x68
Control response timed out after 30 sec
Response not received
ctrl_app_resp_callback:439 Failure
ctrl_app_send_req:1157 Sending req 107
wifi_connect_ap:104 sync proc for req 107
process_ctrl_rx_msg:657 msg type: 0x3 msg id: 0x12d
CB available for 301
process_ctrl_rx_msg:673: event cb available
EVENT: ESP INIT
ctrl_app_event_callback:105 processing event 301
1970-01-01 00:04:35 >  App EVENT: ESP INIT
ctrl_rx_thread:846 processed ctrl rx msg
Control response timed out after 30 sec
Response not received

Requested operation complete
Sleeping for some time just to showcase heartbeat
ctrl_app_send_req:1157 Sending req 121
Disable Heartbeat
config_heartbeat:158 sync proc for req 121
process_ctrl_rx_msg:657 msg type: 0x2 msg id: 0xdd
ctrl_app_parse_resp:294 msgid 221
Success for ctrl resp msg id 221
process_ctrl_rx_msg:746 sync proc for 221
ctrl_rx_ind:616
ctrl_rx_thread:846 processed ctrl rx msg
get_response:923 data: 0x15f30a0
ctrl_app_resp_callback:453 processing resp 221
Heartbeat operation successful
reset_event_callback:1063 event: 301
set_event_callback:1055 event: 301
reset_event_callback:1063 event: 302
set_event_callback:1055 event: 302
reset_event_callback:1063 event: 303
set_event_callback:1055 event: 303
reset_event_callback:1063 event: 304
set_event_callback:1055 event: 304
Exiting..#

mantriyogesh, 2022-04-07:
Hello @vijay-yaga 
Can you please confirm if data ready pin is correctly configured, intact and not loose?

mantriyogesh, 2022-04-11:
Anything worked ahead @vijay-yaga ?

vijay-yaga, 2022-04-11:
Hi @mantriyogesh
its connecing to wifi now we are working on callback events

mantriyogesh, 2022-04-11:
Sure. Can you please educate us what was the issue, might help us in similar situation ahead.

Also please do connect here if you have issues ahead.

vijay-yaga, 2022-04-11:
Hi @mantriyogesh 
thanks for your support data ready pin and Irq pins we are used different we make changes in driver file so its working now

jcormier, 2022-04-28:
> In short, you need all to point same commit, ESP32 and host.

Thanks. Just got tripped up by this. Ended up checking out the release/v0.4 tag.

Is it recommended to try the master branch for both ESP and ARM code.  Or stick with the release?

mantriyogesh, 2022-04-29:
Overall, both approaches hold good. Both 0.4 (being latest release) and master are supported.

- In case of [0.4](https://github.com/espressif/esp-hosted/releases/tag/release%2Fv0.4) Binaries for ESP and [associated git commit](https://github.com/espressif/esp-hosted/commit/74e707b604280ecb2eafa60ba5ed6993ce57b457) is to be used for ARM side kernel module building.
- In case of master, ESP code is to be built, and same code is to be used for ESP and ARM.

mantriyogesh, 2022-05-04:
@vijay-yaga,
Could you please help to close the issue(s) if they are solved as per your expectations?

closed 2022-05-04 by vijay-yaga