espressif/esp-hosted #115: Need  help for adding log in  ESP hosted

anoop18pant, 2022-04-06:
hi,

We are using ESP hosted solution and with our custom board and ESP32C3 devkit -M1 we are able to communicate successfully.

But now we have assembled the ESP32 module on our custom board but after reset we are not able to get the INIT event.
So we want help from you to put log in the function where ESP host driver received data from ESP slave device.
Can you please point us the name of the function so that we can check the data if it corrupted or not?

Your help will be appreciated.

Thanks

mantriyogesh, 2022-04-07:
Hello @anoop18pant 

No worries, we will support to get hosted working.
the function you are lookin at host is: [process_priv_communication](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_driver/esp32/main.c#L464) -> [process_event](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_driver/esp32/main.c#L369) -> [process_init_event](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_driver/esp32/spi/esp_spi.c#L182)

Can you please confirm:
1. `uname -a` output
2. Have you got the Device tree for spi configured-> user space spidev driver working -> used spidev disabler to see `ls -lrt /dev/spi*` does not list that driver -> Used porting guide to setup expected GPIOs like Reset Pin, Handshake Pin, Data Ready pin?
3. Are the SPI pin numbers configured exactly as per the documentation?

mantriyogesh, 2022-04-07:
After above points, Can you please lower the frequency just to try if the event is getting received?
https://github.com/espressif/esp-hosted/blob/master/host/linux/host_driver/esp32/spi/esp_spi.c#L352 →

Change
```
esp_board.max_speed_hz = spi_clk_mhz * NUMBER_1M;
```
to 
```
esp_board.max_speed_hz = 1 * NUMBER_1M;
```

mantriyogesh, 2022-04-08:
Hello @anoop18pant,

Were you able to check on this?

anoop18pant, 2022-04-08:
hi Yogesh,

Sorry yesterday was quiet busy in debugging this issue finally we could find that its related to soldering of module on custome board.
Will update you soon once i will get confirmation its due to hardware soldering.

mantriyogesh, 2022-05-09:
Can we please this issue as thread has moved forward to new issue https://github.com/espressif/esp-hosted/issues/120

closed 2022-05-09 by anoop18pant