espressif/esp-hosted #116: RP2040 support?

beyonlo, 2022-04-11:
Hello.

Are there intention to support the RP2040 to the esp-hosted? 

Will be nice access the WiFi from the RP2040.

Thank you.

mantriyogesh, 2022-04-11:
Did you mean [this](https://www.raspberrypi.com/documentation/microcontrollers/rp2040.html) MCU device?
We had thought for this, but this is definitely not happening in the immediate future.

Anyway, That should not stop you from using RP2040. You can refer the MCU based code. ESP32 side code will be just the same. For Host side, it should be easy to port to RP2040.

beyonlo, 2022-04-11:
> Did you mean [this](https://www.raspberrypi.com/documentation/microcontrollers/rp2040.html) MCU device? We had thought for this, but this is definitely not happening in the immediate future. [...]

Yes, I'm talking about that, the RaspberryPi MCU RP2040.

Thank you!