espressif/esp-hosted #118: ESP32-C3 Fails to boot

peterfox, 2022-04-14:
I've flashed my esp32-c3 board with the hosted 0.4 firmware but it fails with the following:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d1546
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16cc
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d40
SHA-256 comparison failed:
Calculated: 8ac330b1134deb0e9c8ca3dc6b602b7f2e85eeb75853cb6b9cb34623fbf69630
Expected: 47e9d520f2677a0708cbc684a7fddb3a689cc776ef3a459736cfb864cb12e3b7
Attempting to boot anyway...
entry 0x403ce000
I (53) boot: ESP-IDF v4.4-rc1-8-g0b46ac1732-dirty 2nd stage bootloader
I (53) boot: compile time 19:43:58
I (54) boot: chip revision: 3
I (57) boot.esp32c3: SPI Speed      : 80MHz
I (62) boot.esp32c3: SPI Mode       : DIO
I (66) boot.esp32c3: SPI Flash Size : 2MB
I (71) boot: Enabling RNG early entropy source...
E (77) flash_parts: partition 4 invalid - offset 0x110000 size 0x100000 exceeds flash chip size 0x200000
E (87) boot: Failed to verify partition table
E (92) boot: load partition table error!
```

I assume this is because there isn't enough space as my chip only has 2MB of flash and needs more than that to run?

I don't need the SPI functionality, just the BLE over UART. Is there a known way to drop the SPI functionality and compile the hosted firmware with only the UART BLE functionality?

Command I used after erasing the flash in case that's the issue:

```shell
esptool.py --chip esp32c3 -p /dev/cu.usbserial-1430 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size detect \
0x0 esp_hosted_bootloader_esp32c3_spi_uart_v0.4.bin \
0x8000 esp_hosted_partition-table_esp32c3_spi_uart_v0.4.bin \
0xd000 esp_hosted_ota_data_initial_esp32c3_spi_uart_v0.4.bin \
0x10000 esp_hosted_firmware_esp32c3_spi_uart_v0.4.bin
```

mantriyogesh, 2022-04-18:
Hello @peterfox,

Thank you for your exact query.

Although I think only UART is definitely doable, but we have never tried only UART combination.
Give us some time, we will try this combination at our place and let you know if that works fine.

mantriyogesh, 2022-04-18:
Hello @peterfox 
If your use case is only to use the UART, you can actually skip the ESP-Hosted solution at all and use the ESP IDF [hci example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/hci/controller_hci_uart_esp32c3_and_esp32s3) directly. This will set up the ESP in HCI controller only mode. You just have to connect the UART pins at Linux host and set up the pins in UART mode similar to [bt_init()](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/rpi_init.sh#L76)

Hope this helps.

peterfox, 2022-04-18:
Awesome @mantriyogesh , I'll give that a go and see how it turns out.

mantriyogesh, 2022-05-09:
Hello @peterfox 

Can you please update your observations?
Also can you please close the issue?

peterfox, 2022-05-09:
@mantriyogesh sadly I struggled to make the HCI examples work fully.

closed 2022-05-09 by peterfox

mantriyogesh, 2022-05-09:
@peterfox 
Can you please state exact issues you had faced, we can try to help you out.
With the issue, please mention exact IDF release used, IDF commit and HCI logs if you can get, will be much helpful.