espressif/esp-hosted #121: ESP32-S3 supported？

Xiehanxin, 2022-05-23:


mantriyogesh, 2022-05-23:
Hello @Xiehanxin,

**ESP32-S3** is **not** yet supported. We think to support it but yet we do not have timeline planned for it as if now.

Xiehanxin, 2022-05-23:
> Hello @Xiehanxin,
> 
> **ESP32-S3** is **not** yet supported. We think to support it but yet we do not have timeline planned for it as if now.

OK，thanks

mantriyogesh, 2022-07-07:
Hello @Andreas-Hofer,

Understood clearly.
I assume you are using Linux.

Some answers:
1. We have ported the ESP32-S3 solution already for Wi-Fi on SPI transport. It is in review & testing cycle. We expect it to be out by next week.
2. Because of our other priority work, it got delayed. As such, there are no technical issues.
3. Porting would be very similar to existing code. In fact, the Linux kernel driver doesn't much change (apart from addition of chip).
4. If in case you had already using Linux solution which working with ESP32 on SPI transport, in no or minimal efforts you can use it on ESP32-S3.

Question:
1. Can you please state the throughput requirements?
2. What is the Linux platform you would be using, and what would be the kernel version?

I will notify here once we merge Linux based SPI solution for ESP32-S3.

Andreas-Hofer, 2022-07-07:
Hi mantriyogesh,

thank you for your quick reaction. I want to quickly answer to your questions:

1. Today we consider high throughput as having a lower priority. As long as throughput is more or less on par with ESP32-S2 over SPI, it is acceptable for our application. At this time, we use ESP32 with SDIO transport interface since we found the higher throughput of ESP32 with SDIO being more attractive at first, but in the meantime we consider long-term availability as equally important. This is why now we would accept the lower throughput of SPI as documented for ESP32-S2, and presumably ESP32-S3.
2. The Linux platform used in our product is STM32MP153C running Yocto, Linux kernel version is either 5.4 or 5.10

You did not comment on BLE support for ESP32-S3 yet. Will BLE be supported as well, and when will BLE support be available?

We will have to decide whether we stick to ESP32 + SDIO or if we change our design to ESP32-S3 + SPI.

Regards
Andreas

Andreas-Hofer, 2022-07-11:
Hi @mantriyogesh 

great, nice to hear. I will then wait for you "... notify here once we merge Linux based SPI solution for ESP32-S3".

I don't want to adapt our design for the ESP32-S3 before we are sure that it supports esp-hosted.

Best Regards
Andreas Hofer

mantriyogesh, 2022-07-26:
Hello @Andreas-Hofer 

Apologize for the delay and appreciate your patience.
We have added ESP32-S3 `SPI` & `SPI+UART` support under commit, https://github.com/espressif/esp-hosted/commit/4688eacf152f294e257c1e5091af589dc9fb05f9 

Please flash the ESP32-S3 with source option for now.

Please beware, the ESP-Hosted repo page should look a bit differently, as another variant of our solution, `ESP-Hosted-NG` is added in release. This Next Generation is targeted for Linux with wpa_supplicant (using cfg802.11) support.
Prior solution is continued over branch, [ESP-Hosted_MCU_Host](https://github.com/espressif/esp-hosted/commits/ESP-Hosted_MCU_Host).

Andreas-Hofer, 2022-07-27:
Hi @mantriyogesh

very good, thank you so much for releasing the ESP32-S3 port. Just some questions for clarification:

- Does the ESP32-S3 port support WiFi and Bluetooth?
- ESP32-S3 SPI means "use SPI for WiFi **AND** for Bluetooth?"
- ESP32-S3 SPI+UART means "use SPI for WiFi and use UART for Bluetooth?"
- Is the release that supports ESP32-S3 legacy esp-hosted or is it already the "new generation" esp-hosted-ng?

I will discuss with my colleagues now how to proceed. Since our ESP32 based design is quite advanced, we will have to decide whether we will switch the design to ESP32-S3 now or in 2027, when ESP32 could become obsolete. I for my part would prefer to switch to ESP32-S3 now in order to avoid re-certification of our product in 2027. But we'll see. Some colleagues are on vacation, so I don't expect a decision within the next days.

Thank you again for your effort. This is really a great project!

Regards
Andreas Hofer

mantriyogesh, 2022-07-27:
Hello @Andreas-Hofer ,

1. ESP32-S3 supports Wi-Fi and Bluetooth both.
2. Yes
3. Yes
4. ESP32-S3 support is added only into `ESP-Hosted` variant (legacy), as if now. \
    All current chipsets will be eventually added into `Esp-Hosted-NG` as well, but you can expect some time for that to happen.

Overall, we aim to support Linux on `Esp-Hosted-NG` going ahead and \
All MCUs on `ESP-Hosted`. Both the branches will proceed and would be supported.

Thank you for all your good words.
Please do connect if you face issues.

mantriyogesh, 2022-09-24:
Hello!
Any updates here? Did the things work out fine for you @Andreas-Hofer & @Xiehanxin ?

Andreas-Hofer, 2023-02-28:
Hi @mantriyogesh,

I am sorry for my late reply. I wanted to have our work stabilized before I answer you.
This is the case right now. We managed to operate a RasPI 4 and our target design (ST Microelectronics STM32MP1xx) together with the ESP32-S3 and the esp-hosted FG (First Generation) branch.

We connected the ESP32-S3 with the SPI interface of the mentioned hardware platforms. The measured performance values for the RasPI 4 match more or less your results. The performance when connected to our STM32MP1xx is unfortunately much lower, about 50% of the RasPI values. We are investigating why this is the case. Of course we know that the CPU performance of a 650MHz clocked dual-core Cortex-A7 cannot match the performance of the RasPI 4, but maybe this is not the only reason for the lower performance.

We now would have to write some kind of control application. But our software guys would prefer by far the NG (new generation) esp-hosted branch in order to be compatible with our other Linux based solutions (wpa_supplicant, iw etc). A quick try to use the RasPI4 with ESP32-S3 and the esp-hosted NG branch failed, dito with our STM32MP1xx. In this regard I therefore have some questions:

- Is there a  roadmap for the NG  supoort for the ESP32-S3?
- Can we help to make the FG support for the ESP32-S3 become true?
- Can you give us some hints which parts of the code would have to be adapted for the -S3 support?

In any case I want to thank you again for this great piece of software. Really awsome!

Andreas-Hofer, 2023-02-28:
Hi @mantriyogesh,

Great, I am pleased to hear that apparently the S3 support of the NG code is close to release.
We need the NG support for ESP32-S3, but I think we prefer the evaluation of the NG code on our hardware configuration only after it is released.

Thank you!
Andreas

delladalle, 2023-03-17:
Hello @mantriyogesh,
any news on this? So far I was just reading and following your discussion. But now I also would like to address our interest in a NG for the ESP32-S3 for quite the same reasons as Andreas-Hofer mentioned.
If possible I would to love have some firmware and check and test it out.

Dieter

Philipansari, 2023-04-21:
Hi @mantriyogesh 

are there any source files for the ESP32-S3 available yet, or do you know when they will be released?
If it is possible i would love to get the firmware files for it.

Thanks in advance!
Philip