espressif/esp-hosted #122: 0.4版本在Linux raspberrypi 5.15.32-v7+似乎没有工作（does't work）

xxzling, 2022-06-04:
我尝试在Linux raspberrypi 5.15.32-v7+内核的版本上编译host驱动，但是它似乎工作不正常，我先后尝试过使用esp32，esp32-s2在raspi-3b+，raspi-zero上构建，但是均没有成功，无论是sdio还是spi模式使用python ap_scan_list.py时均提示：
open: : No such file or directory
wifi_ap_scan_list:943 Failed to process rx data
Failed to get AP scan list
这花费了我很多时间，我尝试对sdio模式下DATA0-DATA3添加10k上拉电阻，但是同样如此
我所使用的引脚：

24 | IO15 | CS0
23 | IO14 | SCLK
21 | IO12 | MISO
19 | IO13 | MOSI
25 | GND | Ground
15 | IO2 | Handshake
13 | IO4 | Data Ready
31 | EN | ESP32 Reset

编译日志：
`Building for spi protocol
rm -f commands.so *.o
gcc -C -Wall -I/home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_control/python_support/../../../host_common/include/ -I/home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_control/python_support/../../../../common/protobuf-c/ -I/home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_control/python_support/../../../../common/include/ /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_control/python_support/../../../../common/protobuf-c/protobuf-c/protobuf-c.c \
/home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_control/python_support/../c_support/platform_wrapper.c \
/home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_control/python_support/../../../../common/esp_hosted_config.pb-c.c \
/home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_control/python_support/../../../host_common/transport_pserial.c \
-o commands.so -shared -fPIC /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_control/python_support/../../../host_common/commands.c
rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabihf- -C /lib/modules/5.15.32-v7+/build M=/home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32 clean
make[1]: 警告: jobserver 不可用: 正使用 -j1。添加 “+” 到父 make 的规则。
make[1]: 进入目录“/usr/src/linux-headers-5.15.32-v7+”
  CLEAN   /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32/Module.symvers
make[1]: 离开目录“/usr/src/linux-headers-5.15.32-v7+”
make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabihf- -C /lib/modules/5.15.32-v7+/build M=/home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32 modules
make[1]: 警告: jobserver 不可用: 正使用 -j1。添加 “+” 到父 make 的规则。
make[1]: 进入目录“/usr/src/linux-headers-5.15.32-v7+”
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
  You are using:           arm-linux-gnueabihf-gcc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
  CC [M]  /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32/esp_bt.o
  CC [M]  /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32/main.o
  CC [M]  /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32/spi/esp_spi.o
  CC [M]  /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32/esp_serial.o
  CC [M]  /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32/esp_rb.o
  LD [M]  /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32/esp32_spi.o
  MODPOST /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32/Module.symvers
  CC [M]  /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32/esp32_spi.mod.o
  LD [M]  /home/pi/Desktop/esp32_hosted/esp-hosted-release-v0.4/host/linux/host_driver/esp32/esp32_spi.ko
make[1]: 离开目录“/usr/src/linux-headers-5.15.32-v7+”
esp32 module inserted 
/dev/esps0 device created
RPi init successfully completed
`

dmesg:
'[ 2578.318255] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 2578.824414] Bluetooth: Core ver 2.22
[ 2578.824557] NET: Registered PF_BLUETOOTH protocol family
[ 2578.824568] Bluetooth: HCI device and connection manager initialized
[ 2578.824594] Bluetooth: HCI socket layer initialized
[ 2578.824610] Bluetooth: L2CAP socket layer initialized
[ 2578.824635] Bluetooth: SCO socket layer initialized
[ 2600.746598] esp32_spi: loading out-of-tree module taints kernel.
[ 2600.747982] esp_reset, ESP32: Resetpin of Host is 6
[ 2600.748131] esp_reset, ESP32: Triggering ESP reset.
[ 2600.748751] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
[ 3143.805756] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 3725.068093] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 4422.376076] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 4443.441032] esp_reset, ESP32: Resetpin of Host is 6
[ 4443.441181] esp_reset, ESP32: Triggering ESP reset.
[ 4443.441698] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
'
esplog:
'
I (305) cpu_start: Application information:
I (310) cpu_start: Project name:     network_adapter
I (316) cpu_start: App version:      release0.4-60-g22180f7
I (322) cpu_start: Compile time:     Jan 24 2022 17:37:14
I (328) cpu_start: ELF file SHA256:  3c57b77ed96c07ad...
I (334) cpu_start: ESP-IDF:          v4.2.2-529-g85a36b86d0-dirty
I (341) cpu_start: Single core mode
I (345) heap_init: Initializing. RAM available for dynamic allocation:
I (352) heap_init: At 3FFD1428 len 0002ABD8 (170 KiB): DRAM
I (3▒▒ff敹▒▒▒ At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (364) cpu_start: Pro cpu start user code
I (385) spi_flash: detected chip: generic
I (385) spi_flash: flash io: dio
I (386) cpu_start: Starting scheduler on PRO CPU.
I (389) NETWORK_ADAPTER: *********************************************************************
I (389) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.4     
I (399) NETWORK_ADAPTER:                 Transport used :: SPI                  
I (409) NETWORK_ADAPTER: *********************************************************************
I (419) NETWORK_ADAPTER: Supported features are:
I (429) NETWORK_ADAPTER: - WLAN over SPI
I (439) SPI_DRIVER: Using SPI interface
I (439) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (449) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1469) wifi:wifi driver task: 3ffdf568, prio:23, stack:6656, core=0
I (1469) system_api: Base MAC address is not set
I (1469) system_api: read default base MAC address from EFUSE
I (1469) wifi:wifi firmware version: fcafbf0
I (1479) wifi:wifi certification version: v7.0
I (1479) wifi:config NVS flash: disabled
I (1479) wifi:config nano formating: disabled
I (1489) wifi:Init data frame dynamic rx buffer num: 32
I (1489) wifi:Init management frame dynamic rx buffer num: 32
I (1499) wifi:Init management short buffer num: 32
I (1499) wifi:Init dynamic tx buffer num: 32
I (1509) wifi:Init static rx buffer size: 1600
I (1509) wifi:Init static rx buffer num: 10
I (1519) wifi:Init dynamic rx buffer num: 32
I (1519) phy_init: phy_version 1300,2887b9c,Dec 16 2020
I (1609) wifi:enable tsf
I (1609) wifi:mode : null
I (1609) NETWORK_ADAPTER: Initial set up done'

感谢回复，我希望能获得帮助。

xxzling, 2022-06-05:
At the same time, I noticed that no matter what combination is used, when ifconfig-a is used, no new network interface can be generated,but I'm sure there's no problem with the physical link @mantriyogesh

mantriyogesh, 2022-06-06:
Hello @xxzling 

Could you please try using **master** at both, ESP32 (flash manually) and Raspberry Pi and share ESP side log, ESP side sdkconfig and Raspberry Pi dmesg log?

xxzling, 2022-06-06:
Thanks for your help, I use ESP now esp_ hosted_ firmware_ esp32_ spi_ v0.4. Zip and raspberry pie master. Do I need to manually build the esp32 firmware and provide a complete log?

mantriyogesh, 2022-06-06:
Yes that is correct.
Try using master commit, build and flash esp32 firmware and use the same commit at raspberry pi to build the kernel module.

xxzling, 2022-06-06:
OK, thanks for your suggestion. I'll try and submit it later

mantriyogesh, 2022-06-06:
In case you do not want to go for building the ESP, you also use the git commit where the 0.4 release was tagged, i.e. 74e707b604280ecb2eafa60ba5ed6993ce57b457

mantriyogesh, 2022-06-23:
Is it working for you now?

mantriyogesh, 2022-08-03:
Any updates @xxzling ?