espressif/esp-hosted #128: beaglebone black couldn't receive init message from esp32 wroom

vuhailongkl97, 2022-07-08:
Hi 
I'm try to porting this project to beaglebone black board.  every thing seem ok about reset action, spi communication test. 
however when i flash built firmware to esp and load kernel module at release/v0.4. 

bbb trigger to reset esp -> ok
bbb regist network interface -> ok

esp ->bbb : init message -> nok

below is my pin connection. could you please guide me how to debug this ? thanks.

BBB pin label | BBB pin ID | ESP32 pin ID | ESP32 pin function
-- | -- | -- | --
P9_23 | 49 | 21 | GPIO_HANDSHAKE
P9_25 | 117 | 22 | GPIO_DATA_READY
SPI0_CS0 | P9_17 | 5 | CS
SPI0_SCLK | P9_22 | 18 | SCLK
SPI0_MISO D0 | P9_21 | 19 | MISO
SPI0_MOSI D1 | P9_18 | 23 | MOSI
P9_12 | 60 | RESET | RESET

mantriyogesh, 2022-07-08:
Hello @vuhailongkl97 

Good amount of trials you have done. Thank you for stating all the tests and results.
Caution! Long reply ahead with many hyperlinks.

Understanding from your log:
1. You are using ESP Binaries from https://github.com/espressif/esp-hosted/releases/tag/release%2Fv0.4 and using commit **https://github.com/espressif/esp-hosted/commit/74e707b604280ecb2eafa60ba5ed6993ce57b457 for building the host kernel module** (basically same commit of release). It is important to keep ESP and host at same commit, otherwise software will not sync and fail
2. You had tested SPI and works fine. I assume you had tested the 'spidev' or some other user-space application with all existing pin configurations and SPI pins (CLK,MISO,MOSI,CS) are working for you. If not, it is better to test spidev first.
3. When you use [spidev disabler](https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/host/linux/host_control/rpi_init.sh#L143), and [build & load kernel module](https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/host/linux/host_control/rpi_init.sh), the ESP resets correctly. But the first transaction doesn't happen.

Some pointers:
1. Are `GPIO_HANDSHAKE` and `GPIO_DATA_READY` configured correctly (as per your expected numbers) at
  1.1. Software level:
  ESP
  https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/esp/esp_driver/network_adapter/main/spi_slave_api.c#L107-L108
  and at host
  https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/host/linux/host_driver/esp32/spi/esp_spi.h#L22-L24 
  1.2 Hardware level
  These pins are not used elsewhere (not part of other device tree or software)?
2. Is the chip select is configured correct to your SPI bus and chip select from device tree
https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/host/linux/host_driver/esp32/spi/esp_spi.c#L353-L354
  2.1 ```ls -lrt /dev/spi*``` should not list your device as that specific bus instance will/should be disabled in [spi_disabler](https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/host/linux/host_control/spidev_disabler.dts#L8)
3. **Can you please send use output of 'dmesg', ESP side log, and log how do you load the driver?**
4. Can you try to decrease SPI frequency at [esp_board.max_speed_hz](https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/host/linux/host_driver/esp32/spi/esp_spi.c#L352)
```
esp_board.max_speed_hz = /*spi_clk_mhz*/  1 *  NUMBER_1M;
```
5. Try other SPI modes at both places,
  5.1 ESP: [spi_slave_interface_config_t slvcfg](https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/esp/esp_driver/network_adapter/main/spi_slave_api.c#L466)
  5.2 Host: [esp_board.mode](https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/host/linux/host_driver/esp32/spi/esp_spi.c#L351)
6. If  spidev works (SPI pins are tested fine) but esp32_spi driver doesn't, Can you please put some logs inside to understand GPIO_HANDSHAKE and GPIO_DATA_READY are correctly set from ESP32.
  6.1 [spi_data_ready_interrupt_handler](https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/host/linux/host_driver/esp32/spi/esp_spi.c#L74)
  6.2 [spi_interrupt_handler](https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/host/linux/host_driver/esp32/spi/esp_spi.c#L83)

vuhailongkl97, 2022-07-08:
Hello @mantriyogesh  thanks for your reply.
1. I checked out the source code to tag `release/v0.4` and build & flash for both host and esp.
2. I tested /dev/spidev0.0 and spidev program.
3.  I write dtoverlay to disable spidev0.0 by my self. /dev/spidev0.0 wasn't existed after I setup to load my disabler spi.
at some pointer which you pointed out
1,2,3 -> I tried. I will try 4,5,6 in your suggestion.

mantriyogesh, 2022-07-08:
Thanks for heads up.
On point 2, I hope you have tested both, TX and RX in between ESP and host.

vuhailongkl97, 2022-07-08:
yes, I tested TX, RX the message was transceived between ESP and host properly. I run spidev example on host and https://github.com/espressif/esp-idf/blob/master/examples/peripherals/spi_slave/receiver/main/app_main.c on esp

vuhailongkl97, 2022-07-09:
@mantriyogesh thanks. the problem was resolved.
the problem is my data_ready in host which was reserved by other module, easily to fix it by change the pin to another one.

Now I'm can test some features on wifi properly. I would like to merge my some changes for  support beaglebone board to your repo after I finish the testing. Is it ok ?

mantriyogesh, 2022-07-09:
Hello @vuhailongkl97,

Glad to see that you could localize the issue and get the system working.

It will immensely help future users when they run into similar issue.
Some of the options I can think of, like GPIOs, SPI details (bus, CS used, mode, frequency) used, Device Tree, overlay source, any customizations in the code. It is up to you, how much you could share.

Further, In case you observe any code issues, Please do trigger Pull Request on repo, we will review and take your contribution into master with credits.

closed 2022-07-09 by vuhailongkl97

mantriyogesh, 2022-07-11:
Hello @vuhailongkl97 

Please help us to close the issue if the issue is resolved as per your expectations.

vuhailongkl97, 2022-07-11:
hello @mantriyogesh 
I closed this issue yesterday, I don't understand what you want ?

mantriyogesh, 2022-07-11:
My apologies. The browser didn't update cached status.
Thank you.