espressif/esp-hosted #132: network_adapter Build Error

lgl88911, 2022-07-16:
I got the following error when compiling network_adapter:
```
In file included from /mnt/e/westz/esp/esp-hosted/esp/esp_driver/network_adapter/main/app_main.c:47:
/mnt/e/westz/esp/esp-idf/components/driver/deprecated/driver/periph_ctrl.h:7:2: warning: #warning driver/periph_ctrl.h header is no longer used, and will be removed in future versions. [-Wcpp]
    7 | #warning driver/periph_ctrl.h header is no longer used, and will be removed in future versions.
      |  ^~~~~~~
/mnt/e/westz/esp/esp-hosted/esp/esp_driver/network_adapter/main/app_main.c: In function 'esp_update_ap_mac':
/mnt/e/westz/esp/esp-hosted/esp/esp_driver/network_adapter/main/app_main.c:196:47: error: 'MACSTR' undeclared (first use in this function)
  196 |                 snprintf(mac_str,BSSID_LENGTH,MACSTR,MAC2STR(ap_mac));
      |                                               ^~~~~~
/mnt/e/westz/esp/esp-hosted/esp/esp_driver/network_adapter/main/app_main.c:196:47: note: each undeclared identifier is reported only once for each function it appears in
/mnt/e/westz/esp/esp-hosted/esp/esp_driver/network_adapter/main/app_main.c:196:54: error: implicit declaration of function 'MAC2STR' [-Werror=implicit-function-declaration]
  196 |                 snprintf(mac_str,BSSID_LENGTH,MACSTR,MAC2STR(ap_mac));
      |                                                      ^~~~~~~
/mnt/e/westz/esp/esp-hosted/esp/esp_driver/network_adapter/main/app_main.c: In function 'app_main':
/mnt/e/westz/esp/esp-hosted/esp/esp_driver/network_adapter/main/app_main.c:697:15: error: implicit declaration of function 'esp_read_mac' [-Werror=implicit-function-declaration]
  697 |         ret = esp_read_mac(mac, ESP_MAC_BT);
      |               ^~~~~~~~~~~~
/mnt/e/westz/esp/esp-hosted/esp/esp_driver/network_adapter/main/app_main.c:697:33: error: 'ESP_MAC_BT' undeclared (first use in this function); did you mean 'ESP_MAX_IF'?
  697 |         ret = esp_read_mac(mac, ESP_MAC_BT);
      |                                 ^~~~~~~~~~
      |                                 ESP_MAX_IF
/mnt/e/westz/esp/esp-hosted/esp/esp_driver/network_adapter/main/app_main.c:752:9: error: implicit declaration of function 'tcpip_adapter_init' [-Werror=implicit-function-declaration]
  752 |         tcpip_adapter_init();
      |         ^~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
```
I used the latest idf version and the compilation method is as follows

```
git clone --recursive https://github.com/espressif/esp-idf.git --depth=1
cd esp-idf
./install.sh esp32c3

cd esp-hosted/esp/esp_driver/network_adapter/
idf.py fullclean
idf.py set-target esp32c3
idf.py -p /dev/ttyS14 build flash
```
Does compiling network_adapter require special configuration or a specific version of esp-idf?

lgl88911, 2022-07-16:
Update: Use esp-idf 4.4 build Pass

mantriyogesh, 2022-07-17:
Please use supported ESP-IDF release as per your ESP chipset being used from [documentation](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/Linux_based_readme.md#22-esp-idf-setup)

mantriyogesh, 2022-07-17:
Please feel free to connect if you face any more issues.

lgl88911, 2022-07-17:
@mantriyogesh Thanks, I find the information form your link
[image]

closed 2022-07-17 by lgl88911