espressif/esp-hosted #133: esp32-c3 spi communication checksum failed

TreeNewbeer, 2022-07-25:
I am using esp32-c3 spi(25M or 50M) connected with my host MCU, and found that the spi communication is not so stable and will raise checksum failed exception at some time, when I attach a logic analyser to the spi bus and not found the "checksum failed data" appeared. Is any idea for the further debugging?

[image]

TreeNewbeer, 2022-07-25:
BTW, my esp-hosted software version is release v0.4, and host driver is ported by myself which running on a Zephyr-RTOS MCU system

mantriyogesh, 2022-07-25:
Nice to hear, that you had ported the MCU solution over Zephyr RTOS.
We will check this issue and get back to you shortly.

TreeNewbeer, 2022-07-26:
Thanks for the quick reply
Today I found that esp-hosted has released v1.0.1 and also seen that there are some spi relatived bugfixs in the lastest release, I will also update my esp-hosted software version to the lastest release and do some test to figure out whether  this issue is appears again.

mantriyogesh, 2022-07-26:
Yes! The the new release is done, basically, there are two variants:
1. ESP-Hosted-NG - next generation: This is Linux only solution, where you can configure with help of cfg-80211.
2. ESP-Hosted: This is suggested solution for MCU. Yes! As you have noticed, There are multiple bug fixes gone in. Please do let us know if you have problems with this variant latest code.

TreeNewbeer, 2022-09-13:
I have upgraded my esp-hosted version to the latest MCU-Host branch(f209c91) and found that the problem is still exit (spi lost data packages). I also used iperf to test udp throughput and found the package lost rate is very high.
 (BTW, it was a long day when upgraded the code...)
[image]

Is it any idea to fix this problem?

mantriyogesh, 2022-09-13:
Sorry to hear, that you had to put efforts to upgrade the solution.
I hope you had not manually merged. Just by `git pull` and `git checkout ESP-Hosted_MCU_Host` should have done the trick, if the ESP-Hosted code was used through separate folder / git submodule.

Anyway, Can you please let us know some things:
1. What are exact server and client side iperf commands you use.
2. iperf version
3. SPI mode used
4. SPI clock used
5. Is the checksum enabled/disabled now? Do you still observe checksum related errors?
6. Can you verify if it is not issue of your network stack and OS context switching issues or ticking issues?
As I see pattern of iperf, that it is dropping in some timely fashion. If you are using MCU, you might have better control to check the task histogram, that might get some better idea.

Points 1->5 are to understand your system and we will try to mimic your setup to reproduce the issue.
Point 6 is some exercise you can try at your side which will help us to understand your system.

@TreeNewbeer, This might take a little slow path to resolve because of nature of issue & as many things to be tried out at both ends. But we will get to bottom of this issue.

TreeNewbeer, 2022-09-14:
Hi @mantriyogesh， thanks for the quick reply again!
For your question:
1. iperf test command:
    - server: iperf -s -u -i 1
    - client: zperf udp upload 172.22.18.177 5001 1 2048 5M
2. iperf version:
    - server: iperf version 2.0.5 (08 Jul 2010) pthreads
    - client: zephyr [zperf](https://docs.zephyrproject.org/latest/connectivity/networking/api/zperf.html) 
3. spi host configuration:
    - mode: master
    - data_bit_len: 32
    - clock: 25M/50M (problem will also occur in two clock configuration)
    - sample mode: SPI_CPOL1_CPHA0
4. spi slave (esp32c3) configuration:
    ```c
   spi_slave_interface_config_t slvcfg={
		.mode=SPI_MODE_2,
		.spics_io_num=GPIO_CS,
		.queue_size=SPI_QUEUE_SIZE,
		.flags=SPI_SLAVE_BIT_LSBFIRST,
		.post_setup_cb=spi_post_setup_cb,
		.post_trans_cb=spi_post_trans_cb
	};
    ```
5. I don't understand what's " checksum enabled/disabled" since I can't find some related configuration about here, is it has been implemented in recent code? I found this issue by adding some print code in checksum check code block.
6. I am using zephyrproject(v3.1) with native network stack, maybe it is a good idea to debug the network stack later. But I guess the problem is not because of the network stack though I could not 100% make sure since the UDP processing is not so complex and I really met the issue in spi checksum.

TreeNewbeer, 2022-09-14:
The reason why I manually merged is that I add some my own cmd/event code into esp-hosted in my own method, when I merging the latest code and found that some feature has been official supported and I need to remove some old code or refactor it in more elegant method such as the "Event report" feature, now just do some modify in the protobuf desp file and add it's handler. ：）

mantriyogesh, 2022-09-14:
Although exact environment is hard to reproduce, we will try to establish a similar environment to debug the problem.
If it is feasible for you, you can send a port thought PR. we can mimic the environment to recreate exact issue.

We can even merge that PR with credits to you, if we support that environment in future.

TreeNewbeer, 2022-09-14:
I have pushed my porting code and opened a PR https://github.com/espressif/esp-hosted/pull/160.
It seems that it is hard to reproduce this issue for you on STM32 platform, maybe I should get a STM32 devkit to run the MCU project and figure out why there is no problem on STM32...
I will also write a SPI  communication test case to exactly verify whether the issue is about SPI or others. Thanks for your help!

mantriyogesh, 2022-09-14:
Thank you @TreeNewbeer for your efforts and time for the PR.
We will try to dig in further how easily we could test based on your changes.

TreeNewbeer, 2022-09-15:
Hi @mantriyogesh , I found this issue is a little bit complex, I found there are some problem really about the network stack, and the checksum failed problem maybe related with hardware PCB layout. If you really could not find some big problem in my porting code, you can stop the further debugging and save your time, I am also working on it and will let you know my key progress if your are interest in it.

For the PR, if you have plan that porting to zephyrproject and let more MCU platform which support zephyr can run esp-hosted, I will help you to do some code clean/refactor in this PR and get it merged.

mantriyogesh, 2022-09-16:
Thank you @TreeNewbeer, For all your all the efforts including the debugging.
We will definitely be interested to educate ourselves about your finding. That also may be helpful for other people having similar issues.

Regarding the PR, Please allow me some time to respond, I will try to check how we can try to proceed ahead.