espressif/esp-hosted #134: GPIO pin confusion

Oquirella, 2022-07-27:
In rpi_init.sh the following pins are set to enable bluetooth:
    sudo raspi-gpio set 15 a0 pu
    sudo raspi-gpio set 14 a0 pu
    sudo raspi-gpio set 16 a3 pu
    sudo raspi-gpio set 17 a3 pu

But in SPI_setup.md ( ESP32-S3 setup)  these pins are not connected?

mantriyogesh, 2022-07-27:
Hello @Oquirella ,

Thank you for your query.
Actually, the above said pins are only required when you use transport combination, `SPI + UART`. Above pins are only required if Bluetooth traffic is sent through UART.

To detail further, the transport combinations for ESP32-S3,
- SPI only
  - Wi-Fi and Bluetooth, both running over SPI
  - Above Pins are not required as the Bluetooth data is handled through HCI interface from `ESP-Hosted kernel module`
- SPI + UART
  - Wi-Fi runs over SPI & Bluetooth runs over UART.
  - Above pins **are** required. Bluetooth traffic is directly offloaded through Linux kernel driver of HCI over UART.

Please refer `1.3 Setup` on [Linux_based_readme.md](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/Linux_based_host/Linux_based_readme.md) for additional information.

Hope this help to clear.

Oquirella, 2022-07-27:
What do in need to adjust in the devicetree? Because i am using a imx6ull evaluation kit with arduino headers. I don't know what to do with the pins

Oquirella, 2022-07-27:
I get the following errors when inserting the module:

[  648.322010] Failed to obtain SPI master handle
[  648.761621] Failed Init SPI device
modprobe: ERROR: could not insert 'esp32_spi': No such device

mantriyogesh, 2022-07-27:
Can you please help closing the issue?

mantriyogesh, 2022-10-03:
@Oquirella can we please close the issue?

closed 2022-10-03 by Oquirella