espressif/esp-hosted #137: porting done: iMX6UL Yocto

rforro, 2022-07-29:
Many of you are looking for Yocto recipes so here is mine, you can freely use it. It fetches current NG repository, builds esp32_sdio kernel module, configures reboot pin and sets up this module as automatically loadable at startup. 

My system configuration:
- Yocto dunfell
- Kernel 4.19.75

**esp-hosted_1.0.1.bb** 
```
SUMMARY = "ESP-Hosted solution provides a way to use ESP board as a communication processor i.e. host for Wi-Fi and Bluetooth/BLE connectivity"
HOMEPAGE = "https://github.com/espressif/esp-hosted/"
LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://../LICENSE;md5=7f43e699e0a26fae98c2938092f008d2"

SRC_URI = "gitsm://github.com/espressif/esp-hosted.git;branch=master"

SRCREV = "release/v${PV}"

S = "${WORKDIR}/git/host"

inherit module

EXTRA_OEMAKE_append_task-install = " -C ${STAGING_KERNEL_DIR} M=${S}"
EXTRA_OEMAKE += "KERNEL=${STAGING_KERNEL_DIR}"
EXTRA_OEMAKE += "target=sdio"

RPROVIDES_${PN} += "kernel-module-esp-hosted"
```

Don't forget to setup resetpin. Create modprobe configuration and place there definition of the GPIO pin where you connected ESP32_EN by adding following into `machine.conf` or `local.conf` 
```
KERNEL_MODULE_PROBECONF += "esp32_sdio"
module_conf_esp32_sdio = "options esp32_sdio resetpin=1"
```
This will create a modprobe configuration file under `/etc` with your resetpin. Now, instead of `insmod` you should call `modprobe esp32_sdio` only. But on embedded device is kinda impractical to load it manually, so setup it up as an auto-loadable module:
```
KERNEL_MODULE_AUTOLOAD += "esp32_sdio"
```

MangeshMalusare, 2022-07-29:
Hello @rforro,
Thank you for sharing.. We appreciate your contribution!

Thanks,
Mangesh

Oquirella, 2022-07-29:
@rforro Can you also share your device tree and setup?

mantriyogesh, 2022-08-03:
@rforro 
Can you please share your device tree for @Oquirella reference?

BTW @Oquirella ,

The device tree is very specific to the SoC. If the SoC differs, Device Tree would differ.
So might be good as reference.

One of the such device tree we used (stm32mp157f-dk2.dts.zip) in past for porting ST Linux:
[st_lin_sdio_working.patch.gz](https://github.com/espressif/esp-hosted/files/9248602/st_lin_sdio_working.patch.gz)
[st_lin_code_change_for_SPI__over_08a6de0__27aug_sr.patch.gz](https://github.com/espressif/esp-hosted/files/9248600/st_lin_code_change_for_SPI__over_08a6de0__27aug_sr.patch.gz)
[stm32mp157f-dk2.dts.zip](https://github.com/espressif/esp-hosted/files/9248604/stm32mp157f-dk2.dts.zip)

mantriyogesh, 2022-08-03:
Porting guide also can be helpful to port solution to your Linux.
https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/Linux_based_host/porting_guide.md
and 
https://github.com/espressif/esp-hosted/blob/master/docs/porting_guide.md

Oquirella, 2022-08-16:
I get the following messages in my console:

[   90.082798] Bus freq set to 24000000 start...
[   90.083191] Bus freq set to 24000000 done!
[   90.121672] 
[   90.121672] Bus freq set to 400000000 start...
[   90.141809] Bus freq set to 400000000 done!
[   99.602626] 
[   99.602626] Bus freq set to 24000000 start...
[   99.603008] Bus freq set to 24000000 done!

Any idea why this happends?

mantriyogesh, 2022-08-16:
This looks unrelated to ESP-Hosted.
Anyway, can you please attach complete compressed kern.log

Oquirella, 2022-08-16:
I rebooted and then the issue disappeared.

I get the following message in my logs:

ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]

But i don't get any devide when using ifconfig -a:

can0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:28 

can1      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:29 

eth0      Link encap:Ethernet  HWaddr 00:04:9F:06:F5:5B  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth1      Link encap:Ethernet  HWaddr 00:04:9F:06:F5:5A  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

mantriyogesh, 2022-08-16:
@Oquirella Can you please update the logs in the original issue of yours, https://github.com/espressif/esp-hosted/issues/135, where it may make more sense.

meera1310, 2022-09-23:
I am having problem when I include this recipe in my 
yocto dunfell
kernel 5.10

- **ERROR: esp-1.0-r0 do_compile: oe_runmake failed

ERROR: esp-1.0-r0 do_compile: Execution of '/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/temp/run.do_compile.530083' failed with exit code 1
ERROR: Logfile of failure stored in: /home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/temp/log.do_compile.530083**
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 16 KERNEL_SRC=/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source KERNEL=/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source target=spi KERNEL_PATH=/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source KERNEL_VERSION=5.10.83-cip1-yocto-standard CC=aarch64-poky-linux-gcc   -fuse-ld=bfd -fmacro-prefix-map=/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0=/usr/src/debug/esp/1.0-r0                      -fdebug-prefix-map=/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0=/usr/src/debug/esp/1.0-r0                      -fdebug-prefix-map=/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/recipe-sysroot-native=  -fdebug-prefix-map=/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source=/usr/src/kernel LD=aarch64-poky-linux-ld.bfd   AR=aarch64-poky-linux-ar  O=/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-build-artifacts KBUILD_EXTRA_SYMBOLS=
| make: *** No targets specified and no makefile found.  Stop.
| ERROR: oe_runmake failed

Oquirella, 2022-09-23:
Try my recipe :

DESCRIPTION = "Driver to use Espressif SoCs and modules as a communication co-processor"
HOMEPAGE = "https://github.com/f-secure-foundry/mxs-dcp"

LICENSE = "GPLv3"
LIC_FILES_CHKSUM = "file://${WORKDIR}/git/LICENSE;md5=7f43e699e0a26fae98c2938092f008d2"

SRC_URI = "gitsm://github.com/espressif/esp-hosted.git;protocol=https;branch=ESP-Hosted_MCU_Host"

SRCREV = "${AUTOREV}"

require esp-hosted.inc

S = "${WORKDIR}/git/host/linux/host_driver/esp32"

do_configure_prepend() {
    sed -i 's/#define SPI_DATA_READY_PIN.*/#define SPI_DATA_READY_PIN 69/g' ${S}/spi/esp_spi.h
    sed -i 's/#define HANDSHAKE_PIN.*/#define HANDSHAKE_PIN 70/g' ${S}/spi/esp_spi.h    
}

inherit module

EXTRA_OEMAKE_append_task-install = " -C ${STAGING_KERNEL_DIR} M=${S}"
EXTRA_OEMAKE += "KERNEL=${STAGING_KERNEL_DIR}"
EXTRA_OEMAKE += "target=spi"

KERNEL_MODULE_PROBECONF += "esp32_spi"
KERNEL_MODULE_AUTOLOAD += "esp32_spi"
module_conf_esp32_spi = "options esp32_spi resetpin=67"

mantriyogesh, 2022-09-23:
Hello @meera1310 
I am not understanding the exact issue you have faced.
I suppose it is something to do with building. May be, @rforro can help. 

@Oquirella 
Can you please add the log into your original issue, 135?

meera1310, 2022-09-23:
Hello Oquirella,
                         Thank you so much for your response. Actually, i couldn't find esp-hosted.inc file. After commenting that line I am getting errors listed below. I also changed

LICENSE = "Apache-2.0" because that license was throwing some error

ERROR: esp-1.0-r0 do_compile: oe_runmake failed
ERROR: esp-1.0-r0 do_compile: Execution of '/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/temp/run.do_compile.533483' failed with exit code 1
ERROR: Logfile of failure stored in: /home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/temp/log.do_compile.533483
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 16 KERNEL_SRC=/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source KERNEL=/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source target=spi KERNEL_PATH=/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source KERNEL_VERSION=5.10.83-cip1-yocto-standard CC=aarch64-poky-linux-gcc   -fuse-ld=bfd -fmacro-prefix-map=/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0=/usr/src/debug/esp/1.0-r0                      -fdebug-prefix-map=/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0=/usr/src/debug/esp/1.0-r0                      -fdebug-prefix-map=/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/recipe-sysroot-native=  -fdebug-prefix-map=/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source=/usr/src/kernel LD=aarch64-poky-linux-ld.bfd   AR=aarch64-poky-linux-ar  O=/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-build-artifacts KBUILD_EXTRA_SYMBOLS=
| rm -rf *.o sdio/*.o spi/*.o *.ko
| make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabihf- -C /home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source M=/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32 clean
| make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
| make[1]: Entering directory '/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source'
| make[2]: Entering directory '/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-build-artifacts'
| make[2]: Leaving directory '/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-build-artifacts'
| make[1]: Leaving directory '/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source'
| make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabihf- -C /home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source M=/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32 modules
| make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
| make[1]: Entering directory '/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source'
| make[2]: Entering directory '/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-build-artifacts'
|   CC [M]  /home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp_bt.o
| aarch64-poky-linux-gcc: error: unrecognized argument in option '-mabi=apcs-gnu'
| aarch64-poky-linux-gcc: note: valid arguments to '-mabi=' are: ilp32 lp64
| aarch64-poky-linux-gcc: error: unrecognized command line option '-mapcs'; did you mean '--specs'?
| aarch64-poky-linux-gcc: error: unrecognized command line option '-mno-sched-prolog'; did you mean '-Wno-sign-promo'?
| aarch64-poky-linux-gcc: error: unrecognized command line option '-msoft-float'
| make[3]: *** [/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source/scripts/Makefile.build:280: /home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp_bt.o] Error 1
| make[2]: *** [/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source/Makefile:1822: /home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32] Error 2
| make[2]: Leaving directory '/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-build-artifacts'
| make[1]: *** [Makefile:185: __sub-make] Error 2
| make[1]: Leaving directory '/home/glide/projects/renesas_5.10_kernel/build/tmp/work-shared/smarc-rzg2lc/kernel-source'
| make: *** [Makefile:53: all] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/temp/run.do_compile.533483' failed with exit code 1
ERROR: Task (/home/glide/projects/renesas_5.10_kernel/build/../meta-renesas/recipes-esp/esp/esp.bb:do_compile) failed with exit code '1'

Oquirella, 2022-09-23:
This is your real problem:

| aarch64-poky-linux-gcc: error: unrecognized argument in option '-mabi=apcs-gnu'
| aarch64-poky-linux-gcc: note: valid arguments to '-mabi=' are: ilp32 lp64
| aarch64-poky-linux-gcc: error: unrecognized command line option '-mapcs'; did you mean '--specs'?
| aarch64-poky-linux-gcc: error: unrecognized command line option '-mno-sched-prolog'; did you mean '-Wno-sign-promo'?
| aarch64-poky-linux-gcc: error: unrecognized command line option '-msoft-float'

meera1310, 2022-09-23:
yes

meera1310, 2022-09-23:
As I changed arch and cross compile , now I am facing these errors

ERROR: modpost: "hci_register_dev" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!
| ERROR: modpost: "hci_recv_frame" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!
| ERROR: modpost: "hci_unregister_dev" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!
| ERROR: modpost: "bt_err" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!
| ERROR: modpost: "hci_alloc_dev" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!
| ERROR: modpost: "hci_free_dev" [/home/glide/projects/renesas_5.10_kernel/build/tmp/work/smarc_rzg2lc-poky-linux/esp/1.0-r0/git/host/linux/host_driver/esp32/esp32_spi.ko] undefined!

mantriyogesh, 2022-09-23:
@meera1310 

Next onwards, please raise a separate request to track a separate issue.
You can always tag any other issue inside that new issue.

This will really help us & others in long run.
Thank you for your understanding.

lakshitave, 2022-09-26:
Hello,
In my recipe I want to configure that esp should take spidev1.0 driver instead of spi. How could I include this in my recipe or there is any other method for this.

mantriyogesh, 2022-09-26:
Yocto recipe related question, if someone can help, Please check, https://github.com/espressif/esp-hosted/issues/162

mantriyogesh, 2022-09-26:
@meera1310 

I think those errors are arising because you are not able to load the hci related function definitions.
I am not sure how can you include bluetooth module in your yocto, but in general, if you do `sudo modprobe bluetooth`, these errors should go away in non-yocto linux.

Further, there is an open issue, https://github.com/espressif/esp-hosted/issues/154, which might be related to you.

lakshitave, 2022-09-27:
Hello,
I am getting error in my logs. I am sharing the logs which I am getting:
root@smarc-rzg2lc:~# modprobe esp32_spi
[   61.702523] rpc-if-spi rpc-if-spi: chipselect 0 already in use
[   61.714222] Failed to add new SPI device
[   62.134610] ------------[ cut here ]------------
[   62.139479] kobject: '(null)' (00000000af576277): is not initialized, yet kobject_put() is being called.
[   62.154519] WARNING: CPU: 1 PID: 235 at lib/kobject.c:752 kobject_put+0xd8/0x118
[   62.161996] Modules linked in: esp32_spi(O+) vspm_if(O) vspm(O) mmngrbuf(O) mmngr(O)
[   62.169949] CPU: 1 PID: 235 Comm: modprobe Tainted: G        W  O      5.10.83-cip1-yocto-standard #1
[   62.179221] Hardware name: Renesas SMARC EVK based on r9a07g044c2 (DT)
[   62.185811] pstate: 40400005 (nZcv daif +PAN -UAO -TCO BTYPE=--)
[   62.191880] pc : kobject_put+0xd8/0x118
[   62.195772] lr : kobject_put+0xd8/0x118
[   62.199649] sp : ffff8000197dba70
[   62.203005] x29: ffff8000197dba70 x28: 0000000000000100 
[   62.208415] x27: 0000000000000002 x26: ffff8000197dbda0 
[   62.213815] x25: ffff00000f4eb680 x24: 0000000000000024 
[   62.219213] x23: 0000000000000000 x22: ffff800008a37000 
[   62.224611] x21: ffff00000aca1b00 x20: ffff800008a6c000 
[   62.230009] x19: ffff800008a6c520 x18: 0000000000000000 
[   62.235406] x17: 0000000000000000 x16: 0000000000000000 
[   62.240801] x15: 00002bf9500b0ce6 x14: 000000000000014a 
[   62.246196] x13: 000000000000014a x12: 0000000000000000 
[   62.251591] x11: 0000000000000005 x10: 0000000000000930 
[   62.256985] x9 : ffff8000197db7a0 x8 : ffff00000aca2490 
[   62.262381] x7 : 0000000000000000 x6 : ffff00003fdfc8a0 
[   62.267776] x5 : ffff00003fdfc8a0 x4 : 0000000000000000 
[   62.273168] x3 : ffff00003fe037b0 x2 : ffff00003fdfc8a0 
[   62.278563] x1 : 93f32aa990440900 x0 : 0000000000000000 
[   62.283957] Call trace:
[   62.286472]  kobject_put+0xd8/0x118
[   62.290027]  cdev_del+0x2c/0x40
[   62.293281]  esp_serial_cleanup+0x20/0x60 [esp32_spi]
[   62.298432]  spi_exit+0xa0/0x130 [esp32_spi]
[   62.302797]  esp_init_interface_layer+0x16c/0x198 [esp32_spi]
[   62.308638]  esp_init+0x150/0x1000 [esp32_spi]
[   62.313148]  do_one_initcall+0x74/0x1d0
[   62.317052]  do_init_module+0x54/0x218
[   62.320864]  load_module+0x1ebc/0x2408
[   62.324677]  __do_sys_finit_module+0xbc/0xf8
[   62.329010]  __arm64_sys_finit_module+0x20/0x30
[   62.333611]  el0_svc_common.constprop.4+0x8c/0x180
[   62.338463]  do_el0_svc+0x24/0x90
[   62.341845]  el0_svc+0x14/0x20
[   62.344966]  el0_sync_handler+0x90/0xb8
[   62.348857]  el0_sync+0x160/0x180
[   62.352222] ---[ end trace 9e0623897f56984e ]---
[   62.366380] ------------[ cut here ]------------
[   62.373391] refcount_t: underflow; use-after-free.
[   62.390505] WARNING: CPU: 0 PID: 235 at lib/refcount.c:28 refcount_warn_saturate+0xf8/0x148
[   62.398962] Modules linked in: esp32_spi(O+) vspm_if(O) vspm(O) mmngrbuf(O) mmngr(O)
[   62.406926] CPU: 0 PID: 235 Comm: modprobe Tainted: G        W  O      5.10.83-cip1-yocto-standard #1
[   62.416211] Hardware name: Renesas SMARC EVK based on r9a07g044c2 (DT)
[   62.422807] pstate: 40400005 (nZcv daif +PAN -UAO -TCO BTYPE=--)
[   62.428891] pc : refcount_warn_saturate+0xf8/0x148
[   62.433753] lr : refcount_warn_saturate+0xf8/0x148
[   62.438592] sp : ffff8000197dba60
[   62.441956] x29: ffff8000197dba60 x28: 0000000000000100 
[   62.447359] x27: 0000000000000002 x26: ffff8000197dbda0 
[   62.452760] x25: ffff00000f4eb680 x24: 0000000000000024 
[   62.458157] x23: 0000000000000000 x22: ffff800008a37000 
[   62.463554] x21: ffff00000aca1b00 x20: ffff800008a6c000 
[   62.468952] x19: ffff800008a6c520 x18: 0000000000000000 
[   62.474348] x17: 0000000000000000 x16: 0000000000000000 
[   62.479744] x15: 00008faf6495b3b8 x14: 000000000000006d 
[   62.485140] x13: 000000000000006d x12: 0000000000000000 
[   62.490535] x11: 000000000000000b x10: 0000000000000930 
[   62.495932] x9 : ffff8000197db790 x8 : ffff00000aca2490 
[   62.501329] x7 : 0000000000000000 x6 : ffff00003fde58a0 
[... 104 lines cut ...]
[   63.286536]  esp_init+0x150/0x1000 [esp32_spi]
[   63.291051]  do_one_initcall+0x74/0x1d0
[   63.294958]  do_init_module+0x54/0x218
[   63.298771]  load_module+0x1ebc/0x2408
[   63.302584]  __do_sys_finit_module+0xbc/0xf8
[   63.306920]  __arm64_sys_finit_module+0x20/0x30
[   63.311519]  el0_svc_common.constprop.4+0x8c/0x180
[   63.316374]  do_el0_svc+0x24/0x90
[   63.319756]  el0_svc+0x14/0x20
[   63.322876]  el0_sync_handler+0x90/0xb8
[   63.326769]  el0_sync+0x160/0x180
[   63.330164] Code: a90153f3 aa0003f3 aa0103f4 f9400000 (f9400400) 
[   63.336325] ---[ end trace 9e0623897f569851 ]---

Broadcast message from systemd-journald@smarc-rzg2lc (Sun 2020-09-20 10:44:57 UTC):

kernel[182]: [   63.129817] Internal error: Oops: 96000004 [#1] PREEMPT SMP

Broadcast message from systemd-journald@smarc-rzg2lc (Sun 2020-09-20 10:44:57 UTC):

kernel[182]: [   63.330164] Code: a90153f3 aa0003f3 aa0103f4 f9400000 (f9400400)

Segmentation fault

Please suggest me a solution for this.

meera1310, 2022-09-27:
Will you please help me for adding esp drivers [](https://github.com/espressif/esp-hosted/tree/ESP-Hosted_MCU_Host/esp/esp_driver/network_adapter)in yocto image?
yocto version - **3.1.14 (dunfell)**
kernel - **5.10.83-cip1-yocto-standard**

mantriyogesh, 2022-10-03:
Hello @meera1310 ,

I am not exactly familiar with yocto. But you can take references from this issue and several others where they are actually using ESP-Hosted using yocto recipe.
ESP-Hosted solution actually is agnostic of yocto and should work in both, yocto or non-yocto cases.