espressif/esp-hosted #138: HCI scan errors when Using esp32C3 hosted mode

anoop18pant, 2022-07-29:
hi,

We are using ESP32C3 hosted mode we are observing one issue when we have a lot many BLE devices then we are observing the HCI device errors time outs etc....

The random errors are occurring.

Any help pointer will be appreciated.

mantriyogesh, 2022-07-29:
Yes, we will surely look into the issue.
Can you please help to close other issues where you have received the solution?

Please send us the logs,
1. host
- dmesg
- btmon log while test running
- bluetoothctl log
2. esp
- firmware log

TreeNewbeer, 2022-08-01:
I also met a related issue and I found that it always occurs in ble adv report. I think this problem is about the adv report flow control, the esp-idf has some configuration macros are related to here, but I have no time to varify that whether changes those configurations will fix it or not.

TreeNewbeer, 2022-08-01:
[image]

Those configuration macros I found are here. Hopefully it will make some sense.

mantriyogesh, 2022-08-16:
Hello @anoop18pant ,
By any chance are you using hcitool? hcitool is deprecated for latest blueZ releases.
You can use 'bluetoothctl' instead.

Could you please provide ESP, Host Kern log, bluetoothctl and btmon log to understand the issue you faced?

mantriyogesh, 2022-08-19:
Have you got it working @anoop18pant ?

mantriyogesh, 2022-09-06:
Any updates @anoop18pant ?

anoop18pant, 2022-09-06:
After changing the clock from 1 mhz to 20 mhz seems the issue is not occurring.

Though not sure about it

mantriyogesh, 2022-09-06:
Great, thanks.

One possibility that, the data is not getting pushed fast within expected timers expected by BlueZ, because of slower frequency.
Can you please help closing the issue?

anoop18pant, 2022-09-06:
After clock speed change not observing the change.

closed 2022-09-06 by anoop18pant