espressif/esp-hosted #139: Ehternet MII support

CoderScribe, 2022-07-29:
Hello, has consideration also been made to support the Ethernet MAC? Then you could offer three interfaces WiFi+BT/BLE+Ethernet in a single IC.

mantriyogesh, 2022-08-16:
Hello @CoderScribe,

Thank you for your ideas.
We will definitely check the feasibility for this.

mike-2020, 2024-03-01:
Has this been implemented? It could generate some interesting use cases if such function is available.

mantriyogesh, 2024-03-02:
Actually, We right now do not have Ethernet modules.
But anyway the net-if interface is just the same as that of Wi-Fi, it should not much differ in terms of the porting.

1. **Rx path** : `SPI/SDIO bus <-- Wi-Fi`  == replace by ==>  `SPI/SDIO bus <-- eth`

 https://github.com/espressif/esp-hosted/blob/415226f6a4fe9be5fb0f80231873a79428cc875f/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c#L240-L266 ,
A. `wlan_sta_rx_callback()` to be replaced with say `eth_rx_callback()`
B. `esp_wifi_internal_free_rx_buffer()` to be replaced with `eth_l2_free()` (or better, `eth_netif->driver_free_rx_buffer()`

**Tx path**: `SPI/SDIO bus --> Wi-Fi`  == replace by ==>  `SPI/SDIO bus --> eth`
C. `esp_wifi_internal_tx` to `esp_eth_transmit()`

mike-2020, 2024-03-03:
Thanks for the response. Do you mean it can be as simple as replacing the functions mentioned above?
What about I need both WIFI network and Ethernet network avaiable at the same time? Can I present two network card in the host OS: one for ESP32 WIFI and one for ESP32 Ethernet? (but still use one single SDIO interface for communication between ESP32 and the host)

mantriyogesh, 2024-03-04:
Yes! this is definitely do-able. I will have to check 'max' esp-netif instances limit (which by the way is not this small !). What you need is just one more extra esp-netif instance.

In Espressif another team, people use even multiple Ethernet ports to use the separate networks or even create bridge in two netifs.

mike-2020, 2024-03-10:
Considering this Ethernet integration intention (especially, present network cards for WIFI and Ethernet respectively), which solution is better match? FG or NG?

mantriyogesh, 2024-03-10:
Integration should be much similar.
FG vs NG depends upon how do you intend to use wifi at Linux host.
If you wish complete control, FG is better as it provides full control from application with simple API calls.

NG is preferred when wifi you want to get controlled by standard wpa_supplicant.

For new ethernet control, easier to integrate first using FG in my opinion, as your application would have complete control of new eth.