espressif/esp-hosted #140: Unable to read data before 30 seconds timeout

take-42, 2022-08-03:
Hello.
Trying to realize AP mode and STA mode of WiFi by connecting RaspberryPi 3 Model B and ESP-WROOM-32 with SPI.

I followed the steps in the ESP-Hosted Comprehensive Guide and wired as described in [1.1.1 ESP32 setup for Wi-Fi and BT/BLE connectivity Setup over SPI](https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/Linux_based_host/SPI_setup.md#111-esp32-setup).
For the ESP32 firmware, I download the pre-built v4.0 and wrote 4 .bin files in ESP-IDF on Windows.

After running
```
$ cd host/linux/host_control/
$ . /rpi_init.sh spi
```
and then
```
$ cd host/linux/host_control/python_support/
$ sudo python3 test.py get_connected_ap_info
```

The following message is displayed, and it does not seem to be responding.

```
esp_hosted_driver_read, Unable to read data before 30 seconds timeout
transport_pserial_data_handler:154 Failed to read RX data
wifi_set_mode:369 Failed to process rx data
Error in setting mode
esp_hosted_driver_read, Unable to read data before 30 seconds timeout
transport_pserial_data_handler:154 Failed to read RX data
wifi_get_mode:294 Failed to process rx data
...
```

What could be the cause of these occurrences?

Or, when I tried ESP-Hoted-NG, I succeeded in connecting in STA mode, so I hope this can use both AP and STA mode, is there a schedule set?

take-42, 2022-08-03:
Here is ESP console and dmesg log,

<details>
<summary>ESP_log</summary>

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:12
load:0x3fff0028,len:7432
load:0x40078000,len:15100
ho 0 tail 12 room 4
load:0x40080400,len:3904
entry 0x400806b8
I (80) boot: Chip Revision: 3
I (81) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (44) boot: ESP-IDF v4.0.3-215-gbe7df8bce9-dirty 2nd stage bootloader
I (44) boot: compile time 17:19:23
I (45) boot: Enabling RNG early entropy source...
I (50) boot: SPI Speed      : 40MHz
I (54) boot: SPI Mode       : DIO
I (58) boot: SPI Flash Size : 4MB
I (63) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (81) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (88) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (96) boot:  3 factory          factory app      00 00 00010000 00100000
I (103) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (111) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (118) boot: End of partition table
I (123) boot: Defaulting to factory image
I (127) boot_comm: chip revision: 3, min. application chip revision: 0
I (135) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x202a8 (131752) map
I (191) esp_image: segment 1: paddr=0x000302d0 vaddr=0x3ffbdb60 size=0x03d40 ( 15680) load
I (197) esp_image: segment 2: paddr=0x00034018 vaddr=0x40080000 size=0x00400 (  1024) load
I (198) esp_image: segment 3: paddr=0x00034420 vaddr=0x40080400 size=0x0bbf0 ( 48112) load
I (227) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x8eab8 (584376) map
I (436) esp_image: segment 5: paddr=0x000cead8 vaddr=0x4008bff0 size=0x123b0 ( 74672) load
I (486) boot: Loaded app from partition at offset 0x10000
I (486) boot: Disabling RNG early entropy source...
I (486) cpu_start: Pro cpu up.
I (490) cpu_start: Application information:
I (495) cpu_start: Project name:     network_adapter
I (500) cpu_start: App version:      release0.4-60-g22180f7
I (506) cpu_start: Compile time:     Jan 24 2022 17:19:16
I (513) cpu_start: ELF file SHA256:  0f66794e8494e214...
I (519) cpu_start: ESP-IDF:          v4.0.3-215-gbe7df8bce9-dirty
I (525) cpu_start: Starting app cpu, entry point is 0x400812fc
I (0) cpu_start: App cpu up.
I (536) heap_init: Initializing. RAM available for dynamic allocation:
I (543) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (549) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (555) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (561) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (567) heap_init: At 3FFC80E0 len 00017F20 (95 KiB): DRAM
I (573) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (579) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (586) heap_init: At 4009E3A0 len 00001C60 (7 KiB): IRAM
I (592) cpu_start: Pro cpu start user code
I (610) spi_flash: detected chip: generic
I (611) spi_flash: flash io: dio
I (611) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (623) NETWORK_ADAPTER: *********************************************************************
[... 386 lines cut ...]
[    6.919027] random: crng init done
[    6.919050] random: 7 urandom warning(s) missed due to ratelimiting
[    7.100479] 8021q: 802.1Q VLAN Support v1.8
[    7.304352] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    7.356816] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    7.559179] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    7.648545] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    7.648736] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    9.211279] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[    9.213428] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   11.705065] fuse init (API version 7.27)
[   41.918048] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[   42.205643] Bluetooth: Core ver 2.22
[   42.205717] NET: Registered protocol family 31
[   42.205720] Bluetooth: HCI device and connection manager initialized
[   42.205735] Bluetooth: HCI socket layer initialized
[   42.205743] Bluetooth: L2CAP socket layer initialized
[   42.205769] Bluetooth: SCO socket layer initialized
[   65.564479] esp32_spi: loading out-of-tree module taints kernel.
[   65.565785] esp_reset, ESP32: Resetpin of Host is 6
[   65.565915] esp_reset, ESP32: Triggering ESP reset.
[   65.568730] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
pi@raspberrypi:~/esp/esp-hosted-v0.4/host/linux/host_control/python_support $ 
```
</details>

mantriyogesh, 2022-08-03:
Hello @take-42 ,

Possible duplicate of https://github.com/espressif/esp-hosted/issues/122?

take-42, 2022-08-03:
Hey @mantriyogesh 
Thanks for your reply.

I thought almost the same, but when I run `ifconfig -a`, `ethap0` and `ethsta0` are created.
Also `/dev/esps0` is present.

mantriyogesh, 2022-08-03:
I suggest you to use one of:
1. For ESP-Hosed-**N**G: Flash ESP from https://github.com/espressif/esp-hosted/releases/tag/release%2Fv1.0.1 and use host code from: https://github.com/espressif/esp-hosted/tree/master/host
2. For ESP-Hosted-**F**G: Flash ESP from code at https://github.com/espressif/esp-hosted/tree/ESP-Hosted_MCU_Host/esp/esp_driver/network_adapter using correct IDF version from documentation matching your ESP chipset. and use host code from https://github.com/espressif/esp-hosted/tree/ESP-Hosted_MCU_Host/host/linux

Basically, ESP & host be used from one branch. For ESP-Hosted-NG, use master. For  ESP-Hosted-FG, use branch, `ESP-Hosted_MCU_Host`

Can you please just confirm if this is done?

mantriyogesh, 2022-08-03:
Please refrain from changing sdkconfig manually.
Please follow `idf.py menuconfig` as said in documentation.

Can you please give ESP log on `ESP-Hosted_MCU_Host`?
It is differs from your original shared log.
Please check reference log:
[esp32_spi_ESP-Hosted_MCU_Host.log](https://github.com/espressif/esp-hosted/files/9250751/esp32_spi_ESP-Hosted_MCU_Host.log)

Note `slave_ctrl: event ESPInit` in the end.

take-42, 2022-08-04:
Is it also different from the console log when flashing with idf.py below?
Sorry, I looked in the `ESP-Hosted_MCU_Host` after flashing but could not find the log file.

<details><summary>log</summary>

```
PS C:\Users\xxxx\Desktop\esp-hosted-v0.4\esp\esp_driver\network_adapter> idf.py build flash monitor
Executing action: all (aliases: build)
Running ninja in directory c:\users\xxxx\desktop\esp-hosted-v0.4\esp\esp_driver\network_adapter\build
Executing "ninja all"...
[1/4] cmd.exe /C "cd /D C:\Users\xxxx\Desktop\ESP-HOSTED-...esp/esp_driver/network_adapter/build/network_adapter.bin"network_adapter.bin binary size 0xd2a90 bytes. Smallest app partition is 0x100000 bytes. 0x2d570 bytes (18%) free.
[2/4] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\xxxx\Desktop\esp-hosted-v0.4\esp\esp_driver\network_adapter\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/xxxx/Desktop/esp-hosted-v0.4/esp/esp_driver/network_adapter/build/bootloader/bootloader.bin"
Bootloader binary size 0x62c0 bytes. 0xd40 bytes (12%) free.
Executing action: flash
Serial port COM7
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Running ninja in directory c:\users\xxxx\desktop\esp-hosted-v0.4\esp\esp_driver\network_adapter\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\Users\xxxx\Desktop\ESP-HOSTED-...esp/esp_driver/network_adapter/build/network_adapter.bin"network_adapter.bin binary size 0xd2a90 bytes. Smallest app partition is 0x100000 bytes. 0x2d570 bytes (18%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\xxxx\Desktop\esp-hosted-v0.4\esp\esp_driver\network_adapter\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/xxxx/Desktop/esp-hosted-v0.4/esp/esp_driver/network_adapter/build/bootloader/bootloader.bin"
Bootloader binary size 0x62c0 bytes. 0xd40 bytes (12%) free.
[2/3] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v...p-idf-v4.4.1/components/esptool_py/run_serial_tool.cmake"esptool.py esp32 -p COM7 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 bootloader/bootloader.bin 0x10000 network_adapter.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin
esptool.py v3.3-dev
Serial port COM7
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 8c:4b:14:15:a4:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x000e2fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Compressed 25280 bytes to 15816...
Writing at 0x00001000... (100 %)
Wrote 25280 bytes (15816 compressed) at 0x00001000 in 0.8 seconds (effective 263.9 kbit/s)...
Hash of data verified.
Compressed 862864 bytes to 537813...
Writing at 0x00010000... (3 %)
Writing at 0x0001d5e3... (6 %)
Writing at 0x0002858c... (9 %)
Writing at 0x000327f6... (12 %)
Writing at 0x00037fb9... (15 %)
Writing at 0x0003de67... (18 %)
Writing at 0x00043d2b... (21 %)
Writing at 0x00049f6b... (24 %)
Writing at 0x0004fd0c... (27 %)
Writing at 0x00055c6e... (30 %)
Writing at 0x0005b80a... (33 %)
Writing at 0x0006123d... (36 %)
Writing at 0x00066ad4... (39 %)
Writing at 0x0006c152... (42 %)
Writing at 0x000718d6... (45 %)
Writing at 0x00077272... (48 %)
Writing at 0x0007d90a... (51 %)
Writing at 0x0008349e... (54 %)
Writing at 0x0008862a... (57 %)
Writing at 0x0008d9b3... (60 %)
Writing at 0x00092c89... (63 %)
[... 98 lines cut ...]
I (590) spi_flash: flash io: dio
I (595) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (609) NETWORK_ADAPTER: *********************************************************************
I (615) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.4
I (623) NETWORK_ADAPTER:                 Transport used :: SPI
I (631) NETWORK_ADAPTER: *********************************************************************
I (637) NETWORK_ADAPTER: Supported features are:
I (643) NETWORK_ADAPTER: - WLAN over SPI
I (645) ESP_BT: - BT/BLE
I (649) ESP_BT:    - HCI Over SPI
I (651) ESP_BT:    - BT/BLE dual mode
I (661) BTDM_INIT: BT controller compile version [5688ed5]
I (661) system_api: Base MAC address is not set
I (665) system_api: read default base MAC address from EFUSE
I (671) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (993) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 8c-4b-14-15-a4-22

I (993) SPI_DRIVER: Using SPI interface
I (993) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1001) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (2015) NETWORK_ADAPTER: Initial set up done
```

</details>

take-42, 2022-08-04:
First of all, the 30 second timeout no longer occurs after running `idf.py build flash monitor`.
I continue to be unable to run the tests specified in `test.py` as options (some unspecified tests being run?), but I will try some more. I will try a few more.

Thank you very much!

closed 2022-08-04 by take-42