espressif/esp-hosted #141: Matching fimrware for ESP-Hosted-NG Release 1.0.1

miwee, 2022-08-04:
The release files for ESP-Hosted-NG Release 1.0.1 have only Linux host driver. There is no ESP firmware files. Can we use ESP firmware binaries from https://github.com/espressif/esp-hosted/releases/tag/release%2Fv0.4

MangeshMalusare, 2022-08-04:
That firmware version is not compatible with NG release.

Please use binaries provided with NG release:
https://github.com/espressif/esp-hosted/releases/download/release%2Fv1.0.1/ESP-Hosted-NG-release_1.0.1.tgz

ajita02, 2022-08-04:
@miwee,
The binaries for ESP32 you can get from [ESP-Hosted-NG-release_1.0.1.tgz](https://github.com/espressif/esp-hosted/releases/download/release%2Fv1.0.1/ESP-Hosted-NG-release_1.0.1.tgz). Its present at release [ESP-Hosted-NG-release page](https://github.com/espressif/esp-hosted/releases/tag/release%2Fv1.0.1).

Thanks,
Ajita

mantriyogesh, 2022-08-04:
Short answer:
For `ESP-Hosted-NG`, Use [ESP-Hosted-NG-release_1.0.1.tgz](https://github.com/espressif/esp-hosted/releases/download/release%2Fv1.0.1/ESP-Hosted-NG-release_1.0.1.tgz)

Long Answer:
- For `ESP-Hosted` _i.e._ `ESP-Hosted-FG`
  - Flashing release 0.4 release with ready-made binaries
    - Use https://github.com/espressif/esp-hosted/releases/tag/release%2Fv0.4 binaries to flash ESP32.
    - Make sure you use https://github.com/espressif/esp-hosted/commit/74e707b604280ecb2eafa60ba5ed6993ce57b457 for Linux kernel building
    - Flashing ESP32 on latest code manually (includes recent bug-fixes)
      - Switch to [ESP-Hosted_MCU_Host](https://github.com/espressif/esp-hosted/tree/ESP-Hosted_MCU_Host) and follow https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/README.md
      - Use same code commit for flashing ESP32 and for host kernel building
-  For `ESP-Hosted-NG` 
   - Follow https://github.com/espressif/esp-hosted/blob/master/docs/README.md
   - This will eventually land in section `3.1 Setup` → [Hardware and Software Setup](https://github.com/espressif/esp-hosted/blob/master/docs/setup.md) 
   - Then [Select Hosted config](https://github.com/espressif/esp-hosted/blob/master/docs/setup.md#2-select-hosted-config). Unfold expected config. For example, `SPI Only`
   - In `Software setup` you can find steps how to flash `ESP-Hosted-NG-release_1.x.y.tgz`

miwee, 2022-08-04:
Thanks for the detailed info. At one place I read that Firmware source code is not released due to some technical issue. Our requirement is we are running older ESP-hosted v0.4 in SPI Mode 1 on both sides, as we discovered that works best in our case. 

Had the source for Firmware been available, we could have done this modification. Hence I request to please release SPI only Firmware binaries for SPI Mode 1 for ESP32-C3.

mantriyogesh, 2022-08-04:
ESP32-C3 and other chipsets which are already on ESP-Hosted-FG are surely going to come shortly.
By the time you can use `ESP-Hosted-FG`, if that works for you.

closed 2022-08-04 by miwee