espressif/esp-hosted #142: Esp Hosted INIT EVENT not received from MCU

Whitehouse112, 2022-08-09:
Hi everyone,
I'm trying to implement ESP-hosted for MCU solution but I'm stuck: i'm working with an esp32-wroom-32e connected through SPI to a MCU (stm32u575) but i'm unable to get the INIT_EVENT after esp32 startup. I receive only `0x0 0x0 0x0 0x0...`. I'm working with esp-hosted (branch `ESP-Hosted_MCU_Host`, commit: f209c91) on MCU, while `esp_hosted_firmware_esp32_spi_v04.bin`v has been flashed with esptool.py over ESP32. I'm sure SPI and all connections are correct since i can read correctly any values send from MCU to ESP thanks to logic analyzer (in the images below I send `0xaa` just to test the communication: the same result is obtained sending an empty buffer).

Someone has faced similar issue?

Thanks

[image]

[image]

mantriyogesh, 2022-08-11:
Hello @Whitehouse112 

Thank you for your issue.
Sorry for delay in response.

Is it possible to flash the ESP binaries on the same commit you have used for MCU host?

mantriyogesh, 2022-08-11:
Also can you please let me know which IOC file you had used to create STM workspace?

Whitehouse112, 2022-08-12:
You're right, I flashed the wrong version. With firmware v0.3 everything is fine!
Thanks

> Hello @Whitehouse112
> 
> Thank you for your issue. Sorry for delay in response.
> 
> Is it possible to flash the ESP binaries on the same commit you have used for MCU host?

closed 2022-08-12 by Whitehouse112