espressif/esp-hosted #143: esp32-c3-mini-1u-n4 device not responding for AT commands

Vinayaka-bs, 2022-08-11:
hi,

i'm using esp32-c3-mini-1u-n4 device. 
after the reset i can able to receive the below log on terminal:

ESP-ROM:esp32c3-api1-20210207
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_ëÃSP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x18c8
load:0x403ce000,len:0x8dc
load:0x403d0000,len:0x2d90
entry 0x403ce000
I (31) boot: ESP-IDF qa-test-v4.3.3-20220423 2nd stage bootloader
I (31) boot: compile time 09:06:43
I (32) boot: chip revision: 3
I (35) boot.esp32c3: SPI Speed      : 40MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (44) boot.esp32c3: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 nvs              WiFi data        01 02 00010000 0000e000
I (88) boot:  3 at_customize     unknown          40 00 0001e000 00042000
I (95) boot:  4 ota_0            OTA app          00 10 00060000 001d0000
I (102) boot:  5 ota_1            OTA app          00 11 00230000 001d0000
I (110) boot: End of partition table
I (114) esp_image: segment 0: paddr=00060020 vaddr=3c140020 size=2a5e0h (173536) map
I (161) esp_image: segment 1: paddr=0008a608 vaddr=3fc91200 size=03c6ch ( 15468) load
I (165) esp_image: segment 2: paddr=0008e27c vaddr=40380000 size=01d9ch (  7580) load
I (168) esp_image: segment 3: paddr=00090020 vaddr=42000020 size=133aech (1260268) map
I (451) esp_image: segment 4: paddr=001c3b14 vaddr=40381d9c size=0f384h ( 62340) load
I (466) esp_image: segment 5: paddr=001d2ea0 vaddr=50000000 size=00014h (    20) load
I (467) esp_image: segment 6: paddr=001d2ebc vaddr=50000018 size=00010h (    16) load
I (478) boot: Loaded app from partition at offset 0x60000
I (478) boot: Disabling RNG early entropy source...
module_name:MINI-1

max tx power=78,ret=0

2.4.1

After this if I send AT (along with CR+LF) through terminal. it was not responding. also not responding for any other command.
please help me to fix this issue. 

thank you. 

regards,
vinay

mantriyogesh, 2022-08-11:
Hello @Vinayaka-bs 

We do not support AT commands. Fot AT commands, you can refer [ESP-AT](https://github.com/espressif/esp-at)
Alternatively, if you feel this is ESP-IDF issue, please raise it at [ESP-IDF](https://github.com/espressif/esp-idf)

Vinayaka-bs, 2022-08-11:
Thank you for your response.

 

Regards,

vinay

mantriyogesh, 2022-08-12:
Could you please close this issue?

closed 2022-08-12 by Vinayaka-bs