espressif/esp-hosted #144: Source code for ESP-Hosted-NG Firmware

amitv87, 2022-08-15:
Could not find source code for the firmwares published in the latest release, is it supposed to be a binary only distribution?

mantriyogesh, 2022-08-16:
Hello @amitv87 

Thank you for your query.
As if now, The ESP side source is not provided with last release of ESP-Hosted-NG. Source is available for host code.

There were many libraries and manual steps involved for end user, which could have lead to unexpected user experience, this is the reason, we provided in binary format.

Anyway, we aim for providing open source solution, which we are working on right now. Most probably next release would be with source included.

amitv87, 2022-12-20:
Any progress in this direction?

mantriyogesh, 2022-12-20:
Unfortunately not yet.

whstudio123, 2023-01-30:
@mantriyogesh Anything new？The code really matters for learning and modifying.
有什么更新的进展吗？即使是一个体验不好的一个版本？