espressif/esp-hosted #148: SDIO and SPI parameters

giuliobenetti, 2022-08-22:
Hello,

do you plean to use a DT node or just pass SDIO and SPI settings as parameters?
This to understand in which way to contribute. Linux is moving more and more towards DT but I see you're using 'resetpin' as parameter.
SPI has a problem with Linux >= 5.16 since an API is not there anymore.

Can you please provide me the direction you want to take so I can work and contribute to it?

Thank you!

mantriyogesh, 2022-08-23:
Understood your concern.
I will come up what we can do in some time, to keep all pins, at both DT and/or through parameters.

Can you please explain, what do you mean by `SPI has a problem with Linux >= 5.16 since an API is not there anymore.`

giuliobenetti, 2022-08-23:
> Can you please explain, what do you mean by `SPI has a problem with Linux >= 5.16 since an API is not there anymore.`

Starting from Linux 5.16 spi_busnum_to_master() has been removed since no driver was using it anymore:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=bdc7ca008e1f5539e891187032cb2cbbc3decb5e
This makes driver to fail and I still have to find a work-around for it. Reimplement the function locally is a very big portion of code since spi_busnum_to_master() has a lot of dependencies. Maybe we can achieve its function through another API.

giuliobenetti, 2022-08-23:
@mantriyogesh my idea was to contribute with the fix for this problem and to do that I needed to know how you wanted to face the busses parameters.

But if you provide the fix it’s even better :-)

mantriyogesh, 2022-08-23:
Buses parameters should be by Device Trees.
But to know the actual pin used (non-standard GPIOs, like Data ready), we will anyway have to inform the module. So overall, it is better that we will keep the GPIOs configured through both, DT and through module parameters both.

giuliobenetti, 2022-08-23:
Ok, thank you for explaining me. I close this Issue then.

mantriyogesh, 2022-08-23:
Please re-open this issue. We will get you a working fix, then we can close.

giuliobenetti, 2022-08-23:
Ok, thank you

ricoheller, 2022-11-15:
hi @mantriyogesh and @giuliobenetti 

I did some changes in the code so I am using only information from DTS.
My DTS is now like this:

```
/* bluetooth 1 */
&ecspi4 {
    fsl,spi-num-chipselects = <1>;
    pinctrl-names = "default";
    pinctrl-0 = <&pinctrl_ecspi4 &pinctrl_ecspi4_cs>;
    cs-gpios = <&gpio3 3 GPIO_ACTIVE_LOW>;
    status = "okay";

    bt1: spi@0 {
        compatible = "espressif,esp32-c3";
        reg = <0>;
        spi-max-frequency = <10000000>; /* shall not exceed 60 MHz with ESP32-C3. */
        enable-gpios = <&gpio4 15 GPIO_ACTIVE_HIGH>;
        handshake-gpios = <&gpio4 14 GPIO_ACTIVE_HIGH>;
        dataready-gpios = <&gpio4 13 GPIO_ACTIVE_HIGH>;
        status = "okay";
    };
};
```

Code patch is attached.

[0005-esp32-parameter-pins.patch.txt](https://github.com/espressif/esp-hosted/files/10012210/0005-esp32-parameter-pins.patch.txt)

mantriyogesh, 2022-11-15:
Interesting @ricoheller.
Thank you for your contribution.