espressif/esp-hosted #150: ESP32 C3 new version release date

vijay-yaga, 2022-08-24:
We are good to hear that hosted application working with wpa-supplicant
but we are not using ESP32 WROOM 32 using ESP32 C3
when can we expect the new hosted firmware version for C3 

Thankyou
Vijay

mantriyogesh, 2022-08-24:
Hello @vijay-yaga 

Thank you for your query.
We are right now working over it. You can expect in upcoming release in couple of weeks.

vijay-yaga, 2022-08-25:
Hi @mantriyogesh 
Thanks for your support 
currently i am trying to compile  new release esp-hosted-release-v1.0.1 using the cross compiler arm-linux-gnueabihf-
but i am getting error in the function  process_rx_packet in line (skb_put_data(eap_skb, skb->data, skb->len);)
**/main.c:661:4: error: implicit declaration of function ‘skb_put_data’**
can you provide me the solution to compile the esp32 wroom 32 release-v1.0.1 as soon as possible
thank you

regards
vijay

mantriyogesh, 2022-08-29:
Can you please provide:
1. Git commit you are using in Linux side?
2. Command `uname -a` output
3. Complete build log of compilation
4. Does including [#include <linux/skbuff.h](https://elixir.bootlin.com/linux/latest/source/include/linux/skbuff.h)>) help?

vijay-yaga, 2022-08-29:
1.Git commit you are using in Linux side?
A. we are downloaded the esp-hosted-release-v1.0.1 and we are trying to cross compile to arm board 
2.Command uname -a output
A. Linux ubuntu 5.4.0-124-generic #140~18.04.1-Ubuntu SMP Fri Aug 5 11:43:34 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
 we have given cross compile and kernel path in the Makefile as shown below
 
**CONFIG_ENABLE_MONITOR_PROCESS = n

# Toolchain Path
CROSS_COMPILE := /home/user/work/toolchain_hf/bin/arm-linux-gnueabihf-
# Linux Kernel header
KERNEL := /home/user/NUC970_Buildroot-master/output/build/linux-master
# Architecture
ARCH := arm

#Default interface is sdio
MODULE_NAME=esp32_spi

#Targets passed overrrides default value
ifeq ($(target), sdio)
	MODULE_NAME=esp32_sdio
endif

ifeq ($(target), spi)
	MODULE_NAME=esp32_spi
endif

ifeq ($(CONFIG_ENABLE_MONITOR_PROCESS), y)
	EXTRA_CFLAGS += -DCONFIG_ENABLE_MONITOR_PROCESS
endif

EXTRA_CFLAGS += -I$(PWD)/include -I$(PWD)

ifeq ($(MODULE_NAME), esp32_sdio)
	EXTRA_CFLAGS += -I$(PWD)/sdio
	module_objects += sdio/esp_sdio.o sdio/esp_sdio_api.o
endif

ifeq ($(MODULE_NAME), esp32_spi)
	EXTRA_CFLAGS += -I$(PWD)/spi
	module_objects += spi/esp_spi.o
endif

PWD := $(shell pwd)

obj-m := $(MODULE_NAME).o
$(MODULE_NAME)-y := esp_bt.o main.o esp_cmd.o esp_wpa_utils.o esp_cfg80211.o $(module_objects)

all: clean
	make ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) -C $(KERNEL) M=$(PWD) modules

clean:
	rm -rf *.o sdio/*.o spi/*.o *.ko
	make ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) -C $(KERNEL) M=$(PWD) clean**
3.Complete build log of compilation

**rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=arm CROSS_COMPILE=/home/user/work/toolchain_hf/bin/arm-linux-gnueabihf- -C /home/user/NUC970_Buildroot-master/output/build/linux-master M=/home/user/work/esp-hosted-release-v1.0.1/host clean
make[1]: Entering directory '/home/user/NUC970_Buildroot-master/output/build/linux-master'
  CLEAN   /home/user/work/esp-hosted-release-v1.0.1/host/.tmp_versions
make[1]: Leaving directory '/home/user/NUC970_Buildroot-master/output/build/linux-master'
make ARCH=arm CROSS_COMPILE=/home/user/work/toolchain_hf/bin/arm-linux-gnueabihf- -C /home/user/NUC970_Buildroot-master/output/build/linux-master M=/home/user/work/esp-hosted-release-v1.0.1/host modules
make[1]: Entering directory '/home/user/NUC970_Buildroot-master/output/build/linux-master'
  CC [M]  /home/user/work/esp-hosted-release-v1.0.1/host/esp_bt.o
  CC [M]  /home/user/work/esp-hosted-release-v1.0.1/host/main.o
/home/user/work/esp-hosted-release-v1.0.1/host/main.c: In function ‘process_rx_packet’:
/home/user/work/esp-hosted-release-v1.0.1/host/main.c:657:8: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
    eth = skb_put(eap_skb, ETH_HLEN);
        ^
/home/user/work/esp-hosted-release-v1.0.1/host/main.c:662:4: error: implicit declaration of function ‘skb_put_data’ [-Werror=implicit-function-declaration]
    skb_put_data(eap_skb, skb->data, skb->len);
    ^
cc1: some warnings being treated as errors
scripts/Makefile.build:277: recipe for target '/home/user/work/esp-hosted-release-v1.0.1/host/main.o' failed
make[2]: *** [/home/user/work/esp-hosted-release-v1.0.1/host/main.o] Error 1
Makefile:1441: recipe for target '_module_/home/user/work/esp-hosted-release-v1.0.1/host' failed
make[1]: *** [_module_/home/user/work/esp-hosted-release-v1.0.1/host] Error 2
make[1]: Leaving directory '/home/user/NUC970_Buildroot-master/output/build/linux-master'
Makefile:44: recipe for target 'all' failed
make: *** [all] Error 2**

4.Does including #include <linux/skbuff.h>help?
A. No Still we are getting same error 

thank you for your response  and we are waiting for your solution please provide ASAP

regards
vijay

vijay-yaga, 2022-09-14:
Hi @mantriyogesh 

can you please suggest me some solution we are stuck in the project

regards
vijay

mantriyogesh, 2022-09-14:
Actually, we tried to build over the 5.15 kernel (latest kernel supported by Raspberry) & 4.19.115-v7+ (old one).
Both kernel versions are building fine.

For the specific kernel module, we will have to build complete kernel, which might take some time.
But anyway, we do not observe such issue in building.

Should I assume by, `Linux ubuntu 5.4.0-124-generic https://github.com/espressif/esp-hosted/issues/140~18.04.1-Ubuntu SMP Fri Aug 5 11:43:34 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
this is your native (laptop) kernel.
Is it the same for target cross-compile kernel (arm) in your case?

mantriyogesh, 2022-09-16:
Hello @vijay-yaga 

It looks other people are also facing issue with kernel building, but on different kernel version.

Can you please check if some code changes as part of https://github.com/espressif/esp-hosted/issues/156 works to you?
Code change submitted by author, @wegherfe in awaiting PR, https://github.com/espressif/esp-hosted/pull/157.

wegherfe, 2022-09-16:
Hello.
I faced the same issue. The code change is simple, just edit `include/linux/skbuff.h` to add the new inline function definition, like below:

```C
static inline void *skb_put_data(struct sk_buff *skb, const void *data,
				 unsigned int len)
{
	void *tmp = skb_put(skb, len);

	memcpy(tmp, data, len);

	return tmp;
}
```
I did not pass this change to my PR because it is out of ESP tree.
Maybe this definition could be protected with following guard and added to esp.h main header:

```C
#if LINUX_VERSION_CODE < KERNEL_VERSION(4,13,0)
// Put function definition here
#endif /* LINUX_VERSION_CODE < KERNEL_VERSION(4,13,0) */
```

vijay-yaga, 2022-09-24:
Hi @mantriyogesh 
we have successfully  compiled the version esp-hosted-release-v1.0.1 in linux kernel version 4.8 but we are facing a command response issue and the log is given below 

**insmod esp32_spi.ko resetpin=67
esp32_spi: loading out-of-tree module taints kernel.
master bus num -1030619136
ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
/mnt/sda1 #
Received ESP bootup event
EVENT: 3
EVENT: 2
EVENT: 0
EVENT: 1
esp32: process_fw_data ESP chipset's last reset cause:
POWERON_RESET
esp32: ESP Firmware version: 1.0.1
ESP chipset detected [esp32]
random: nonblocking pool is initialized
esp32: Command[1] timed out
esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: 0xffffffea
ESP peripheral capabilities: 0x78
ESP Bluetooth init
Capabilities: 0x78. Features supported are:
         * WLAN on SPI
         * BT/BLE
           - HCI over SPI
           - BT/BLE dual mode**

 eps32 log is given below

**ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14848
load:0x40080400,len:3792
entry 0x40080694
I (27) boot: ESP-IDF v4.4.1-472-gc9140caf8c-dirty 2nd stage bootloader
I (27) boot: compile time 21:39:54
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 factory          factory app      00 00 00010000 00100000
I (99) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (106) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (114) boot: End of partition table
E (118) esp_image: image at 0x110000 has invalid magic byte (nothing flashed here?)
E (126) boot: OTA app partition slot 0 is not bootable
I (132) boot_comm: chip revision: 1, min. application chip revision: 0
I (139) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=16370h ( 90992) map
I (181) esp_image: segment 1: paddr=00026398 vaddr=3ffbdb60 size=05548h ( 21832) load
I (190) esp_image: segment 2: paddr=0002b8e8 vaddr=40080000 size=04730h ( 18224) load
I (197) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=77d08h (490760) map
I (375) esp_image: segment 4: paddr=000a7d30 vaddr=40084730 size=18e3ch (101948) load
I (417) esp_image: segment 5: paddr=000c0b74 vaddr=50000000 size=00010h (    16) load
I (432) boot: Loaded app from partition at offset 0x10000
I (432) boot: Disabling RNG early entropy source...
I (444) cpu_start: Pro cpu up.
I (444) cpu_start: Starting app cpu, entry point is 0x4008125c
I (0) cpu_start: App cpu up.
I (460) cpu_start: Pro cpu start user code
I (460) cpu_start: cpu freq: 240000000
I (460) cpu_start: Application information:
I (465) cpu_start: Project name:     network_adapter
I (470) cpu_start: App version:      293978e
I (475) cpu_start: Compile time:     Jul 20 2022 21:39:33
I (481) cpu_start: ELF file SHA256:  4989d9e9bda49b86...
I (487) cpu_start: ESP-IDF:          v4.4.1-472-gc9140caf8c-dirty
I (494) heap_init: Initializing. RAM available for dynamic allocation:
I (501) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (507) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (513) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (520) heap_init: At 3FFC74E8 len 00018B18 (98 KiB): DRAM
I (526) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (532) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (539) heap_init: At 4009D56C len 00002A94 (10 KiB): IRAM
I (546) spi_flash: detected chip: generic
I (549) spi_flash: flash io: dio
I (555) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (568) FW_MAIN: *********************************************************************
I (574) FW_MAIN:                 ESP-Hosted Firmware version :: 1.0.1              
I (582) FW_MAIN:                 Transport used :: SPI
I (588) FW_MAIN: *********************************************************************
I (594) FW_MAIN: Supported features are:
I (598) FW_MAIN: - WLAN over SPI
I (602) FW_BT: - BT/BLE
I (604) FW_BT:    - HCI Over SPI
I (608) FW_BT:    - BT/BLE dual mode
I (610) FW_MAIN: Capabilities: 0x78
I (624) system_api: Base MAC address is not set
I (624) system_api: read default base MAC address from EFUSE
I (628) BTDM_INIT: BT controller compile version [5688ed5]
I (632) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (962) FW_MAIN: ESP Bluetooth MAC addr: a8-48-fa-8f- a-7e

I (964) FW_SPI: Using SPI interface
I (964) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (970) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1978) FW_MAIN: Initial set up done**

can you guide me to solve this issue 

Regards
vijay

mantriyogesh, 2022-09-24:
Hello @vijay-yaga 

Kernel building is added till 3.10 kernel version in latest commits of ESP-Hosted, which should ease you out on building side.

WRT recent query, it is even puzzling, why the ESP → Host SPI first transaction worked fine. But first transaction from Host→ESP did not work.

As I remember, you had used ESP-Hosted-FG _i.e._ First generation as well. Is that correct? If yes, did you face issues like this there?

mantriyogesh, 2022-09-24:
Can you please use the latest master and try again with all prints with '**KERN_DEBUG**' enabled in esp_hosted_ng/host/esp_cmd.c?

Please note that there is directory structure change in latest commit. I suggest to back up your changes if any and clean up and git checkout to master, git pull and merge your changes again if any.

mantriyogesh, 2022-09-26:
@vijay-yaga  As I remember, you also had used ESP-Hosted-FG (First Generation). Was it working on this same set-up?

vijay-yaga, 2022-09-27:
Hi @mantriyogesh 
yes we have used release 4.0 before and its working fine in esp32 c3 module but in that i was unable to detect the disconnected state of WIFI
now we are using esp32 wroom32D module and master branch NG

Regards
vijay

mantriyogesh, 2022-09-27:
Actually, In ESP-Hosted-FG, we have provided the async control path on master, which will actually leverage you with events like Wi-Fi disconnect. Other async events like heartbeat, SoftAP station left the association etc. are also added, which you can easily use and integrate with existing solution.

WRT NG,
Considering FG has worked earlier, I am not going to suspect basic connection and basic issues what we had already cleared while porting to FG in your case. To understand scenario better, We would have to give you a debug patch & debug ESP binary to further understand the issue.
Will that be okay with you?

vijay-yaga, 2022-09-28:
Hi @mantriyogesh 
yes please provide me the debug patch and binaries so that we can analyze the problem and we can get some solution 

Regards
vijay

mantriyogesh, 2022-09-29:
Extremely sorry for the delay, I will try to give it today.

mantriyogesh, 2022-09-30:
Hello @vijay-yaga 

PFA [GH_150_30sep22.tgz](https://github.com/espressif/esp-hosted/files/9683501/GH_150_30sep22.tgz).
This is not a release firmware. This contains a lot of debug info & hence unfit for production environment.
Please refer readme.txt in the tarball for further steps.

vijay-yaga, 2022-10-01:
Hi @mantriyogesh 

we capture the debug log from esp 32 module and linux kernel
please find the below log for esp32 module

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14848
load:0x40080400,len:3792
entry 0x40080694
I (27) boot: ESP-IDF v4.4.1-593-g6d85d53cee-dirty 2nd stage bootloader
I (27) boot: compile time 14:29:28
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 factory          factory app      00 00 00010000 00100000
I (99) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (106) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (114) boot: End of partition table
I (118) boot: Defaulting to factory image
I (123) boot_comm: chip revision: 1, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=15918h ( 88344)                                                                                        p
I (170) esp_image: segment 1: paddr=00025940 vaddr=3ffb0000 size=038bch ( 14524)                                                                                        ad
I (176) esp_image: segment 2: paddr=00029204 vaddr=40080000 size=06e14h ( 28180)                                                                                        ad
I (188) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=73464h (472164)                                                                                        p
I (359) esp_image: segment 4: paddr=000a348c vaddr=40086e14 size=0dd88h ( 56712)                                                                                        ad
I (383) esp_image: segment 5: paddr=000b121c vaddr=50000000 size=00010h (    16)                                                                                        ad
I (393) boot: Loaded app from partition at offset 0x10000
I (393) boot: Disabling RNG early entropy source...
I (405) cpu_start: Pro cpu up.
I (405) cpu_start: Starting app cpu, entry point is 0x400811e4
I (0) cpu_start: App cpu up.
I (422) cpu_start: Pro cpu start user code
I (422) cpu_start: cpu freq: 240000000
I (422) cpu_start: Application information:
I (426) cpu_start: Project name:     network_adapter
I (432) cpu_start: App version:      df0fe64-dirty
I (437) cpu_start: Compile time:     Sep 30 2022 15:11:50
I (443) cpu_start: ELF file SHA256:  770a8f25366a28b3...
I (449) cpu_start: ESP-IDF:          v4.4.1-593-g6d85d53cee-dirty
I (456) heap_init: Initializing. RAM available for dynamic allocation:
I (463) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (469) heap_init: At 3FFB7688 len 00028978 (162 KiB): DRAM
I (476) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (482) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (488) heap_init: At 40094B9C len 0000B464 (45 KiB): IRAM
I (496) spi_flash: detected chip: generic
I (499) spi_flash: flash io: dio
I (504) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (518) FW_MAIN: ***************************************************************                                                                                        ***
I (524) FW_MAIN:                 ESP-Hosted Firmware version :: 1.0.1                                                                                                   
I (532) FW_MAIN:                 Transport used :: SPI
I (538) FW_MAIN: ***************************************************************                                                                                        ***
I (544) FW_MAIN: Supported features are:
I (548) FW_MAIN: - WLAN over SPI
I (552) FW_MAIN: Capabilities: 0x20
[... 1390 lines cut ...]
spi_rx: c2a91180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00        ..............

Received ESP bootup event
EVENT: 3
EVENT: 2
EVENT: 0
EVENT: 1
esp32: process_fw_data ESP chipset's last reset cause:
POWERON_RESET
esp32: ESP Firmware version: 1.0.1
ESP chipset detected [esp32]
spi_tx: c2aa7720: 00 00 01 00 06 00 0c 00 00 00 00 00 01 00 00 00  ................
spi_tx: c2aa7730: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
esp32: Command[1] timed out
esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
esp32: add_network_iface() failure, ret: -1
network iterface init failed
ESP peripheral capabilities: 0x20
Capabilities: 0x20. Features supported are:
         * WLAN on SPI
  
where i am getting the issue can you please suggest the some solution so that i can use esp32 wroomd in my project

Regards
vijay

mantriyogesh, 2022-10-03:
1. Not understanding, why logs highlighted 
[image]
from `console+tail_var_log_kern.log` in attached tarball, have not captured in logs.

Is the patch provided in tarball applied correctly?
Part of readme.txt:
[image]

2. Observation from your above host log,
```sh
spi_tx: c2aa7720: 00 00 01 00 06 00 0c 00 00 00 00 00 01 00 00 00 ................
spi_tx: c2aa7730: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
```
is sent from host (spi tx). but there is no mapping RX transaction at ESP32.
I was actually expecting `rx_cmd` prints, like one in reference logs:
[image]

2.1 The question is, are all the connections, specially **handshake** and **MOSI** are connected & functioning correctly?
2.2 Handshake pin working or not can be localized with additional manual print **over debug patch** in `spi_interrupt_handler()`
https://github.com/espressif/esp-hosted/blob/da792ce76501831a007202d719f4bd95a95a3e0d/esp_hosted_ng/host/spi/esp_spi.c#L74-L81
2.3 Although I do not have suspect on this, but on similar basis, data ready pin working or not can be localized by adding additional manual print **over debug patch** in `spi_data_ready_interrupt_handler()`
https://github.com/espressif/esp-hosted/blob/da792ce76501831a007202d719f4bd95a95a3e0d/esp_hosted_ng/host/spi/esp_spi.c#L65-L72
2.4 Can you verify / replace jumper wire on MOSI?
2.5 Is there any change in device tree since your working FG solution in terms of all SPI, Data ready, Reset & handshake pins?

mantriyogesh, 2022-10-08:
Is there any update @vijay-yaga ?

mantriyogesh, 2022-10-10:
Any update / issues @vijay-yaga

vijay-yaga, 2022-10-10:
Hi @mantriyogesh 
we attached patch applied correctly 
and one more thing we have connected esp32 pins
handshake pin connected to Io02(pin no 24)
data read pin connected to Io04(pin no 26)
reset pin connected  to EN(pin no 03)
we have added all debug messages and changed the debug log level

kernel log is given below

**esp_reset, ESP32: Triggering ESP reset.
ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
spi data ready interrupt handeler 1
spi data ready interrupt handeler 2spi interrupt handeler 1
spi interrupt handeler 2esp: esp_spi_work:403 trans_ready:1 rx_pending:1
spi interrupt handeler 1spi interrupt handeler 2
esp: process_rx_buf:319 <3>spi_rx: c2a71160: 03 00 00 00 22 00 0c 00 94 00 00 00 01 00 13 00  ...."...........
spi_rx: c2a71170: 12 03 01 00 02 01 0a 00 01 20 01 07 01 00 01 01  ......... ......
spi_rx: c2a71180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00        ..............
esp: process_rx_packet:657 internal
esp: esp_spi_work:403 trans_ready:1 rx_pending:0

Received ESP bootup event
esp: process_event_esp_bootup:179
esp: process_event_esp_bootup:191 <6>EVENT: 3
EVENT: 2
EVENT: 0
EVENT: 1
esp32: process_fw_data ESP chipset's last reset cause:
POWERON_RESET
esp32: ESP Firmware version: 1.0.1
ESP chipset detected [esp32]
esp: esp_add_card:402
esp: esp_commands_setup:1173 esp: recycle_cmd_node:154
esp: recycle_cmd_node:154 esp: recycle_cmd_node:154
esp: recycle_cmd_node:154 esp: recycle_cmd_node:154
esp: recycle_cmd_node:154 esp: recycle_cmd_node:154
esp: recycle_cmd_node:154 esp: recycle_cmd_node:154
esp: recycle_cmd_node:154 esp: recycle_cmd_node:154
esp: recycle_cmd_node:154 esp: recycle_cmd_node:154
esp: recycle_cmd_node:154 esp: recycle_cmd_node:154
esp: recycle_cmd_node:154 esp: recycle_cmd_node:154
esp: recycle_cmd_node:154 esp: recycle_cmd_node:154
esp: recycle_cmd_node:154 <7>esp: esp_commands_setup:1194 ESP_CMD_INIT_DONE set
esp: add_network_iface:375
esp: esp_cfg80211_register:322 esp: esp_cfg80211_register:354
esp: esp_cfg80211_register:357 0
esp: wait_and_decode_cmd_resp:169
esp: wait_and_decode_cmd_resp:181 <7>esp32: Processing Command [0x1]
esp: esp_spi_work:403 trans_ready:1 rx_pending:0
spi_tx: c2a70860: 00 00 01 00 06 00 0c 00 00 00 00 00 01 00 00 00  ................
spi_tx: c2a70870: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
spi interrupt handeler 1
spi interrupt handeler 2esp: process_rx_buf:319
esp: esp_spi_work:403 trans_ready:1 rx_pending:0esp: wait_and_decode_cmd_resp:188
esp: wait_and_decode_cmd_resp:192 <3>esp32: Command[1] timed out
esp: recycle_cmd_node:154
esp: wait_and_decode_cmd_resp:241 <3>esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
esp: add_network_iface:391
esp: add_network_iface:396 <3>esp32: add_network_iface() failure, ret: -1
network iterface init failed
esp: process_event_esp_bootup:303
ESP peripheral capabilities: 0x20
Capabilities: 0x20. Features supported are:
         * WLAN on SPI
esp: process_event_esp_bootup:309**

esp32 module log is given below

**I (1656) FW_MAIN: Initial set up done
[... 80 lines cut ...]
I (610) wifi_init: tcpip mbox: 32
I (614) wifi_init: udp mbox: 6
I (616) wifi_init: tcp mbox: 6
I (620) wifi_init: tcp tx win: 5744
I (624) wifi_init: tcp rx win: 5744
I (626) wifi_init: tcp mss: 1440
I (630) wifi_init: WiFi IRAM OP enabled
I (634) wifi_init: WiFi RX IRAM OP enabled
I (638) FW_SPI: Using SPI interface
I (640) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (648) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1656) FW_MAIN: Initial set up done
I (1664) FW_SPI: SPI_TX:
I (1666) FW_SPI: 0x3ffc928c   03 00 00 00 22 00 0c 00  94 00 00 00 01 00 13 00  |...."...........|
I (1666) FW_SPI: 0x3ffc929c   12 03 01 00 02 01 0a 00  01 20 01 07 01 00 01 01  |......... ......|
I (1746) FW_SPI: SPI_TX:
I (1746) FW_SPI: 0x3ffb9448   ff 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (1748) FW_SPI: 0x3ffb9458   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|**

we have changed jumper wires and we re verified the all hardware connections every thing is fine

can you provide me the which pins i have to connect the spi to my controller board because there are two spi in esp32 wroom32D

regards
vijay

mantriyogesh, 2022-10-11:
I could not understand what do you mean by

> and one more thing we have connected esp32 pins
handshake pin connected to Io02(pin no 24)
data read pin connected to Io04(pin no 26)
reset pin connected to EN(pin no 03)

Can you please let me know, what the connections are?
Also are you using Raspberry Pi or some different Linux?

mantriyogesh, 2022-10-11:
[image]

These are the pin connections at ESP side supported in [GH_150_30sep22.tgz](https://github.com/espressif/esp-hosted/files/9683501/GH_150_30sep22.tgz)

You have to stick with ESP side connections.
> pin no 24, 25 & 3 
Are these pins used at Host Linux? If so, yes, **At Linux side**, you can choose any free (unused in Device Tree, unclaimed by any function or application)  GPIOs for Data Ready, Handshake & Reset Pin

vijay-yaga, 2022-10-11:
Hi @mantriyogesh 
we are using Buildroot Linux 
[image]
We connected to same pins as shown above and 24,26 and 3 pins are free at host side and they working fine 
and we are using the esp32 module only

earlier also we have worked FG version in Buildroot Linux only 

as per log esp32 to host is working but host to esp32 is having some problem and we are trying to chnage the module and check mean wile can you tell me is there any issue in spi communication or some other thing based on Log provided above 

Regards 
vijay

mantriyogesh, 2022-10-11:
Actually, I see that ESP ⇒ Host msg is being sent.
So there is no problem in Data Ready, SPI CLK, SPI MISO, SPI CS.
Remaining pins, SPI MOSI & Handshake could be doubted. From logs, Even the handshake is getting interrupts. Only MOSI is little doubtful. Try changing cable for MOSI may be?

Some suggestions/question before you replace the ESP32 module:

1. <ins>Imp</ins>: Is this same Linux & ESP you had tested FG, and it was working fine?
2. As you are ESP32, can you please try commenting `cs_change` line in esp_spi.c
 
[image]

As described in [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#124-peripheral-configurations)?

3. Reduce clock to 1MHz

4. Change the SPI mode for both ESP and Linux, for example `SPI mode 3` at both ESP & Linux

5. Try SPI mode mismatch. Why It may work? SPI hardwares behave differently WRT time. Some times, you might face timing issues for ESP32. Please refer: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_slave.html#restrictions-and-known-issues
How to adjust the timing is a different thing to do, but we can easily judge if SPI mode is mismatched in slave and host. 
If this works, there would be different methods to adjust timing.
 
Point (3), (4) & partially (5) cannot be tested for NG as You might need binaries from me for different SPI modes or frequency.
But alternatively you can try ESP-Hosted-FG for (3), (4), (5) time being.

vijay-yaga, 2022-10-14:
Hi @mantriyogesh 
mean while we are woriking on esp-hosted-fg for c3 can you provide me the binaries for esp-hosted-fg for c3 module

Reagrds
vijay

vijay-yaga, 2022-10-18:
Hi @mantriyogesh can you please share the esp-hosted-fg version ESP32-C3 Module prebuilt binaries ASAP. I would like to check control path API from the FG version itself, as we are running out of time.

mantriyogesh, 2022-10-18:
Hello @vijay-yaga 
You can directly build them from latest **master** checked out.

For NG, as because we have not yet exposed the source code, we had to provide the binaries in case of ESP32. (ESP32-C3 is also available now with NG, I can give you for evaluation, if you need)
For FG, code can directly be built and flashed for ESP32-C3.

Please make sure the same flavour (NG/FG) & same git commit version is used for ESP and Host building.

mantriyogesh, 2022-10-18:
You can use latest master with 71381089694e1fbca4d7f15889019413fa1ff875 commit.
Do let us know if you have any issues..

vijay-yaga, 2022-10-18:
Hi @mantriyogesh 

ya ok we will compile the esp32-c3 binaries and we will test the FG code and in parallel provide me the ESP32-C3 NG details so that we test the NG code in c3

ricoheller, 2022-10-20:
Hello @mantriyogesh 

We have a custom board running Yocto Linux with two ESP32-C3 modules.
Communication with ESP modules is via SPI.

We have already ported the ESP Hosted **NG** (esp32_spi).
I will try the evaluation binaries you have provided.

Please let me know if we can further help you with the tests.

mantriyogesh, 2022-11-04:
Hello @vijay-yaga 

@ricoheller has already tried and tested the NG for C3.
Did you get a chance to evaluate the NG for C3?

vijay-yaga, 2022-11-09:
Hi @mantriyogesh 

we have successfully ported the ESP32 C3 NG code and we have tested  and its working fine we are able to connect the internet using WPA_SUPPLICANT thanks for your support 

in the mean while can you please provide me the binaries for production level so that we can proceed the production 

thank you
vijay

mantriyogesh, 2022-11-10:
Hello @vijay-yaga 

Thank you for evaluation.
Quick question, Had you faced the issue similar to #169?

WRT C3 release,
We are expecting ESP32-C3 will be released for NG by end of next week.
Code wise, there would be no (or minimal) difference in ESP binaries.

vijay-yaga, 2022-11-10:
Hi @mantriyogesh 

We have not using Bluetooth in our custom board so we have removed Bluetooth initialisation and HCI interface
We have commented the below code for avoiding  the HCI related compilation issues

**/*if (hci_recv_frame(hdev, skb)) {
#else
			if (hci_recv_frame(skb)) {
#endif
				hdev->stat.err_rx++;
			} else {
				esp_hci_update_rx_counter(hdev, *type, skb->len);
			}*/**
Thank you for your support we will use binaries what you given to us 

Regards
vijay

mantriyogesh, 2022-11-10:
Hello @vijay-yaga 

Point noted. The change you made looks fine.
This selective building of Bluetooth will also be supported in upcoming release.

vijay-yaga, 2023-01-03:
Hi mantriyogesh

we have successfully compiled and compiled and loaded the hosted application V1.0.2 and binaries for ESP32-C3
we are able to communicate the ESP32-C3 module using spi communication
we are able to do insert the esp32_spi.ko and ESP32-C3 module is detecting and communicating
but while scanning the WIFI SSID using (iwlist) command we are getting error as shown below
iwlist espsta0 scan
esp_cfg80211_scan
espsta0 No scan results
[Hosted_driver_LOG.txt](https://github.com/espressif/esp-hosted/files/10335516/Hosted_driver_LOG.txt)
[ESP32_C3_LOG.txt](https://github.com/espressif/esp-hosted/files/10335523/ESP32_C3_LOG.txt)

mantriyogesh, 2023-04-19:
Hello @vijay-yaga 
Sorry for the late response.
Have you got it working?