espressif/esp-hosted #151: Esp-hosted NG 1.0.1 SPI raspberry problem

micheledarold, 2022-08-26:
I'm trying the next generation version 1.0.1 in spi mode but I have this problem

kernel Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

dmesg
[   72.102905] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[   72.223712] Bluetooth: Core ver 2.22
[   72.223759] NET: Registered protocol family 31
[   72.223763] Bluetooth: HCI device and connection manager initialized
[   72.223774] Bluetooth: HCI socket layer initialized
[   72.223782] Bluetooth: L2CAP socket layer initialized
[   72.223803] Bluetooth: SCO socket layer initialized
[   89.210479] esp32_spi: loading out-of-tree module taints kernel.
[   89.211874] esp_reset, ESP32: Triggering ESP reset.
[   89.428801] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]

and nothing else, I think the problem is related to OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status

device tree
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

dtoverlay=disable-bt
dtparam=spi=on
core_freq=250
core_freq_min=250

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

i don't undestand how to use spidev_disabler.dts I must manually add to my config.txt?

thank you and best regards

michele

mantriyogesh, 2022-08-29:
Hello @micheledarold 

Considering you are using `ESP-Hosted-NG`,
1. Have you used the release binaries from https://github.com/espressif/esp-hosted/releases/tag/release%2Fv1.0.1 for Flashing `ESP32`?
2. What is git commit of ESP-Hosted at your Raspberry Pi?
3. Is the SPI connected correctly with jumper cables < 10cm?
4. As you use [rpi_init.sh](https://github.com/espressif/esp-hosted/blob/master/host/rpi_init.sh#L139-L142), `spi_diabler.dts` will be automatically built and overlay will be inserted.

It is important to get `Received ESP bootup event` as shown in 
https://github.com/espressif/esp-hosted/blob/master/docs/setup.md -> `2 Select hosted config` -> `SPI Only` -> `Build and load kernel module` -> `Sample log`. This is first message sent from ESP32 to RaspberryPi on SPI transport, which is *missing* in your dmesg log. That means SPI connectivity is not yet established.

I hope you have followed complete documentation, from `SPI Only` -> `Software setup`, `Build and load kernel module`.

You can ignore this warning
```sh
overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
```

micheledarold, 2022-08-30:
hi @mantriyogesh thank you for your reply I close the issue, I should have some problems with wiring, I've moved to my custom board and evrything is working.
Thank you for your support.

closed 2022-08-30 by micheledarold