espressif/esp-hosted #152: esp32c3 can't work in channel 12 and 13

lgl88911, 2022-08-26:
# Env
ESP32C3
ESP-Hosted Release V0.4

# Issue
Channel 1~11 work well. BUT when wifi AP channel work in 12 or 13, the esp32c3 can't connect to AP, below is log:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16c4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
I (30) boot: ESP-IDF v4.4.1-581-gee2029c38a 2nd stage bootloader
I (30) boot: compile time 11:50:59
I (30) boot: chip revision: 3
I (33) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (71) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  3 factory          factory app      00 00 00010000 00100000
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (101) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (109) boot: End of partition table
I (113) boot: Defaulting to factory image
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=1c618h (116248) map
I (144) esp_image: segment 1: paddr=0002c640 vaddr=3fc91200 size=032c8h ( 13000) load
I (147) esp_image: segment 2: paddr=0002f910 vaddr=40380000 size=00708h (  1800) load
I (150) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=935a8h (603560) map
I (253) esp_image: segment 4: paddr=000c35d0 vaddr=40380708 size=109e8h ( 68072) load
I (266) esp_image: segment 5: paddr=000d3fc0 vaddr=50000010 size=00010h (    16) load
I (271) boot: Loaded app from partition at offset 0x10000
I (271) boot: Disabling RNG early entropy source...
I (286) cpu_start: Pro cpu up.
I (294) cpu_start: Pro cpu start user code
I (294) cpu_start: cpu freq: 160000000
I (294) cpu_start: Application information:
I (297) cpu_start: Project name:     network_adapter
I (303) cpu_start: App version:      release/v0.4-5-gcd31d49-dirty
I (309) cpu_start: Compile time:     Aug 14 2022 11:48:55
I (315) cpu_start: ELF file SHA256:  912b6c5a9e3c71f2...
I (321) cpu_start: ESP-IDF:          v4.4.1-581-gee2029c38a
I (328) heap_init: Initializing. RAM available for dynamic allocation:
I (335) heap_init: At 3FC996B0 len 00026950 (154 KiB): DRAM
I (341) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (348) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (355) spi_flash: detected chip: mxic
I (359) spi_flash: flash io: dio
I (363) sleep: Configure to isolate all GPIO pins in sleep state
I (369) sleep: Enable automatic switching of GPIO sleep configuration
I (377) coexist: coexist rom version 9387209
I (382) cpu_start: Starting scheduler.
I (386) NETWORK_ADAPTER: *********************************************************************
I (386) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.4
I (396) NETWORK_ADAPTER:                 Transport used :: SPI
I (406) NETWORK_ADAPTER: *********************************************************************
I (416) NETWORK_ADAPTER: Supported features are:
I (426) NETWORK_ADAPTER: - WLAN over SPI
I (426) ESP_BT: - BT/BLE
I (436) ESP_BT:    - HCI Over SPI
I (436) ESP_BT:    - BLE only
I (446) BTDM_INIT: BT controller compile version [05195c9]
I (446) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07
I (476) system_api: Base MAC address is not set
I (476) system_api: read default base MAC address from EFUSE
I (486) BTDM_INIT: Bluetooth MAC: 60:55:f9:77:5f:b2

I (486) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 60-55-f9-77-5f-b2

I (496) SPI_DRIVER: Using SPI interface
I (496) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (506) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1516) pp: pp rom version: 9387209
I (1516) net80211: net80211 rom version: 9387209
I (1526) NETWORK_ADAPTER: Initial set up done
I (9626) slave_commands: Get station mac address
I (9626) slave_commands: mac [60:55:f9:77:5f:b0]
I (9646) slave_commands: station mode set
I (12626) slave_commands: Retry to connect to the AP
I (14306) slave_commands: Retry to connect to the AP
I (15356) slave_commands: Retry to connect to the AP
I (17046) slave_commands: Retry to connect to the AP
I (18116) slave_commands: Retry to connect to the AP
I (19806) slave_commands: Password incorrect for SSID
```
From the spec of esp32c3, it can be seen that it supports channel 12~13. 
[image]

From commands.c, you can see that the softap maximum channel is limited to 11
```
#define MAX_CHNL_NO             11
```
But I didn't find any restrictions on station, maybe it's in IDF?

# Problem
Why limit channel to 11？ how to remove the limit？ Thank！

mantriyogesh, 2022-08-29:
Hello @lgl88911 ,

Understood your problem.
Thank you for trying and testing both the solutions and overall debugging till this point.

1. For your question WRT, we can verify if it is IDF issue or not.
In the same ESP-IDF environment that you are using, just flash and test `examples/wifi/getting_started/station` Wi-Fi example on your ESP32-C3. If that fails, it would ESP-IDF limitation/issue.

2. Do you have sniffer logs, if yes, can you please attach them?

3. If IDF test is pass as above, We will try to recreate and let you know the update.

lgl88911, 2022-09-03:
Hi @mantriyogesh Thanks for reply.
1. test `examples/wifi/getting_started/station` in channel 13 work well
```
I (503) wifi:new:<13,0>, old:<1,0>, ap:<255,255>, sta:<13,0>, prof:1
I (503) wifi:state: init -> auth (b0)
I (523) wifi:state: auth -> assoc (0)
I (533) wifi:state: assoc -> run (10)
W (943) wifi:<ba-add>idx:0 (ifx:0, c0:a5:dd:76:89:7f), tid:0, ssn:5, winSize:64
I (943) wifi:connected with AP8, aid = 2, channel 13, BW20, bssid = c0:a5:dd:76:                                            89:7f
I (943) wifi:security: WPA2-PSK, phy: bgn, rssi: -51
I (953) wifi:pm start, type: 1

I (953) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt                                            _time: 10000
I (1013) wifi:BcnInt:102400, DTIM:1
I (1843) esp_netif_handlers: sta ip: 192.168.101.112, mask: 255.255.255.0, gw: 1                                            92.168.101.1
I (1843) wifi station: got ip:192.168.101.112
I (1843) wifi station: connected to ap SSID
```
2. Can you tell me how to capture "sniffer logs"? 
3.  IDF test is pass as above

mantriyogesh, 2022-09-03:
Thank you @lgl88911 

We will check on this and let you know..

lgl88911, 2022-09-03:
@mantriyogesh I found that this is a random problem, today I tested channel 13 and it can connect again. I encountered a failure once on channel 6. When it failed to connect, restarting the wireless AP did not improve. I will continue to follow this issue and have new information to share with you.

mantriyogesh, 2022-09-04:
Last comment is with ESP-Hosted or IDF?

lgl88911, 2022-09-04:
> Last comment is with ESP-Hosted or IDF?

ESP-Hosted

mantriyogesh, 2022-09-26:
Hello @lgl88911 

Some updates from our side,
1. Our team have tested the channel 6, 12, 13 for 500 iterations.
We did not observe any issue at our place.
2. Is your environment noisy? For example any microwave over or noisy Bluetooth or Noisy Wi-Fi networks around?
3. Anyway, Can you please send Wi-Fi sniffer logs, to understand the issue? Sniffer logs should be easy to capture with your laptop.
Please note, For sniffer logs, please cross-check if M1 to M4 are captured correctly, otherwise we would not be able to decode the captured messages.

mantriyogesh, 2022-10-31:
@lgl88911 any updates?

lgl88911, 2022-11-01:
@mantriyogesh  Sorry for the late reply
I updated the BSSID flow according to the master code https://github.com/madmachineio/esp-hosted/commit/9d6da38928b586a6062d373b06fdf40b2f58dcc8 , there is no problem of connection failure, but this change seems to have no impact on the connection failure, So I still want to clarify the following questions：
> Is your environment noisy?

There will be wifi signals from other APs, but the connection with the mobile phone is normal for the APs that can't be connected by the esp32 host.

> Can you please send Wi-Fi sniffer logs, to understand the issue?

How to sniff? Is there any document you can refer to?

mantriyogesh, 2022-11-02:
Hello @lgl88911 
As I can see on your prior comments, you are using hosted release `release/v0.4`
 
Actually, Many fixes like strncpy had gone in into master. See https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_control.c#L798

It would be perfect if you move to [latest release](https://github.com/espressif/esp-hosted/releases) for FG.
Please make sure that you use the source code in the release itself.
[image]

For sniffer, We have a script we run on the Linux laptop which uses Wireshark software
```sh
#!/bin/sh
MAC_LINK=wlp0s20f3                      #Change this name to your local wifi interface name 
MONITOR_LINK=rtap0
CHANNEL=6                               #Change this channel to one you are currently using

ifconfig $MAC_LINK down
iw dev $MONITOR_LINK del
iw dev $MAC_LINK interface add $MONITOR_LINK type monitor
ifconfig $MONITOR_LINK up
iwconfig $MONITOR_LINK channel $CHANNEL
wireshark -k -I -i $MONITOR_LINK &
```

lgl88911, 2022-11-03:
@mantriyogesh I'm glad to see the release of the first FG version, but it has changed a lot compared with v0.4. At present, V0.4 seems to be stable. Since migration to FG requires some work, there is no plan to upgrade to FG release.
After changing the strncpy of the bssid and ssid, there is no connection problem. I am going to close this issue. Do you have any other concerns?

mantriyogesh, 2022-11-03:
@lgl88911 , thats true, there are many features added in latest release.
Looks good that you are happy with 0.4.

> Do you have any other concerns?
Interesting to get this question! ! We are good if you are good.
We can close the issue.

lgl88911, 2022-11-03:
@mantriyogesh Thanks for your support, I will upgrade to FG release at an appropriate time. Now I close this issue.

closed 2022-11-03 by lgl88911