espressif/esp-hosted #153: setup.md is missing

vuhailongkl97, 2022-08-29:
Dear authors,

I've checked your guideline, I saw that this page is missing many information for the setup (how to build, test). Please consider to fix it.

 
https://github.com/espressif/esp-hosted/blob/master/docs/setup.md

mantriyogesh, 2022-08-29:
Can you please let us know what exact information you feel is missing?
We can try to make it better visible in that case.

vuhailongkl97, 2022-08-29:
I didn't see any instruction about how to build with command lines, I think something like how to run `rpi_init.sh` which need to be added like previous version (before NG)

mantriyogesh, 2022-08-30:
- [ ] Actually, we tried to make it simpler in this version.

- [ ] Can you please see below tabular setups folded

[image]

- [ ] Then unfold the option one by one.
[image]

- [ ] This way it can help get focus on only what transport combination you use.

- [ ] As you follow ahead, you would get the details of rpi_init.sh as well.

[image]

Hope this answers your question.

vuhailongkl97, 2022-08-30:
@mantriyogesh I got it.

closed 2022-08-30 by vuhailongkl97