espressif/esp-hosted #154: [ESP-Hosted_MCU_Host] - enable/disable bluetooth via build flag

mjdietzx, 2022-09-04:
I have an embedded linux (build root) device interfacing w/ ESP via SDIO and only using wifi. I do not need bluetooth, it is causing compiler warnings and runtime errors. The short term solution is for me to figure out the config/dependencies, but really I think I should be able to just pass a build flag. I see other people are commenting out code, I am hoping to install this library from source w/ out mutating files internal to esp-hosted.

Here are runtime errors I currently get: 
```[   10.296000] esp32_sdio: Unknown symbol hci_alloc_dev (err 0)
[   10.320000] esp32_sdio: Unknown symbol bt_err (err 0)
[   10.344000] esp32_sdio: Unknown symbol hci_unregister_dev (err 0)
[   10.368000] esp32_sdio: Unknown symbol hci_recv_frame (err 0)
[   10.388000] esp32_sdio: Unknown symbol hci_register_dev (err 0)
OK
Starting rngd: modprobe: can't load module esp32_sdio (extra/esp32_sdio.ko): unknown symbol in module, or unknown parameter
```

I may be able to add `CONFIG_BT=m` to solve this problem (any other ideas)? I do think esp-hosted should just be configurable to not include bt. I am seeing other people run into this and comment out code which is not ideal.

For reference, here is my build root `esp_sdio_host.mk`:
```
ESP_SDIO_HOST_VERSION = f209c91ca010c0899b37dfbb978b8c1dba1d57c4
ESP_SDIO_HOST_SITE_METHOD = git
ESP_SDIO_HOST_SITE = https://github.com/espressif/esp-hosted.git
ESP_SDIO_HOST_LICENSE = Apache-2.0
ESP_SDIO_HOST_LICENSE_FILES = COPYING
ESP_SDIO_HOST_INSTALL_STAGING = YES
ESP_SDIO_HOST_MODULE_SUBDIRS = host/linux/host_driver/esp32/

ESP_SDIO_HOST_MAKE_OPTS = \
	ARCH=$(KERNEL_ARCH) \
	CROSS_COMPILE=$(TARGET_CROSS)

$(eval $(kernel-module))
$(eval $(generic-package))
```

In the future, I think it would also be great for you to make a PR to BR to have native support there. embedded linux users would love it

mjdietzx, 2022-09-04:
ie here https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/linux/host_driver/esp32/Makefile, I wish I had, for example, a `BT_ENABLED` flag. It could be enabled by default, but in my `esp_sdio_host.mk`, I'd love to be able to also add:
```ESP_SDIO_HOST_MAKE_OPTS = \
	ARCH=$(KERNEL_ARCH) \
	CROSS_COMPILE=$(TARGET_CROSS) \
        BT_ENABLED=no
```
and have it "just work"

mjdietzx, 2022-09-04:
Related, it would be amazing if you could add linux host directly in the package directory and submitted to the Buildroot upstream project. https://buildroot.org

mantriyogesh, 2022-09-05:
Hello @mjdietzx 

Thank you for your valuable suggestion.
We will try to separate so that it will be easy to omit bluetooth references.