espressif/esp-hosted #155: issue: Unsupported tag in event - Raspberry Pi 4 model B + ESP32 SPI connection issue

KashyapPatalia, 2022-09-09:
Hello, everyone.
I am getting following logs for dmesg (I am using Raspberry Pi 4 model B with 32 bit Raspberry Pi OS Lite + ESP32_Devkitc_v4 with pre built binaries of spi v0.4 - SPI connection):

[ 94.113919] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[ 94.243688] Bluetooth: Core ver 2.22
[ 94.243757] NET: Registered PF_BLUETOOTH protocol family
[ 94.243763] Bluetooth: HCI device and connection manager initialized
[ 94.244099] Bluetooth: HCI socket layer initialized
[ 94.244109] Bluetooth: L2CAP socket layer initialized
[ 94.244125] Bluetooth: SCO socket layer initialized
[ 116.410158] esp32_spi: loading out-of-tree module taints kernel.
[ 116.410925] esp_reset, ESP32: Resetpin of Host is 6
[ 116.411147] esp_reset, ESP32: Triggering ESP reset.
[ 116.411536] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
[ 118.822285]
Received INIT event from ESP32 peripheral
[ 118.822306] EVENT: 2
[ 118.822323] EVENT: 1
[ 118.822337] Unsupported tag in event
[ 118.822350] EVENT: 0
[ 118.822364] ESP peripheral capabilities: 0x78
[ 119.575877] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[ 119.575890] Bluetooth: BNEP filters: protocol multicast
[ 119.575902] Bluetooth: BNEP socket layer initialized
[ 119.588172] NET: Registered PF_ALG protocol family
[ 119.608597] cryptd: max_cpu_qlen set to 1000

I do not know this [ 118.822337] Unsupported tag in event. Also I am not able to establish communication over SPI. It always says timeout & Response not received.
Can anyone please help me ?
TIA.

mantriyogesh, 2022-09-09:
Considering that you are using ESP with 0.4 binaries, are you using the same commit, https://github.com/espressif/esp-hosted/commit/74e707b604280ecb2eafa60ba5ed6993ce57b457 at Raspberry Pi?

KashyapPatalia, 2022-09-09:
> 

I am on branch ESP-Hosted_MCU_Host - latest commit

mantriyogesh, 2022-09-09:
Actually, 0.4 release is based on https://github.com/espressif/esp-hosted/commit/74e707b604280ecb2eafa60ba5ed6993ce57b457. So same commit need to be used at Raspberry Pi.

You can also use latest commit at ESP-Hosted_MCU_Host, But the commit need to be used at both the ESP and Host.
You know the Host building part. For ESP, you would need to build and flash the ESP firmware using https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/Linux_based_host/SPI_setup.md#222-source-compilation (considering SPI only transport setup)

KashyapPatalia, 2022-09-09:
> Considering that you are using ESP with 0.4 binaries, are you using the same commit, [74e707b](https://github.com/espressif/esp-hosted/commit/74e707b604280ecb2eafa60ba5ed6993ce57b457) at Raspberry Pi?

Hi. I have now switched the branch to master and HEAD to the commit [74e707b](https://github.com/espressif/esp-hosted/commit/74e707b604280ecb2eafa60ba5ed6993ce57b457) as you mentioned.

Now I am no longer getting the Unsupported tag in event log in dmesg logs. However I am still not able to get raspberry pi talk to ESP32 over SPI.

Still I am getting :
pi@raspberrypi:~/esp-hosted/host/linux/host_control/c_support $ sudo ./test.out get_wifi_mode
Enable heartbeat with duration 20
            
Control response timed out after 30 sec
Response not received

Requested operation complete
Sleeping for some time just to showcase heartbeat
NULL func, failed to call callback
Failed to send control req 121

mantriyogesh, 2022-09-09:
Please provide dmesg log?

KashyapPatalia, 2022-09-09:
> Please provide dmesg log? Better to restart and try scenario and get fresh dmesg log. Also do not forget to rebuild the application (test.out)

Does these steps.
Find here the logs:
[esp32-serial-log.txt](https://github.com/espressif/esp-hosted/files/9533295/esp32-serial-log.txt)
[dmesg-log.txt](https://github.com/espressif/esp-hosted/files/9533297/dmesg-log.txt)

mantriyogesh, 2022-09-09:
Please `make clean;make` in esp-hosted/host/linux/host_control/c_support.

KashyapPatalia, 2022-09-09:
> Please `make clean;make` in esp-hosted/host/linux/host_control/c_support.

This worked !! Thank you. I'll do my further testings and will let you know if anything.

Again, thanks a lot !!

mantriyogesh, 2022-09-09:
Please help closing the bug once you are done.

mantriyogesh, 2022-09-09:
We suggest you to use second alternative, i.e. latest ESP-Hosted_MCU_Host commit at both ESP & Host, as it contains multiple recent bugfixes and additional features.

KashyapPatalia, 2022-09-09:
Hi again. I am still on https://github.com/espressif/esp-hosted/commit/74e707b604280ecb2eafa60ba5ed6993ce57b457 commit and everything related to WiFi station seems to be working fine. However when I am starting softap mode, my station devices are not able to connect to the created softAP of the ESP32.

Here are the ESP32 serial logs:

I (998995) slave_commands: ESP32 SoftAP is avaliable 
I (999001) NETWORK_ADAPTER: Get softap mac address
I (999005) NETWORK_ADAPTER: AP mac [1a:22:22:22:22:22] 
I (1063607) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (1063609) wifi:station: 22:8b:2c:cd:41:56 join, AID=1, bgn, 20
I (1063653) slave_commands: station 22:8b:2c:cd:41:56 join, AID=1
I (1078199) wifi:station: 22:8b:2c:cd:41:56 leave, AID = 1, bss_flags is 134243, bss:0x3ffd5224
I (1078199) slave_commands: station 22:8b:2c:cd:41:56 leave, AID=1
I (1078199) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1

My station device (my android phone in this case) is basically not able to get IP from the ESP32 Access Point. I have a feeling that it would be a basic miss from my side, but can you please help me with this ?
Thanks in advance.

KashyapPatalia, 2022-09-09:
> We suggest you to use second alternative, i.e. latest ESP-Hosted_MCU_Host commit at both ESP & Host, as it contains multiple recent bugfixes and additional features.

Sure, I will try that later once I am done with all the functional tests on this version.

mantriyogesh, 2022-09-09:
Hello @KashyapPatalia 

Actually, IP address could be assigned with any Linux specific utility.
You can check example mentioned in https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/common/c_demo.md#some-points-to-note for softap.

We are agnostic which utility is being used.

KashyapPatalia, 2022-09-12:
Hi @mantriyogesh 
I have successfully integrated ESP32 with Raspberry Pi. It's working fine for both STA and AP mode. Thanks a lot for helping me out. I am now closing this issue.

closed 2022-09-12 by KashyapPatalia

kashyap-patalia, 2022-11-01:
@mantriyogesh Can you please confirm that on the linux host side, the drivers are esp32 module agnostic, right ?  Meaning I can just replace the esp32 module with esp32c3 and it would just work fine. I have now esp32c3 with me. I am trying to interface it with my rpi setup. I am also facing some issues compiling esp32 firmware binaries.

mantriyogesh, 2022-11-01:
Hello @KashyapPatalia 

Yes the linux side code is agnostic wrt esp chipset.
There is cs_change attribute conditionally enabled for Esp32 (already taken care). The module should work for all supported ESP chipset including ESP32-C3. You might have to flash the C3 ESP-Hosted firmware on ESP32-C3.

To bring your notice, we have recently pushed next release, release/fg-0.0.5, which has many bugfixes and feature set added, which you can use.

Can you please attqch the log where you get errors?

kashyap-patalia, 2022-11-01:
>  Can you please attach the log where you get errors?

[build.log](https://github.com/espressif/esp-hosted/files/9907752/build.log)

mantriyogesh, 2022-11-01:
What is ESP-Hosted commit you are using?
I suggest using latest release or master to build the code.
Please note, we have ported the complete ESP code base to **release/v5.0** branch for **ESP-IDF**.
-> In case you are using old ESP-IDF branch to build, git pull and git checkout release/v5.0 , remove the existing toolchain, install new toolchain
-> Go back to the esp building folder rm -rf sdkconfig build/ and set-target and build and flash for your expected chipset.

mantriyogesh, 2022-11-01:
Please make sure that you use same git commit at both, esp and host when you update code.

mantriyogesh, 2022-11-01:
Some of similar errors popped in https://github.com/espressif/esp-hosted/issues/132

I am not sure which branch is used for ESP-IDF, but that is the problems mostly.
If you are going to use latest release for FG (0.0.5), you might need ESP-IDF with `release/v5.0` branch checked out and tool chain installed.

mantriyogesh, 2022-11-02:
Any update @KashyapPatalia ?