espressif/esp-hosted #158: support for WT32-ETH01

beta-tester, 2022-09-09:
hi, i have a [WT32-ETH01](http://www.wireless-tag.com/portfolio/wt32-eth01/) board with WiFi + ETH.

- can i use this board as well under Linux and MCU?
- if so, can i use its ETH port as well?

mantriyogesh, 2022-09-11:
Currently we do not support ETH as possible transport combination.
However we will definitely check if this is possible option to develop.

mantriyogesh, 2022-09-28:
Thank you for pointing out that @beta-tester .

Yes. It is based on ESP-AT repo.
If that solves your purpose, you can surely use it.

beta-tester, 2022-09-28:
> It is based on ESP-AT repo. If that solves your purpose, you can surely use it.

i still would like to see "ESP-Hosted" on those modules as well.

mantriyogesh, 2022-09-28:
Thank you for all the support.
Definitely, we are looking up how we can support. But it might take some time.