espressif/esp-hosted #159: How to get the firmware of esp?

liangyingy, 2022-09-13:
I can't find the firmware files for esp in the current repo.

mantriyogesh, 2022-09-13:
Files meaning the binary files or the source code of firmware?

liangyingy, 2022-09-13:
Either binary or firmware source code is fine, preferably source code.

mantriyogesh, 2022-09-13:
Hello @liangyingy,

Thank you for your query.

- For ESP-Hosted-NG (Next Generation)
  - Branch
    - Please refer [master](https://github.com/espressif/esp-hosted) branch
  - Releases
    - [NG Release 1.0.1](https://github.com/espressif/esp-hosted/releases/tag/release%2Fv1.0.1) 
  - Source code
    - ESP: Not yet available, because of some technical issues, will be released in upcoming releases
    - Host: [host](https://github.com/espressif/esp-hosted/tree/master/host)
- For ESP-Hosted-FG (First Generation)
  - Branch
    - Please refer [ESP-Hosted_MCU_Host](https://github.com/espressif/esp-hosted/tree/ESP-Hosted_MCU_Host) branch at https://github.com/espressif/esp-hosted
  - Releases
    - [FG release 0.4](https://github.com/espressif/esp-hosted/tree/release/v0.4)
  - Source code
    - ESP
      - [esp/esp_driver/network_adapter](https://github.com/espressif/esp-hosted/tree/ESP-Hosted_MCU_Host/esp/esp_driver/network_adapter) This is the latest source code
    - Host
      - [host](https://github.com/espressif/esp-hosted/tree/ESP-Hosted_MCU_Host/host)

Please check the links hyperlinks provided.
Complete procedure including, How to use ESP, How to build kernel module at host, How to use Wi-Fi / Bluetooth and much more can be referred through the documentation.
Please connect back if something not clear.

liangyingy, 2022-09-13:
Thank you for your answer!

closed 2022-09-13 by liangyingy