espressif/esp-hosted #161: ESP32 Unable to detect Flash Size

wildbiotiger, 2022-09-15:
I'm using a [AT25SF321B](https://www.mouser.com/datasheet/2/698/adet_s_a0010543939_1-2956529.pdf) as the SPI flash for an ESP32-D0WD-V3 IC. I can successfully detect the flash chip, but the size isn't able to be detected. Is there a way to ignore the flash size during runtime? Before we had manually commented out the do_core_init check in the last firmware version (v0.4). Is there something incompatible with the SPI Flash we are using? It seems to correctly flash the chip using the write_flash command, but booting fails.

Output of flash_id command
```
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: f0:08:d1:4c:fb:48
Uploading stub...
Running stub...
Stub running...
Manufacturer: 1f
Device: 8701
Detected flash size: Unknown
Hard resetting via RTS pin...
```

Output once it tries to boot
```
I (13) boot: ESP-IDF v4.4.1-472-gc9140caf8c-dirty 2nd stage bootloader
I (13) boot: compile time 21:33:21
I (13) boot: chip revision: 3
I (16) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (23) boot.esp32: SPI Speed      : 40MHz
I (28) boot.esp32: SPI Mode       : DIO
I (32) boot.esp32: SPI Flash Size : 4MB
I (37) boot: Enabling RNG early entropy source...
I (42) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (61) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (68) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  3 factory          factory app      00 00 00010000 00100000
I (83) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (91) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (98) boot: End of partition table
I (102) boot: Defaulting to factory image
I (107) boot_comm: chip revision: 3, min. application chip revision: 0
I (114) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=15e00h ( 89600) map
I (155) esp_image: segment 1: paddr=00025e28 vaddr=3ffbdb60 size=0553ch ( 21820) load
I (164) esp_image: segment 2: paddr=0002b36c vaddr=40080000 size=04cach ( 19628) load
I (172) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=769dch (485852) map
I (348) esp_image: segment 4: paddr=000a6a04 vaddr=40084cac size=181d4h ( 98772) load
I (389) esp_image: segment 5: paddr=000bebe0 vaddr=50000000 size=00010h (    16) load
I (404) boot: Loaded app from partition at offset 0x10000
I (404) boot: Disabling RNG early entropy source...
I (415) cpu_start: Pro cpu up.
I (416) cpu_start: Starting app cpu, entry point is 0x4008121c
I (410) cpu_start: App cpu up.
I (432) cpu_start: Pro cpu start user code
I (432) cpu_start: cpu freq: 240000000
I (432) cpu_start: Application information:
I (437) cpu_start: Project name:     network_adapter
I (442) cpu_start: App version:      293978e
I (447) cpu_start: Compile time:     Jul 20 2022 21:33:02
I (453) cpu_start: ELF file SHA256:  573ea08fdc881889...
I (459) cpu_start: ESP-IDF:          v4.4.1-472-gc9140caf8c-dirty
I (466) heap_init: Initializing. RAM available for dynamic allocation:
I (473) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (479) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (485) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (492) heap_init: At 3FFD1498 len 0000EB68 (58 KiB): DRAM
I (498) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (504) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (510) heap_init: At 4009CE80 len 00003180 (12 KiB): IRAM
I (518) spi_flash: detected chip: generic
I (521) spi_flash: flash io: dio
E (525) spi_flash: Detected size(0k) smaller than the size in the binary image header(4096k). Probe failed.

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x40081b3e:0x3ffe3a800x400907fd:0x3ffe3aa0 0x4009739d:0x3ffe3ac0 0x400d25ea:0x3ffe3be0 0x400d2746:0x3ffe3c00 0x40081445:0x3ffe3c40 0x400794e0:0x3ffe3c80  |<-CORRUPTED

ELF file SHA256: 573ea08fdc881889

Rebooting...
```

Any ideas on what we can try to fix this issue?

mantriyogesh, 2022-09-16:
Hello @wildbiotiger,

Thank you for trying the ESP-Hosted solution.
This issue could be better answered with ESP-IDF team.

Suggestion: If any GPIOs attached, Can you remove all of them, try to erase flash and re-try flashing?

Can you please check this issue with https://github.com/espressif/esp-idf/issues?

wildbiotiger, 2022-09-16:
I can post the issue there, but the issue is with the SDIO firmware provided in the latest release (v1.0.1). So, I am not able to modify any of the underlying code here. Would you be able to provide the source code that's trying to detect the SPI memory size in the current build? That would help me to frame the question better for the ESP-IDF team.

Other notes:
- SDIO is currently hooked up with the following pulled high through 10k resistors: MTMS, MTDI, MTDO, GPIO2, GPIO4
- When flashing, I pull MTMS, MTDI, MTDO, GPIO2, GPIO4, and GPIO0 low and toggle CHIP_PU to get it into the UART bootloader
- I have also burned the eFuse for VDD_SDIO to be set at 3.3V and verified using an oscilloscope that the voltage to the chip is 3.3V
- No voltage droops or other rail issues have been observed during flashing or anything else, it simply appears to be failing the memory size check
- In both the UART bootloader mode, and the SPI bootloader, the SPI memory size is unable to be detected as I showed above
- There are no other GPIOs connected besides the ones I listed above and the following connected to the SPI flash:
```
SD_DATA0 - (2) Slave Out
SD_DATA1 - (5) Slave In
SD_DATA2 - (7) Hold (IO3)
SD_DATA3 - (3) Write Protect (IO2)
SD_CLK - (6) SCK
SD_CMD - (1) Chip Select
```

mantriyogesh, 2022-09-16:
Can you try to flash ESP-Hosted -FG solution, which is available to flash. There is not much difference other than Wi-Fi related handling. Basically, no change in SDIO or any other functionality.

You can build and flash `ESP32` chipset using 
`ESP-IDF` `release/v4.0` with steps
1. https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/Linux_based_host/Linux_based_readme.md#22-esp-idf-setup
2. https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/Linux_based_host/SDIO_setup.md#222-source-compilation using SDIO

For flashing in fact you can use even IDF examples for SDIO slave driver. If it fails, ESP-IDF would be in better position to understand and resolve these issues quickly. Anyway, Please let us know if any of the above code flashes correctly.

wildbiotiger, 2022-09-16:
Can you publish a build that would disable asserts in the code? That would help get me past THIS particular issue.

mantriyogesh, 2022-09-17:
Forget about ESP-Hosted, 

I want to understand why are you getting, 
`Detected size(0k) smaller than the size in the binary image header(4096k). Probe failed.`

Can you try if SDIO slave example from ESP-IDF from branch, release/v4.0 is getting flashed without these kind of errors.

mantriyogesh, 2022-09-17:
What my point is, if you have something non working at the ESP-IDF level itself, flashing ESP-Hosted would not be much useful and will complicate to understand what is exactly the problem.

Also do you have another chipset, where you can flash the firmware you had flashed?
Possibly, It will help to differentiate the behaviour between working and non working things.

mantriyogesh, 2022-09-17:
Can you please attach sdkconfig ?
Also please check if this issue is related for you:
https://github.com/espressif/esp-idf/issues/4829

wildbiotiger, 2022-09-17:
It’s simply because the chip id the SPI Flash reports generates an invalid size: 0x1F8701 gets passed into [spi_flash_chip_generic.c:129](https://github.com/espressif/esp-idf/blob/5b39159e668d3af5832cedb500ccc4869bf15492/components/spi_flash/spi_flash_chip_generic.c#L129) and returns 0 (I think). So, that’s the place I would start looking.

wildbiotiger, 2022-09-19:
Here's the current sdkconfig from the project.

I think I only had to edit the partition table to expand it for the debug statements and changed DIO to QIO (but I've tried it with DIO with no change in result).

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_INFO is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE=y
CONFIG_BOOTLOADER_LOG_LEVEL=5
CONFIG_BOOTLOADER_SPI_WP_PIN=7
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
# CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT is not set
# CONFIG_SECURE_BOOT_ENABLED is not set
# CONFIG_SECURE_FLASH_ENC_ENABLED is not set
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
# CONFIG_ESPTOOLPY_FLASHMODE_QOUT is not set
# CONFIG_ESPTOOLPY_FLASHMODE_DIO is not set
# CONFIG_ESPTOOLPY_FLASHMODE_DOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
# CONFIG_ESPTOOLPY_FLASHFREQ_40M is not set
# CONFIG_ESPTOOLPY_FLASHFREQ_26M is not set
# CONFIG_ESPTOOLPY_FLASHFREQ_20M is not set
CONFIG_ESPTOOLPY_FLASHFREQ="80m"
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_2MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_16MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
# CONFIG_ESPTOOLPY_BEFORE_NORESET is not set
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_RESET=y
# CONFIG_ESPTOOLPY_AFTER_NORESET is not set
CONFIG_ESPTOOLPY_AFTER="hard_reset"
# CONFIG_ESPTOOLPY_MONITOR_BAUD_9600B is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_57600B is not set
CONFIG_ESPTOOLPY_MONITOR_BAUD_115200B=y
# CONFIG_ESPTOOLPY_MONITOR_BAUD_230400B is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_921600B is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_2MB is not set
[... 681 lines cut ...]
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options
```

wildbiotiger, 2022-09-20:
@mantriyogesh any update on your side? Just to recap:

- Disabling the asserts in the ESP-Hosted build would allow me to bypass the size check if you are able to publish a new build
- Was there anything odd with the response from the SPI Flash that did not match the datasheet or JEDEC specifications? I can report this to the chip manufacturer if so.
- Otherwise, is there anything that should change on the ESP-IDF side with respect to the flash check? It would seem that a custom driver might be a decent option with just overriding the SPI flash size check and probe. If that was the case, how soon could we get this integrated into a new build of the ESP-Hosted firmware?

Please let me know if there's anything that needs to happen on my side.

mantriyogesh, 2022-09-20:
Hello @wildbiotiger ,

Sorry for the late response.
I have asked the ESP-IDF team, if such changes could be published. They might take some time to respond.

In case of urgency, please use the ESP-IDF issue tracker.

mantriyogesh, 2022-09-21:
Is this custom hardware? If yes, what is flash part used on the device?

mantriyogesh, 2022-09-21:
Can you flash 'hello world' example without any changes in menuconfig and get us output?

mantriyogesh, 2022-09-21:
Are you using own flash chip or have you done any changes?

wildbiotiger, 2022-09-21:
@mantriyogesh it is on a custom board. I linked the datasheet for the SPI flash we are using in the first post:
https://www.mouser.com/datasheet/2/698/adet_s_a0010543939_1-2956529.pdf
I’ll have to run the hello world tomorrow.

mantriyogesh, 2022-09-21:
Actually, you can use esp-idf with all your own expected changes, using ESP-Hosted-FG, as the firmware code is available to you and you can do any changes you need in ESP-IDF.

I understand you are running into external flash issues, but getting these up & working is kind of expected before using ESP-Hosted.

Instead of using direct binaries, you can build them in directory https://github.com/espressif/esp-hosted/tree/ESP-Hosted_MCU_Host/esp/esp_driver/network_adapter. 
1. setup IDF, https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/Linux_based_host/Linux_based_readme.md#22-esp-idf-setup -> Change what you need here in IDF
2. Clone ESP-Hosted, https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/Linux_based_host/Linux_based_readme.md#23-esp-hosted-code-repository 
3. build & flash the ESP, https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/docs/Linux_based_host/SPI_setup.md#222-source-compilation

Anyway, I have flagged the flash issue to internal SPI flash team, it might take some time to get response.

wildbiotiger, 2022-09-21:
@mantriyogesh yes, that's what we were doing, and that was build of the MCU firmware was what we were using when I posted the logs originally. We have made the changes there, but we are very interested in getting a version that works with the native Linux Wi-Fi utilities (which is the version that is closed source at the moment).

mantriyogesh, 2022-09-22:
Understood.
I can make the changes and give you. I suppose that's the only change you are looking for.
If not, can you please send the IDF patch and base commit you are using?

mantriyogesh, 2022-09-22:
Hello @wildbiotiger ,

As I get input from the SPI flash team,

```
For this flash, its problem is that, its ID format is not recognized,
and the ID is where the size is usually get from.
```
[image]
```
What I see is that the flash driver used by the app is not compatible with it in the get_size.
Customer need to customize their driver for this point, see: 
https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/examples/storage/custom_flash_driver/components/custom_chip_driver/spi_flash_chip_eon.c

But TBH, This flash will have a lot of places that are not compatible with the default flash driver.
```

Considering the above context, is there any plan B for the SPI flash you can use? As I understand, you might end up in a lot of changes to make it supported. In case this is expected for mass production, you can get in touch with our business team, where they would fasten the support to this problem.

BTW, It is tricky to remember the names, so just explaining the variant/flavours ESP-Hosted supports.
- FG _i.e._ first generation
  - can be used with `Linux` and `MCU`. It has custom control path APIs, you can use.
- NG _i.e._ next generation
  - Specially designed to use Wi-Fi with wpa_supplicant (seam-less Wi-Fi control). Works only with Linux.

Now, In case you want to continue ahead, you have complete source code in above links I had shared previously for FG. You can make the changes in ESP-IDF and get the flashing working on `FG` first. Then let me know the changes in a patch with base commit of ESP-IDF. I will apply that patch to IDF and get you `NG` binary.

Hope this helps.

wildbiotiger, 2022-09-22:
Thanks @mantriyogesh, we will take a look and get back to you on the implementation.

wildbiotiger, 2022-09-23:
@mantriyogesh I've updated a fork of ESP-ID with the AT25 driver here: https://github.com/espressif/esp-idf/pull/9842. Would you please publish a build with this code?

I also found another strange issue somewhat unrelated:

- ESP-IDF version 4.4
- ESP-HOSTED-FG version 0.4
- Enable Log Level Verbose
- SDIO aborts due to recursive mutex in logging function

Backtrace dump from GDB:
```
Reading symbols from build/network_adapter.elf...
(gdb) list *0x40083373
0x40083373 is in lock_acquire_generic (esp32/esp-idf/components/newlib/locks.c:139).
134	    }
135	    BaseType_t success;
136	    if (!xPortCanYield()) {
137	        /* In ISR Context */
138	        if (mutex_type == queueQUEUE_TYPE_RECURSIVE_MUTEX) {
139	            abort(); /* recursive mutexes make no sense in ISR context */
140	        }
141	        BaseType_t higher_task_woken = false;
142	        success = xSemaphoreTakeFromISR(h, &higher_task_woken);
143	        if (!success && delay > 0) {
(gdb) list *0x40081d1a
0x40081d1a is in panic_abort (esp32/esp-idf/components/esp_system/panic.c:402).
397	    esp_apptrace_flush_nolock(ESP_APPTRACE_DEST_TRAX, CONFIG_APPTRACE_POSTMORTEM_FLUSH_THRESH,
398	                              APPTRACE_ONPANIC_HOST_FLUSH_TMO);
399	#endif
400	#endif
401	
402	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
403	    while (1);
404	}
405	
406	/* Weak versions of reset reason hint functions.
(gdb) list *0x40090de1
0x40090de1 is in esp_system_abort (esp32/esp-idf/components/esp_system/esp_system.c:128).
123	    return IDF_VER;
124	}
125	
126	void __attribute__((noreturn)) esp_system_abort(const char *details)
127	{
128	    panic_abort(details);
129	}
(gdb) list *0x4009804a
0x4009804a is in abort (esp32/esp-idf/components/newlib/abort.c:46).
41	
42	    for (size_t i = 0; i < sizeof(str) / sizeof(str[0]); i++) {
43	        strcat(dest, str[i]);
44	    }
45	
46	    esp_system_abort(buf);
47	}
(gdb) 0x40083373
Undefined command: "0x40083373".  Try "help".
(gdb) list *0x40083373
0x40083373 is in lock_acquire_generic (esp32/esp-idf/components/newlib/locks.c:139).
134	    }
135	    BaseType_t success;
136	    if (!xPortCanYield()) {
137	        /* In ISR Context */
138	        if (mutex_type == queueQUEUE_TYPE_RECURSIVE_MUTEX) {
139	            abort(); /* recursive mutexes make no sense in ISR context */
140	        }
141	        BaseType_t higher_task_woken = false;
142	        success = xSemaphoreTakeFromISR(h, &higher_task_woken);
143	        if (!success && delay > 0) {
(gdb) list *0x400834bd
0x400834bd is in _lock_acquire_recursive (esp32/esp-idf/components/newlib/locks.c:167).
162	void IRAM_ATTR _lock_acquire(_lock_t *lock) {
163	    lock_acquire_generic(lock, portMAX_DELAY, queueQUEUE_TYPE_MUTEX);
164	}
165	
166	void IRAM_ATTR _lock_acquire_recursive(_lock_t *lock) {
167	    lock_acquire_generic(lock, portMAX_DELAY, queueQUEUE_TYPE_RECURSIVE_MUTEX);
168	}
169	
170	int IRAM_ATTR _lock_try_acquire(_lock_t *lock) {
171	    return lock_acquire_generic(lock, 0, queueQUEUE_TYPE_MUTEX);
(gdb) list *0x40083536
0x40083536 is in __retarget_lock_acquire_recursive (esp32/esp-idf/components/newlib/locks.c:323).
318	
319	void IRAM_ATTR __retarget_lock_acquire_recursive(_LOCK_T lock)
320	{
321	    check_lock_nonzero(lock);
322	    MAYBE_OVERRIDE_LOCK(lock, &s_common_recursive_mutex);
323	    _lock_acquire_recursive(&lock);
324	}
325	
326	int IRAM_ATTR __retarget_lock_try_acquire(_LOCK_T lock)
327	{
(gdb) list *0x4015370e
0x4015370e is in _vfprintf_r (/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853).
848	/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c: No such file or directory.
(gdb) list *0x40156741
0x40156741 is in vprintf (/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34).
29	/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c: No such file or directory.
(gdb) list *0x4016466d
0x4016466d is in esp_log_writev (esp32/esp-idf/components/log/log.c:200).
195	    esp_log_level_t level_for_tag = s_log_level_get_and_unlock(tag);
196	    if (!should_output(level, level_for_tag)) {
197	        return;
198	    }
199	
200	    (*s_log_print_func)(format, args);
201	
202	}
203	
204	void esp_log_write(esp_log_level_t level,
(gdb) list *0x40096add
0x40096add is in esp_log_write (esp32/esp-idf/components/log/log.c:210).
205	                   const char *tag,
206	                   const char *format, ...)
207	{
208	    va_list list;
209	    va_start(list, format);
210	    esp_log_writev(level, tag, format, list);
211	    va_end(list);
212	}
213	
214	static inline bool get_cached_log_level(const char *tag, esp_log_level_t *level)
(gdb) list *0x400fb4e1
0x400fb4e1 is in sdio_slave_hal_load_buf (esp32/esp-idf/components/hal/sdio_slave_hal.c:667).
662	
663	    if (tail == NULL) {
664	        //no one in the ll, start new ll operation.
665	        sdio_slave_ll_recv_start(hal->slc, desc);
666	        sdio_slave_ll_recv_intr_ena(hal->slc, true);
667	        HAL_LOGV(TAG, "recv_load_buf: start new");
668	    } else {
669	        //restart former ll operation
670	        sdio_slave_ll_recv_restart(hal->slc);
671	        HAL_LOGV(TAG, "recv_load_buf: restart");
(gdb) q
```

wildbiotiger, 2022-09-26:
Hey @mantriyogesh, any chance on getting the pull request integrated and publishing a new build?

mantriyogesh, 2022-09-27:
1. Can you please provide `IDF branch used` and `git commit`, over which you need the MR/patch to be applied?
2. For the recent issue of logging, you can raise the issue in ESP-IDF.
Quick suggestions:
1. disable that log
3. In case want to retain the log, try with, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html#c.ESP_EARLY_LOGI

mantriyogesh, 2022-09-27:
For ESP-Hosted-NG, we are right now limited with IDF branch release/4.4, git commit, 6d85d53c.
If you have time, you can give me a patch over the same commit. Otherwise, I will have to port and then get you the ESP binary (might take a day or two , as I am on some other issue)

Also, I see from prior logs, you are using ESP32 with SDIO transport option. Considering ESP binary creation, would that transport be sufficient for you?

wildbiotiger, 2022-09-27:
@mantriyogesh, I had put that in my last post. Here’s the pull request:
https://github.com/espressif/esp-idf/pull/9842
And the commit
eec8e298d979cd743cff80ebc5495e26ef624406

wildbiotiger, 2022-09-27:
and yes, SDIO transport should be fine. I think the patch I created with the pull request uses v4.4. If it doesn’t just let me know.

wildbiotiger, 2022-09-29:
@mantriyogesh I've added a new pull request targeting just v4.4 here:

https://github.com/espressif/esp-idf/pull/9884

Let me know if you need anything else.

mantriyogesh, 2022-09-29:
Hello Nicholas,

Apologies for delay. I hope this would be useful for you.
[GH_161_29sep22.tgz](https://github.com/espressif/esp-hosted/files/9676000/GH_161_29sep22.tgz) is only ESP firmware with your code over `ESP-IDF release/v4.4 6d85d53c`
The supported commit at Host is [f1d32ab](https://github.com/espressif/esp-hosted/commit/f1d32abd183d6e4dd0cfb118b7a91faf4c1b674b)

Please note, 
1. This is not tested build. Just compiled.
2. You might face issues if you use different host code other than above commit
3. ESP side code is also going to be published in upcoming releases. At that point, you can even sync to master at both side, ESP & Host.

mantriyogesh, 2022-10-03:
Hello @wildbiotiger ,

Are you unblocked now?

wildbiotiger, 2022-10-03:
@mantriyogesh, haven't gotten a chance to test it yet. Work stuff cropping up.

mantriyogesh, 2022-10-03:
No problem. Thanks for the heads up!

wildbiotiger, 2022-10-05:
@mantriyogesh looks like it boots up with the new firmware and correctly detects the AT25 chip. I'm getting some weird errors with the driver insertion though. Seems a bit odd.

- Probing driver while mmc is bound results in a ETIMEOUT (-110)
- Unbinding and binding mmc when module is probed shows "network device detected" but no network interfaces appear and no other info is printed

So, it doesn't LOOK like it's bailing out anywhere, but the driver still isn't getting events or something. Any suggestions on how to proceed?

mantriyogesh, 2022-10-06:
Some suggestions:
1. Please check pull up requirements https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html
2. Please use minimal & good length & equal length jumpers (preferably use PCB like design)
3. Adding host side debug traces in SDIO source files to understand which register is getting success and which is timed out, will be much helpful, in case analyzer is not available
4. Proper logic analyzer dump for example, saleae capture will clear some dust.
5. Any log in dmesg / kern.log?

mantriyogesh, 2022-10-10:
Please do connect, if any updates/issues.

mantriyogesh, 2022-10-31:
Please do connect once you have any update.

blakecantwalk, 2022-11-09:
TSP file is too large to flash like to not have terabytes

blakecantwalk, 2022-11-09:
- - - [ ] well, she said that in the segment set a timer to be like 3547 minutes I've only only had two 123 wired yeah 123 gig flash drives has it you could do a virtual you can charge a virtual machine with like a shell@

mantriyogesh, 2022-11-10:
Hello @blakecantwalk 

We did not understand the comments and their relevancy to the original issue.
Can you please explain the comments a bit?

wildbiotiger, 2023-03-22:
@mantriyogesh FINALLY got around to testing this and noticed in the README that it's compiled for Wi-Fi and BT over SPI (not SDIO as my board is configured for...). Would you please reupload this with the correct configuration?

wildbiotiger, 2023-03-22:
OK, new issue, we are now CONNECTED via SDIO and the driver is recongnized. HOWEVER, the epsta0 does not appear, nor does any other device with wireless capabilities. Any ideas of where to go from here?
dmesg output, looks good?
```
[  938.442187] mmc2: card 0001 removed
[  941.549098] mmc2: queuing unknown CIS tuple 0x01 (3 bytes)
[  941.574322] mmc2: queuing unknown CIS tuple 0x1a (5 bytes)
[  941.592970] mmc2: queuing unknown CIS tuple 0x1b (8 bytes)
[  941.601716] mmc2: queuing unknown CIS tuple 0x80 (1 bytes)
[  941.608487] mmc2: queuing unknown CIS tuple 0x81 (1 bytes)
[  941.614080] mmc2: queuing unknown CIS tuple 0x82 (1 bytes)
[  941.622117] mmc2: queuing unknown CIS tuple 0x80 (1 bytes)
[  941.628835] mmc2: queuing unknown CIS tuple 0x81 (1 bytes)
[  941.634593] mmc2: queuing unknown CIS tuple 0x82 (1 bytes)
[  941.641801] mmc2: new SDIO card at address 0001
[  941.656877] esp_probe: ESP network device detected
[  941.668457] Rx Pre ====== 0
[  941.671276] Rx Pos ======  31
[  941.682656] Tx Pre ======  0
[  941.685562] Tx Pos ======  10
[  941.705330] esp_sdio: probe of mmc2:0001:2 failed with error -22
```

UART Output:
```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14848
load:0x40080400,len:3792
entry 0x40080694
I (27) boot: ESP-IDF v4.4.1-593-g6d85d53cee-dirty 2nd stage bootloader
I (27) boot: compile time 20:34:40
I (28) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 factory          factory app      00 00 00010000 00100000
I (99) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (106) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (114) boot: End of partition table
I (118) boot: Defaulting to factory image
I (123) boot_comm: chip revision: 3, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=15dc0h ( 89536) map
I (171) esp_image: segment 1: paddr=00025de8 vaddr=3ffbdb60 size=055c8h ( 21960) load
I (180) esp_image: segment 2: paddr=0002b3b8 vaddr=40080000 size=04c60h ( 19552) load
I (188) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=76b44h (486212) map
I (364) esp_image: segment 4: paddr=000a6b6c vaddr=40084c60 size=18384h ( 99204) load
I (405) esp_image: segment 5: paddr=000beef8 vaddr=50000000 size=00010h (    16) load
I (420) boot: Loaded app from partition at offset 0x10000
I (420) boot: Disabling RNG early entropy source...
I (432) cpu_start: Pro cpu up.
I (432) cpu_start: Starting app cpu, entry point is 0x4008121c
I (419) cpu_start: App cpu up.
I (448) cpu_start: Pro cpu start user code
I (448) cpu_start: cpu freq: 240000000
I (448) cpu_start: Application information:
I (453) cpu_start: Project name:     network_adapter
I (458) cpu_start: App version:      df0fe64
[... 131 lines cut ...]
        RX packets 1192  bytes 98211 (95.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1761  bytes 226148 (220.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

sit0: flags=128<NOARP>  mtu 1480
        sit  txqueuelen 1000  (IPv6-in-IPv4)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@interceptor:/home/debian#
```

So, if i can get espsta0 to appear, then I think we'd be golden, but I'm not sure what the failure is here.

mantriyogesh, 2023-03-23:
Hello @wildbiotiger 

1.0.1 is old release. Can you please use 1.0.2 release with  [release/ng-v1.0.2](https://github.com/espressif/esp-hosted/tree/release/ng-v1.0.2) as host code base?

Also just to understand, what point you have got sdio working?
Did it work with https://github.com/espressif/esp-hosted/issues/161#issuecomment-1262445841?
I am trying to understand what worked and what did not not (to get to correct line of treatment)

mantriyogesh, 2023-03-23:
Okay.
Let me be truthful, I'm very confused right now.
1. So original issue was flash size change (I hope nothing else). As you had no control, we got the binaies created with your expected change.
Is that you are testing?

2. That is mainly reason that you cannot test on 1.0.2. Because you might want to recreated firmware with those changes. Please confirm?

3. Anytime you were able to load and use espsta0? If yes, which binary and what was the transport used?

4.  Is this first time you are trying to use sdio (never tried or worked with ESP-Hosted sdio transport)?
If yes, check pull up requirement for your esp chipset from https://github.com/espressif/esp-hosted/issues/161#issuecomment-1270204699

wildbiotiger, 2023-03-23:
1. yes, that is what we are testing
2. i just tested with a fresh download of the ng_host master branch, and that results in a successful load of SDIO and no reset, but no driver appears
3. No, that's the part I am stuck on now
4. We have, the pull-ups are correct from what we can probe at the moment.

mantriyogesh, 2023-03-23:
1. Are you using jumper wires for connection? What would be the length?
2. So do you need the firmware changes for flash size with 1.0.2?
3. As I can see the probe was called. But there is some issue in esp log
I can see data path stopped and again started.
First bootup event was not received from esp to host.
Even though the probe was triggered, worth check if the pullups are working.
4. This start-> stop-> start data path happens always?
5. With current 1.0.1 sdio, you using corresponding host code base? Any local changes, please share patch with base commit number

mantriyogesh, 2023-03-23:
Anyway, I feel, worth to change your firmware with your expected code change on 1.0.2. Should I create and send that? Better to check on that.

wildbiotiger, 2023-03-23:
1. no, pcb impedance matched traces
2. yes please
3. yes, that appears to be the case, no bootup event detected
4. no, just on manual driver reload, if i give it the correct reset pin, no start stop occurs
5. using 1.0.2 now for testing, no code changes other than printk's for debugging.

mantriyogesh, 2023-03-23:
4. Resetpin is mandatory

 2) Okay

wildbiotiger, 2023-03-23:
Added a bunch of debug logs prints in case it helps:

key things:
- read_packet is returning no length for the rx work queue, results in halting of packet processing
- not sure what is being xmitted, but there appears to be reads coming back
- looks like something in the packet len register is not coming back formatted correctly?

```
[14234.375175] init: esp reset
[14234.595944] init_adapter: start
[14234.599162] init: done
[14234.601531] init: init transport layer
[14237.578528] mmc2: queuing unknown CIS tuple 0x01 (3 bytes)
[14237.603534] mmc2: queuing unknown CIS tuple 0x1a (5 bytes)
[14237.619366] mmc2: queuing unknown CIS tuple 0x1b (8 bytes)
[14237.633287] mmc2: queuing unknown CIS tuple 0x80 (1 bytes)
[14237.638999] mmc2: queuing unknown CIS tuple 0x81 (1 bytes)
[14237.644562] mmc2: queuing unknown CIS tuple 0x82 (1 bytes)
[14237.653032] mmc2: queuing unknown CIS tuple 0x80 (1 bytes)
[14237.659759] mmc2: queuing unknown CIS tuple 0x81 (1 bytes)
[14237.665348] mmc2: queuing unknown CIS tuple 0x82 (1 bytes)
[14237.672714] mmc2: new SDIO card at address 0001
[14237.685488] esp_probe: ESP network device detected
[14237.705604] esp_read_multi_byte:
[14237.705617] 1f
[14237.710618] esp_read_multi_byte:
[14237.712394] 00
[14237.715719] 00
[14237.717566] 80
[14237.719322] 00
[14237.721075]
[14237.724317] esp_process_interrupt
[14237.727661] esp_process_interrupt pkt!
[14237.731425] esp_process_new_packet_intr: queuing work
[14237.736562] esp_write_multi_byte:
[14237.736566] 00
[14237.739980] 00
[14237.741733] 80
[14237.743486] 00
[14237.745239]
[14237.753616] 00
[14237.753622] f0
[14237.755389] 01
[14237.758685]
[14237.761947] Rx Pre ====== 0
[14237.764748] Rx Pos ======  31
[14237.785336] esp_if_rx_work: work
[14237.789879] esp_read_multi_byte:
[14237.789888] 00
[14237.793226] 10
[14237.794979] 14
[14237.798418] esp_get_packets: check
[14237.803598] esp_get_packets: read pkt
[14237.816610] 00
[14237.816614]
[14237.819876] Tx Pre ======  0
[14237.822765] Tx Pos ======  10
[14237.825743] get_firmware_data: Read pkt done
[14237.833649] esp_read_multi_byte:
[14237.833658] 1f
[14237.838200] 00
[14237.839967] f0
[14237.841721] 01
[14237.843474]
[14237.859165] esp_get_len_from_slave: len: 0
[14237.863291] read_packet: no length 0 0
[14238.251786] getting adapter
[14238.254604] starting tx thread
[14238.259784] esp_write_byte: 01
[14238.274290] esp_sdio: probe of mmc2:0001:2 failed with error -22
```

mantriyogesh, 2023-03-24:
This was used with resetpin?

wildbiotiger, 2023-03-24:
Yes, and I checked that it does reset the device in the logs. 

Nicholas Smith

wildbiotiger, 2023-03-25:
Any update on this? Would like to test something soon.

mantriyogesh, 2023-04-03:
[2023.04.03_ESP32_NG_9884appliedOnIdf5.0.tgz](https://github.com/espressif/esp-hosted/files/11134338/2023.04.03_ESP32_NG_9884appliedOnIdf5.0.tgz)

We have not tested these IDF changes.
This includes ESP32 SDIO NG binaries,
(1) without SDIO timing change
(2) with SDIO timing change

Please check readme.txt for details

mantriyogesh, 2023-04-14:
Please remove per packet logs. It will badly affect timing.

With removed logs, please help to get both side, ESP and host logs.

wildbiotiger, 2023-04-17:
OK, I think that did it. Using the modified SDIO build, the interface initialized and I was able to connect to Wi-Fi and Bluetooth. Only strange thing was I was not able to ping 8.8.8.8 once I connected to my home network. Curl and other web utilities worked fine, though.

What was the fix you implemented on your end?

mantriyogesh, 2023-04-17:
Great that it's working for you.

One binary did not have timing adjustment, other had timing adjustment.
Basically, line as is  uncommented for NG.
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L201

This highly depends upon what timing host follows (& what is configured in device tree of your SoC)

wildbiotiger, 2023-04-17:
ok thanks, any reason the other sdio timing would not be working? Any recommendations on updating our device tree? We are using an omap-hsmmc instance if you need specifics.

mantriyogesh, 2023-04-18:
That's okay, once the timing is fixed, it should not affect.

Device tree might have options mentioned like neg-edge etc.

https://wiki.st.com/stm32mpu/wiki/SDMMC_device_tree_configuration

But keep in mind, one device tree doesn't apply all Linuxes. Every SoC has its own way of implementing and may have own params.

It is worth to mention, try to keep your device tree hot pluggable sdio. So param like,
'non-removable'  should be removed,
Param like 'broken-cd' should be added.

Repeating myself,  One size may not fit all!!
You might have to check what is supported with your SoC and use.

wildbiotiger, 2023-04-18:
OK, dug into the init a bit further on my side. Looks like the MMC bus driver will not detect the ESP32 unless the esp32_sdio module is loaded BEFORE the omap_hsmmc driver, which obviously does not happen because the omap_hsmmc driver is the parent driver of the esp32_sdio driver. 

In order to check this, I was able to simulate an event where, even when the MMC bus detects the card, the ESP bootup event is not received. Is there a way to force the ESP32 in the SDIO driver to reemit the bootup event or run a command to poll for the bootup information? 

If it was just a simple event with no data, I could hook it to a delay or something.

wildbiotiger, 2023-04-18:
As an additional test, I managed to do the following:

1. get the omap_hsmmc driver to recognize the card on boot
2. Manually probe the esp32_sdio driver with modprobe
3. esp32_sdio probe fails with error -110, even though I see the ESP32 resetting via UART
4. The -110 error appears BEFORE the ESP32 completely resets, so there may be something there.

mantriyogesh, 2023-04-19:
I was under impression that you git it working already here : https://github.com/espressif/esp-hosted/issues/161#issuecomment-1511583993

[7 more replies]