espressif/esp-hosted #162: Configure esp32 to take spidev1.0

lakshitave, 2022-09-26:
Hello,
I want to configure that esp32 should take spidev1.0 instead of spi. How could I include it in my recipe or there is any other method to do this.

mantriyogesh, 2022-09-26:
Hello @lakshitave ,

Thank you for your query.

In this case, bus number is 1 and chip select 0.
Please refer,
1. porting guide for ESP-Hosted-First-Generation : https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#124-peripheral-configurations and 
2. porting guide for ESP-Hosted-Next-Generation : https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md

For Yocto recipe, cross-referencing issue https://github.com/espressif/esp-hosted/issues/137

lakshitave, 2022-09-27:
Hello,
I am not getting proper logs as required. I am sharing it. Please have a look to it.

smarc-rzg2lc login: root
[   37.481643] audit: type=1006 audit(1600598671.232:2): pid=223 uid=0 old-auid=4294967295 auid=0 tty=(none) old-ses=4294967295 ses=1 res=1
root@smarc-rzg2lc:~# modprobe esp32_spi
[   61.702523] rpc-if-spi rpc-if-spi: chipselect 0 already in use
[   61.714222] Failed to add new SPI device
[   62.134610] ------------[ cut here ]------------
[   62.139479] kobject: '(null)' (00000000af576277): is not initialized, yet kobject_put() is being called.
[   62.154519] WARNING: CPU: 1 PID: 235 at lib/kobject.c:752 kobject_put+0xd8/0x118
[   62.161996] Modules linked in: esp32_spi(O+) vspm_if(O) vspm(O) mmngrbuf(O) mmngr(O)
[   62.169949] CPU: 1 PID: 235 Comm: modprobe Tainted: G        W  O      5.10.83-cip1-yocto-standard #1
[   62.179221] Hardware name: Renesas SMARC EVK based on r9a07g044c2 (DT)
[   62.185811] pstate: 40400005 (nZcv daif +PAN -UAO -TCO BTYPE=--)
[   62.191880] pc : kobject_put+0xd8/0x118
[   62.195772] lr : kobject_put+0xd8/0x118
[   62.199649] sp : ffff8000197dba70
[   62.203005] x29: ffff8000197dba70 x28: 0000000000000100 
[   62.208415] x27: 0000000000000002 x26: ffff8000197dbda0 
[   62.213815] x25: ffff00000f4eb680 x24: 0000000000000024 
[   62.219213] x23: 0000000000000000 x22: ffff800008a37000 
[   62.224611] x21: ffff00000aca1b00 x20: ffff800008a6c000 
[   62.230009] x19: ffff800008a6c520 x18: 0000000000000000 
[   62.235406] x17: 0000000000000000 x16: 0000000000000000 
[   62.240801] x15: 00002bf9500b0ce6 x14: 000000000000014a 
[   62.246196] x13: 000000000000014a x12: 0000000000000000 
[   62.251591] x11: 0000000000000005 x10: 0000000000000930 
[   62.256985] x9 : ffff8000197db7a0 x8 : ffff00000aca2490 
[   62.262381] x7 : 0000000000000000 x6 : ffff00003fdfc8a0 
[   62.267776] x5 : ffff00003fdfc8a0 x4 : 0000000000000000 
[   62.273168] x3 : ffff00003fe037b0 x2 : ffff00003fdfc8a0 
[   62.278563] x1 : 93f32aa990440900 x0 : 0000000000000000 
[   62.283957] Call trace:
[   62.286472]  kobject_put+0xd8/0x118
[   62.290027]  cdev_del+0x2c/0x40
[   62.293281]  esp_serial_cleanup+0x20/0x60 [esp32_spi]
[   62.298432]  spi_exit+0xa0/0x130 [esp32_spi]
[   62.302797]  esp_init_interface_layer+0x16c/0x198 [esp32_spi]
[   62.308638]  esp_init+0x150/0x1000 [esp32_spi]
[   62.313148]  do_one_initcall+0x74/0x1d0
[   62.317052]  do_init_module+0x54/0x218
[   62.320864]  load_module+0x1ebc/0x2408
[   62.324677]  __do_sys_finit_module+0xbc/0xf8
[   62.329010]  __arm64_sys_finit_module+0x20/0x30
[   62.333611]  el0_svc_common.constprop.4+0x8c/0x180
[   62.338463]  do_el0_svc+0x24/0x90
[   62.341845]  el0_svc+0x14/0x20
[   62.344966]  el0_sync_handler+0x90/0xb8
[   62.348857]  el0_sync+0x160/0x180
[   62.352222] ---[ end trace 9e0623897f56984e ]---
[   62.366380] ------------[ cut here ]------------
[   62.373391] refcount_t: underflow; use-after-free.
[   62.390505] WARNING: CPU: 0 PID: 235 at lib/refcount.c:28 refcount_warn_saturate+0xf8/0x148
[   62.398962] Modules linked in: esp32_spi(O+) vspm_if(O) vspm(O) mmngrbuf(O) mmngr(O)
[   62.406926] CPU: 0 PID: 235 Comm: modprobe Tainted: G        W  O      5.10.83-cip1-yocto-standard #1
[   62.416211] Hardware name: Renesas SMARC EVK based on r9a07g044c2 (DT)
[   62.422807] pstate: 40400005 (nZcv daif +PAN -UAO -TCO BTYPE=--)
[   62.428891] pc : refcount_warn_saturate+0xf8/0x148
[   62.433753] lr : refcount_warn_saturate+0xf8/0x148
[   62.438592] sp : ffff8000197dba60
[   62.441956] x29: ffff8000197dba60 x28: 0000000000000100 
[   62.447359] x27: 0000000000000002 x26: ffff8000197dbda0 
[   62.452760] x25: ffff00000f4eb680 x24: 0000000000000024 
[   62.458157] x23: 0000000000000000 x22: ffff800008a37000 
[   62.463554] x21: ffff00000aca1b00 x20: ffff800008a6c000 
[   62.468952] x19: ffff800008a6c520 x18: 0000000000000000 
[   62.474348] x17: 0000000000000000 x16: 0000000000000000 
[   62.479744] x15: 00008faf6495b3b8 x14: 000000000000006d 
[   62.485140] x13: 000000000000006d x12: 0000000000000000 
[... 107 lines cut ...]
[   63.286536]  esp_init+0x150/0x1000 [esp32_spi]
[   63.291051]  do_one_initcall+0x74/0x1d0
[   63.294958]  do_init_module+0x54/0x218
[   63.298771]  load_module+0x1ebc/0x2408
[   63.302584]  __do_sys_finit_module+0xbc/0xf8
[   63.306920]  __arm64_sys_finit_module+0x20/0x30
[   63.311519]  el0_svc_common.constprop.4+0x8c/0x180
[   63.316374]  do_el0_svc+0x24/0x90
[   63.319756]  el0_svc+0x14/0x20
[   63.322876]  el0_sync_handler+0x90/0xb8
[   63.326769]  el0_sync+0x160/0x180
[   63.330164] Code: a90153f3 aa0003f3 aa0103f4 f9400000 (f9400400) 
[   63.336325] ---[ end trace 9e0623897f569851 ]---

Broadcast message from systemd-journald@smarc-rzg2lc (Sun 2020-09-20 10:44:57 UTC):

kernel[182]: [   63.129817] Internal error: Oops: 96000004 [#1] PREEMPT SMP

Broadcast message from systemd-journald@smarc-rzg2lc (Sun 2020-09-20 10:44:57 UTC):

kernel[182]: [   63.330164] Code: a90153f3 aa0003f3 aa0103f4 f9400000 (f9400400)

Segmentation fault

Please suggest me a solution for this.

mantriyogesh, 2022-09-27:
As I can understand, your SPI bus is not yet claimable.
Have you disabled the spidev for your bus and chip select version?
/dev/spidev.X.Y 
X -> Bus number (1 in your case)
Y -> Chip select (0 in your case)

You might have to do changes in your device tree or SPI disabler to disable your interested spidev.
Once the spidev is disabled correctly, 
```sh
$ ls -lrt /dev/spidev<Bus_num>.<Chip_select>
```
will not list any device.

You can take a look at spi disabler for Raspberry Pi at https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/spidev_disabler.dts . This directly cannot be used on your non-RaspberryPi platform, but could be used as reference.

lakshitave, 2022-09-27:
Again after disabling the spi, I am getting the same logs.
Suggest solution for this.

mantriyogesh, 2022-09-27:
Can you please share the logs (no images of logs please) and your observations in detail?

We can certainly help. Actually getting the SPI device up on the platform is a prerequisite for ESP-Hosted project.

I would like to point you to one of the old discussions, https://github.com/espressif/esp-hosted/issues/101#issuecomment-1046513672 where, you might get some idea of the steps involved.

lakshitave, 2022-09-27:
Hi,
Do I need to add network_adapter (driver) in yocto?

mantriyogesh, 2022-09-27:
How are you using it right now?
Even if you are building and loading & system is able to insert it in, that should be fine.

Currently, as you are able to insert the kernel module and getting above crash, I see that is fine with respect to loading the module. The question is, Is the platform ready to get the module inserted?

In the above post, I sent you some steps in https://github.com/espressif/esp-hosted/issues/101#issuecomment-1046513672 . For example, Device tree, user space driver test, disable spidev and finally loading the kernel module.

Is the spidev getting disabled correctly? What is output of `ls /dev/spidev*` ? Is any other driver claiming the same SPI bus instance in the system (ideally not)?

lakshitave, 2022-09-27:
Here is my spidev output, when I disabled it from dts
root@smarc-rzg2lc:~# ls /dev/spidev*
ls: cannot access /dev/spidev*: No such file or directory

mantriyogesh, 2022-09-27:
1. Was this spi (bus with chip select) tested with spidev or any other user-space driver? This way we rule out any possible issue if the SPI is working or not.
2. Once above is tested & we load the driver, you still face issue of kernel crash? dmesg or /var/log/kern.log logs?

lakshitave, 2022-09-27:
I am using ESP32_Devkit_V4(ESP32-WROOM-32D) as a spi slave device.

For that I am using this yocto recipe

SUMMARY = "ESP-Hosted solution provides a way to use ESP board as a communication processor i.e. host for Wi-Fi and Bluetooth/BLE connectivity"
HOMEPAGE = "[https://github.com/espressif/esp-hosted/"](https://github.com/espressif/esp-hosted/%22)
LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENSE;md5=8264535c0c4e9c6c335635c4026a8022"

SRC_URI = "gitsm://github.com/espressif/esp-hosted.git;branch=master"

SRCREV = "release/v0.4"

S = "${WORKDIR}/git/host/linux/host_driver/esp32"

inherit module

EXTRA_OEMAKE_append_task-install = " -C ${STAGING_KERNEL_DIR} M=${S}"
EXTRA_OEMAKE += "KERNEL=${STAGING_KERNEL_DIR}"
EXTRA_OEMAKE += "target=spi"

RPROVIDES_${PN} += "kernel-module-esp-hosted"

and in local.conf I have appended

KERNEL_MODULE_PROBECONF += "esp32_spi"
module_conf_esp32_spi = "options esp32_spi resetpin=153"
KERNEL_MODULE_AUTOLOAD += "esp32_spi"

From dts tree spidev1.0 is disabled and I have also disabled bluetooth related functionality https://github.com/Ncerzzk/esp-hosted/commit/34405135844630febeed82a74927007dbef29a59

In Makefile I changed architecture arch=arm64 and toolchain path CROSS_COMPILE := /usr/bin/aarch64-linux-gnu-

Now I can successfully generate esp32_spi.ko file but after that my issue is known to you.I am not getting logs "ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]" like this

mantriyogesh, 2022-09-27:
Can you please check:
1. What is your kernel version exactly? 
2. How are you flashing the ESP32?
3. What is the exact binary or git commit that you are using (at ESP & host)?
4. In new terminal, 
```sh
$ tail -F /var/log/kern.log &
```
5. Before loading module, get loaded module list and check if esp32_spi is already loaded or not 
```sh
$ lsmod > lsmod.txt
$ echo "-----------" >> lsmod.txt
$ lsmod | grep -i esp >> lsmod.txt
```
6. insert module
```sh
$ modprobe esp32_spi
$ sleep 5
```
Get logs, in the new terminal and `lsmod.txt` and ESP side logs.

I understand loading module is first step, but once this is fixed, you might have to mimic the script and changes done as part of https://github.com/espressif/esp-hosted/blob/f1d32abd183d6e4dd0cfb118b7a91faf4c1b674b/esp_hosted_fg/host/linux/host_control/rpi_init.sh

lakshitave, 2022-09-28:
Hi,
I am getting the same logs as I am getting before. And also error from makefile.

Suggest solution for this.

lakshitave, 2022-09-28:
Hi, 
I am getting the same logs as I was getting before.
Please suggest solution for this.

mantriyogesh, 2022-09-29:
I had asked some logs in prior step. Can you please attach those logs?

lakshitave, 2022-09-29:
Hello,
These are the logs which I am getting.

 52.179032] Failed to obtain SPI master handle
[   52.599318] ------------[ cut here ]------------
[   52.604234] kobject: '(null)' (00000000b610cbba): is not initialized, yet kobject_put() is being called.
[   52.619225] WARNING: CPU: 0 PID: 225 at lib/kobject.c:752 kobject_put+0xd8/0x118
[   52.626702] Modules linked in: esp32_spi(O+) vspm_if(O) vspm(O) mmngrbuf(O) mmngr(O)
[   52.634648] CPU: 0 PID: 225 Comm: modprobe Tainted: G           O      5.10.83-cip1-yocto-standard #1
[   52.643921] Hardware name: Renesas SMARC EVK based on r9a07g044c2 (DT)
[   52.650509] pstate: 40400005 (nZcv daif +PAN -UAO -TCO BTYPE=--)
[   52.656574] pc : kobject_put+0xd8/0x118
[   52.660461] lr : kobject_put+0xd8/0x118
[   52.664335] sp : ffff8000199c3a70
[   52.667691] x29: ffff8000199c3a70 x28: 0000000000000100 
[   52.673085] x27: 0000000000000002 x26: ffff8000199c3da0 
[   52.678491] x25: ffff00000b232500 x24: 0000000000000024 
[   52.683889] x23: 0000000000000000 x22: ffff800008b7f000 
[   52.689287] x21: ffff00000f07d100 x20: ffff800008b7c000 
[   52.694685] x19: ffff800008b7c520 x18: 0000000000000000 
[   52.700081] x17: 0000000000000000 x16: 0000000000000000 
[   52.705476] x15: 00002cacdfda1550 x14: 0000000000000323 
[   52.710872] x13: 0000000000000323 x12: 0000000000000000 
[   52.716269] x11: 0000000000000004 x10: 0000000000000930 
[   52.721665] x9 : ffff8000199c37a0 x8 : ffff00000f07da90 
[   52.727059] x7 : 0000000000000000 x6 : ffff00003fde68a0 
[   52.732453] x5 : ffff00003fde68a0 x4 : 0000000000000000 
[   52.737847] x3 : ffff00003fded7b0 x2 : ffff00003fde68a0 
[   52.743242] x1 : f80bd7c4e76b5900 x0 : 0000000000000000 
[   52.748635] Call trace:
[   52.751148]  kobject_put+0xd8/0x118
[   52.754701]  cdev_del+0x2c/0x40
[   52.757954]  esp_serial_cleanup+0x20/0x60 [esp32_spi]
[   52.763108]  spi_exit+0xa0/0x140 [esp32_spi]
[   52.767475]  esp_init_interface_layer+0x16c/0x198 [esp32_spi]
[   52.773316]  esp_init+0x150/0x1000 [esp32_spi]
[   52.777826]  do_one_initcall+0x74/0x1d0
[   52.781728]  do_init_module+0x54/0x218
[   52.785540]  load_module+0x1ebc/0x2408
[   52.789354]  __do_sys_finit_module+0xbc/0xf8
[   52.793689]  __arm64_sys_finit_module+0x20/0x30
[   52.798289]  el0_svc_common.constprop.4+0x8c/0x180
[   52.803144]  do_el0_svc+0x24/0x90
[   52.806521]  el0_svc+0x14/0x20
[   52.809635]  el0_sync_handler+0x90/0xb8
[   52.813527]  el0_sync+0x160/0x180
[   52.816893] ---[ end trace 4bfe3e9411b287a3 ]---
[   52.831044] ------------[ cut here ]------------
[   52.843125] refcount_t: underflow; use-after-free.
[   52.855149] WARNING: CPU: 1 PID: 225 at lib/refcount.c:28 refcount_warn_saturate+0xf8/0x148
[   52.863596] Modules linked in: esp32_spi(O+) vspm_if(O) vspm(O) mmngrbuf(O) mmngr(O)
[   52.871562] CPU: 1 PID: 225 Comm: modprobe Tainted: G        W  O      5.10.83-cip1-yocto-standard #1
[   52.880848] Hardware name: Renesas SMARC EVK based on r9a07g044c2 (DT)
[   52.887444] pstate: 40400005 (nZcv daif +PAN -UAO -TCO BTYPE=--)
[   52.893519] pc : refcount_warn_saturate+0xf8/0x148
[   52.898372] lr : refcount_warn_saturate+0xf8/0x148
[   52.903211] sp : ffff8000199c3a60
[   52.906573] x29: ffff8000199c3a60 x28: 0000000000000100 
[   52.911975] x27: 0000000000000002 x26: ffff8000199c3da0 
[   52.917374] x25: ffff00000b232500 x24: 0000000000000024 
[   52.922771] x23: 0000000000000000 x22: ffff800008b7f000 
[   52.928169] x21: ffff00000f07d100 x20: ffff800008b7c000 
[   52.933567] x19: ffff800008b7c520 x18: 0000000000000000 
[   52.938963] x17: 0000000000000000 x16: 0000000000000000 
[   52.944361] x15: 0000a038641cf367 x14: 00000000000001e5 
[   52.949757] x13: 00000000000001e5 x12: 0000000000000000 
[   52.955152] x11: 0000000000000006 x10: 0000000000000930 
[   52.960548] x9 : ffff8000199c3790 x8 : ffff00000f07da90 
[   52.965943] x7 : 0000000000000000 x6 : ffff00003fdfd8a0 
[   52.971340] x5 : ffff00003fdfd8a0 x4 : 0000000000000000 
[... 100 lines cut ...]
[   53.736512]  esp_remove_card+0x4c/0x100 [esp32_spi]
[   53.741484]  spi_exit+0xa8/0x140 [esp32_spi]
[   53.745849]  esp_init_interface_layer+0x16c/0x198 [esp32_spi]
[   53.751690]  esp_init+0x150/0x1000 [esp32_spi]
[   53.756204]  do_one_initcall+0x74/0x1d0
[   53.760109]  do_init_module+0x54/0x218
[   53.763923]  load_module+0x1ebc/0x2408
[   53.767736]  __do_sys_finit_module+0xbc/0xf8
[   53.772071]  __arm64_sys_finit_module+0x20/0x30
[   53.776671]  el0_svc_common.constprop.4+0x8c/0x180
[   53.781527]  do_el0_svc+0x24/0x90
[   53.784905]  el0_svc+0x14/0x20
[   53.788019]  el0_sync_handler+0x90/0xb8
[   53.791910]  el0_sync+0x160/0x180
[   53.795308] Code: a90153f3 aa0003f3 aa0103f4 f9400000 (f9400400) 
[   53.801468] ---[ end trace 4bfe3e9411b287a6 ]---
Segmentation fault

As I am using renesas rzg2lc what should be these 3 parameters:
# Toolchain Path
CROSS_COMPILE := /usr/bin/arm-linux-gnueabihf-
# Linux Kernel header
KERNEL := /lib/modules/$(shell uname -r)/build
# Architecture
ARCH := arm64

mantriyogesh, 2022-09-29:
Okay, Is the module already loaded (at the boot time automatically) and you are trying to again load the module esp32_spi.ko on top?
See lsmod commands in prior post.

Your procedure looks okay. For example setting arch, cross-compile etc.

Kernel 5.10.83 should not have any problems as such.
If you are not reloading the module, only problem remains is 
`Failed to obtain SPI master handle`:

https://github.com/espressif/esp-hosted/blob/f1d32abd183d6e4dd0cfb118b7a91faf4c1b674b/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L358-L365

Have you changed the bus_number & chip select (as mentioned in porting guide)?

lakshitave, 2022-09-29:
After changing this, my logs are
root@smarc-rzg2lc:~# modprobe esp32_spi[   19.237129] ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]

now do i need to add spidev from dts?

mantriyogesh, 2022-09-29:
`spidev` is user-spave driver, which is when used attempts to claim spi bus. So we should disable spidev, which I think you have done already. Now esp32_spi.ko will attempt to use this spi bus. So we really should not try to use spidev again.

If you see https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/rpi_init.sh,

The spidev is disabled and then the insmod is called.
As you do not have any crash now, The next step is to get the peripheral connected and get the first event from ESP to Host as log in dmesg shows. Anyway, I suggest following the complete documentation of ESP-Hosted along with porting guide.

lakshitave, 2022-09-30:
I have added 4 spi pins in /esp-hosted/esp/esp_driver/network_adapter/main/spi_slave_api.c and connected ESP to host. But how it should tested in other linux platform?

mantriyogesh, 2022-09-30:
Do you intent to test
(a) ESP-Hosted
OR
(b) basic SPI using some other user space program?

lakshitave, 2022-09-30:
Hi,
I am testing ESP-Hosted. 
I am getting error after running the script:

esp_hosted_driver_read, Unable to read data before 30 seconds timeout 
transport_pserial_data_handler:154 Failed to read RX data
wifi_get_mac:144 Failed to process rx data
Failed to get softAP mac address
Failure in setting softAP config
ifconfig: SIOCSIFFLAGS: Cannot assign requested address
route: SIOCADDRT: Network is unreachable

Suggest solution for this.

lakshitave, 2022-10-03:
Hello,
Configuration of pins are:
MOSI-473  MISO-474  CLK-472  CS-475

Below I am attaching the image of setup:
[image]

mantriyogesh, 2022-10-03:
Thank you for the image.
1. I don't think with so just 4 wires, ESP-Hosted will work. I believe you have connected all the 8 wires for ESP32 from https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md#111-esp32-setup as per setup image
2. Also, please reduce the length of the wires to possibly 6-10cm. Higher length wires make the communication unreliable.
3. What is the git commit you have used at host and ESP32 ?
It should be exactly the same. For now, I suggest to use master at both places. You would have to flash the ESP binaries after setting up IDF (refer documentation) & build kernel module with same git commit.

lakshitave, 2022-10-07:
Hello,
I am gettings logs as follows when I connect esp32 with the Renesas board.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

Suggest solution for this.

lakshitave, 2022-10-07:
And whenever I don't connect my esp with renesas I am getting this logs from esp

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:12
load:0x3fff0028,len:7432
load:0x40078000,len:15100
ho 0 tail 12 room 4
load:0x40080400,len:3904
entry 0x400806b8
I (79) boot: Chip Revision: 1
I (79) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (43) boot: ESP-IDF v4.0.3-215-gbe7df8bce9-dirty 2nd stage bootloader
I (43) boot: compile time 17:19:23
I (44) boot: Enabling RNG early entropy source...
I (49) boot: SPI Speed      : 40MHz
I (53) boot: SPI Mode       : DIO
I (57) boot: SPI Flash Size : 4MB
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (87) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 factory          factory app      00 00 00010000 00100000
I (102) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (110) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (117) boot: End of partition table
I (122) boot: Defaulting to factory image
I (126) boot_comm: chip revision: 1, min. application chip revision: 0
I (134) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x202a8 (131752) map
I (190) esp_image: segment 1: paddr=0x000302d0 vaddr=0x3ffbdb60 size=0x03d40 ( 15680) load
I (196) esp_image: segment 2: paddr=0x00034018 vaddr=0x40080000 size=0x00400 (  1024) load
I (197) esp_image: segment 3: paddr=0x00034420 vaddr=0x40080400 size=0x0bbf0 ( 48112) load
I (226) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x8eab8 (584376) map
I (435) esp_image: segment 5: paddr=0x000cead8 vaddr=0x4008bff0 size=0x123b0 ( 74672) load
I (485) boot: Loaded app from partition at offset 0x10000
I (485) boot: Disabling RNG early entropy source...
I (485) cpu_start: Pro cpu up.
I (489) cpu_start: Application information:
I (494) cpu_start: Project name:     network_adapter
I (499) cpu_start: App version:      release0.4-60-g22180f7
I (505) cpu_start: Compile time:     Jan 24 2022 17:19:16
I (512) cpu_start: ELF file SHA256:  0f66794e8494e214...
I (517) cpu_start: ESP-IDF:          v4.0.3-215-gbe7df8bce9-dirty
I (524) cpu_start: Starting app cpu, entry point is 0x400812fc
I (0) cpu_start: App cpu up.
I (535) heap_init: Initializing. RAM available for dynamic allocation:
I (542) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (548) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (554) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (560) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (566) heap_init: At 3FFC80E0 len 00017F20 (95 KiB): DRAM
I (572) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (578) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (585) heap_init: At 4009E3A0 len 00001C60 (7 KiB): IRAM
I (591) cpu_start: Pro cpu start user code
I (609) spi_flash: detected chip: generic
I (610) spi_flash: flash io: dio
I (610) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (622) NETWORK_ADAPTER: *********************************************************************
I (628) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.4                        
I (636) NETWORK_ADAPTER:                 Transport used :: SPI                           
I (644) NETWORK_ADAPTER: *********************************************************************
I (650) NETWORK_ADAPTER: Supported features are:
I (654) NETWORK_ADAPTER: - WLAN over SPI
I (658) ESP_BT: - BT/BLE
I (662) ESP_BT:    - HCI Over SPI
I (664) ESP_BT:    - BT/BLE dual mode
I (676) BTDM_INIT: BT controller compile version [14ea243]
I (676) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (682) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1008) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1008) NETWORK_ADAPTER: ESP Bluetooth MAC addr: c4-4f-33-3e-de-33

I (1014) SPI_DRIVER: Using SPI interface
I (1016) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1024) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (2034) wifi:wifi driver task: 3ffd6964, prio:23, stack:6656, core=1
I (2034) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2038) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2046) wifi:wifi firmware version: e58bf82
I (2048) wifi:config NVS flash: disabled
I (2050) wifi:config nano formating: disabled
I (2054) wifi:Init data frame dynamic rx buffer num: 32
I (2058) wifi:Init management frame dynamic rx buffer num: 32
I (2062) wifi:Init management short buffer num: 32
I (2066) wifi:Init dynamic tx buffer num: 64
I (2070) wifi:Init static rx buffer size: 1600
I (2072) wifi:Init static rx buffer num: 16
I (2076) wifi:Init dynamic rx buffer num: 32
I (2080) phy_init: phy_version 4660,0162888,Dec 23 2020
I (2092) wifi:mode : null
I (2092) NETWORK_ADAPTER: Initial set up done

mantriyogesh, 2022-10-07:
Can you just disconnect resetpin GPIO, reset ESP, keeping all other pins connected? Do you see this behaviour?

In general, Please check if your GPIOs are in stale/wrong levels, specially reset pin. Ideally, while choosing any GPIO, please verify if it is unused n the sense that it is not being used GPIO from any other peripherals or device tree.

mantriyogesh, 2022-10-10:
@lakshitave any updates or issues?

mantriyogesh, 2022-10-20:
@lakshitave similar issues faced in https://github.com/espressif/esp-hosted/issues/165 and found out some solutions there. Worth to have a look?

mantriyogesh, 2022-10-31:
@lakshitave Is the issue fixed for you?
Can we please the issue, if you have no open questions?