espressif/esp-hosted #164: Wifi + BT/BLE connectivity for Raspberry Pi 1 B+

tech9492, 2022-10-06:
Hi, 

I have a spare RPI 1B+ and I would like to add Wifi+BT/BLE in order to use it for an IOT project (using mosquitto and node-red). My project would require Wifi and BLE to work simultaneously, Wifi to display a dashboad and BLE to subscribe to a hygrometer/thermometer. 

Hence before purchasing an ESP32-DevKitC and commit myself to this path, I would like to know if ESP-hosted will suit my application? Additionally, I read that ESP32 wifi and bluetooth shares one antenna and cannot be used simultaneously. 

I would love to get the contributors/community's opinion about this.

Thank you.

mantriyogesh, 2022-10-06:
As you had only asked BLE + Wi-Fi, I did not mention, but we also have tested Wi-Fi + A2DP (music streaming) Classic BT at the same time, without any stuttering of Bluetooth.

In this case, the Wi-Fi throughput is a little reduced from peak throughput. Basically, bandwidth and time is shared in between Wi-Fi & Bluetooth.

tech9492, 2022-10-06:
Hi @mantriyogesh 

Thank you for the quick response. Yes, it did clear my doubt about Wi-Fi & BT/BLE operating in parallel. 

Since you mentioned about Linux kernel, I just flashed raspberry pi os (32bit Bullseye) and everything seems to be fine on the RPi 1B+.
Hence, there should not be any issues. I guess only a physical test would be able to confirm that.

Thank you again for the explanation.

tech9492, 2022-10-16:
Hi @mantriyogesh 

I want to confirm some aspects of ESP hosted before I do anything irreversible. I got an ESP32-DevKitC-32D (module: WROOM-32D), this is my first time using a product from the Espressif ecosystem. 

This might sound stupid but I have a doubt about flashing firmware. So, if I need to roll back to factory firmware after having flashed ESP-Hosted NG firmware, should I just flash ESP32 AT binaries to start fresh?

Additionally, can I use jumper wires as a permanent setup? I will be using SPI+UART for its plug & play approach and I am going to prepare 5cm length jumper wires to maintain signal integrity.

Concerning BlueZ, RPi OS(32bit Bullseye) is already running version 5.55. Should I still follow this step [Ensure that bluez is installed on Raspberry Pi and it is downloaded in source format as well](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/bluetooth.md) ?

Lastly regarding power management, RPi and ESP32 should be powered separately? 

Thank you.

mantriyogesh, 2022-10-17:
1. 
> So, if I need to roll back to factory firmware after having flashed ESP-Hosted NG firmware, should I just flash ESP32 AT binaries to start fresh?

Yes, that should do. Any compatible ESP binary can be re-flashed to your board. Re-flashing would be the easiest option to change your flashed image.

In case you want to manually rollback without re-flashing (from standby ota partition), little tricky but do-able. Please check https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html#app-rollback.

2. 
> Additionally, can I use jumper wires as a permanent setup?

Definitely. Just make sure the wires are tightly fitting always. In case they are loose, you might think to replace them.
We suggest getting PCB designed as per your request, which will be better to maintain in long run. Easy alternative also could be to setup a breadboard

3. 
> I am going to prepare 5cm length jumper wires to maintain signal integrity

Great.

4. 
> Concerning BlueZ, RPi OS(32bit Bullseye) is already running version 5.55. Should I still follow this step [Ensure that bluez is installed on Raspberry Pi and it is downloaded in source format as well](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/bluetooth.md) ?

- It is no more needed that you should have BlueZ in source format. We will change the documentation accordingly.
You can just use the any last stable BlueZ in your Linux. 5.55 will also do.
- BT/BLE test procedure might be needed to get your Bluetooth working.
- Bluetooth should work out of the box for your Linux. In case of issues, refer easy `troubleshoot document` at https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Troubleshoot.md and https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Troubleshoot.md for basic issues.

5. 
> Lastly regarding power management, RPi and ESP32 should be powered separately?

Upto you. Make sure the Raspberry Pi is powered with correct power ratings adapter and cable. For the time being, you can draw USB to power ESP & flash it. In general, we power ESP externally from laptop. But there should not be any problems if you draw from RaspberryPi. I hope your application is not power hungry. For example, video streaming. If so, you can assess if SPI is working reliably (peripherals will need power from Raspberry, which will be split across peripherals). In my opinion, there should not be a problem.

tech9492, 2022-10-17:
Hi @mantriyogesh ,

Thank you very much for the explicit explanations.

My main concern about BlueZ was the source format part, since it is not required anymore that is cleared. I will definitely go through the test procedure for BT/BLE & Wi-Fi as per the documentation.

Thank you again. :)

tech9492, 2022-10-18:
Hi @mantriyogesh 

`$ git clone git@github.com:espressif/esp-hosted.git`

Is there any modification to the link of esp-hosted repo? When the above command is executed, it returns authenticity of host cannot be established with an ECDSA key fingerprint and prompt to continue with yes/no/[fingerprint]. How do I proceed?

Regarding the procedure for flashing the ESP32, should I erase before flashing the firmware? 

Lastly, based on  the flash download tool user guide, GPIO0 should be pulled low to enter download mode on ESP32. Does this apply to development board as well i.e ESP32-DevKitC-32D? If so, I should hold down Boot and then press EN just after plugging the usb cable?

Thank you.

mantriyogesh, 2022-10-19:
Thank you for letting us know, the issue you faced.

1.  
> Is there any modification to the link of esp-hosted repo? When the above command is executed, it returns authenticity of host cannot be established with an ECDSA key fingerprint and prompt to continue with yes/no/[fingerprint]. How do I proceed?

I do not face such problem yet. As such, there is no change from our side.
[image]
Can you please try to clean up your local & GitHub ssh keys and add again? See similar problems: https://github.com/ome/devspace/issues/38

2. 
> Regarding the procedure for flashing the ESP32, should I erase before flashing the firmware?

Upto you. (should not matter if you do or not), not needed as such.

3. 
> Lastly, based on the flash download tool user guide, GPIO0 should be pulled low to enter download mode on ESP32. Does this apply to development board as well i.e ESP32-DevKitC-32D? If so, I should hold down Boot and then press EN just after plugging the usb cable?

Try to flash the default way. If doesn't work, then you can. https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html

tech9492, 2022-10-19:
Hi @mantriyogesh 

Please find below a screenshot of terminal. I went through the attached link regarding similar problems but it did not solve the issue. I read that ssh is used to have read/write access to a repo, and not having write access might be causing this issue. Please do correct me if I am wrong. 

Instead can I use https(https://github.com/espressif/esp-hosted.git) to clone the repo? Will that impact any aspect of the cloning?

Thank you

[image]

mantriyogesh, 2022-10-19:
http is also just fine.

mantriyogesh, 2022-10-20:
Thank you for the logs.
1. What is the command you run to load the driver?
2. output of command `hciconfig`
3. Can you please check if [troubleshoot document](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Troubleshoot.md) helps?

tech9492, 2022-10-20:
Hi @mantriyogesh 

1. I followed the [Wi-Fi over SPI + Bluetooth over UART > Build and Run kernel module](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md) . Please find below a screenshot of the commands.
[image]

2. Below is a screenshot of the outputs related to the commands listed in the [troubleshoot document](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Troubleshoot.md) . It include the output for `hciconfig` as well.
[image]

3.  Additionally I have tried the following commands  for Wi-Fi `$ sudo iw dev espsta0 scan` & for Bluetooth `sudo buetoothctl scan on` . Below is a screenshot of the output.
[image]

I am surprised when I clicked on the network icon, it showed a list of Wi-Fi APs. I successfully connected to my Wi-Fi network and internet is working perfectly. However, I have no idea why it aborts the scan command in the terminal.

Since SPI is working properly, is it safe to assume that the bluetooth problem is the UART wiring? 

Right now I am using 5cm jumper cables for both protocol. Can I use different length for each protocol i.e 5cm for SPI and 8cm for UART?

Thank you.

mantriyogesh, 2022-10-21:
Again, thanks a lot for all the detailed logs!!

> However, I have no idea why it aborts the scan command in the terminal.

This is fixed on master with commit, https://github.com/espressif/esp-hosted/commit/009dd5aab86e68beaada8243fa81f66d62e8e19e.

WRT Bluetooth issue,
Can you please confirm if you tried to remove all Bluetooth related softwares and re-installing them as specified in Troubleshoot document?
Also, the user permissions set? In your case user is different, accordingly you might have to change..

tech9492, 2022-10-21:
Hi @mantriyogesh 

I forgot to mention in my previous post concerning two messages that appear at top right corner once the desktop loaded.

1. `Failed to resolve overlay 'disable-bt'` - I guess this is because RPi 1B+ does not have onboard Wi-Fi/BT.
2. `dterror: can't find symbol 'uart0_pins'` - Does this relate to the tx timeout messages?

I have also prepared another set of jumpers of length 8cm, which is my current setup. I carried out the whole troubleshooting procedure with both 5cm and 8cm. I can confirm its running BlueZ 5.55 and as stated previously Wi-Fi is working perfectly.

As for user permissions, I executed `$ sudo usermod -G bluetooth -a <username>` , it did not fix the issue.

> This is fixed on master with commit, https://github.com/espressif/esp-hosted/commit/009dd5aab86e68beaada8243fa81f66d62e8e19e. BTW, you can use just released fg-0.0.5 code base as well.

I do not mind trying, is the first gen more stable? Just to confirm, I have to start from scratch and follow the first gen(SPI+UART) guide?

Thank you.

tech9492, 2022-10-22:
Hi @mantriyogesh 

I just had a try using Wi-Fi & Bluetooth on SPI only setup. I can confirm both are working perfectly, no issues found on `dmesg` log. 

> What are the pros and cons of SPI+UART vs SPI setups?

Is there any performance gain using SPI+UART over SPI?

Thank you.

mantriyogesh, 2022-10-26:
Sorry, I am on vacation till Monday, so there are delays in the response.

Both Wi-Fi and Bluetooth supported simultaneously in both SPI only and SPI+UART.

In SPI only setup, both Wi-Fi and Bluetooth share the same SPI bandwidth.
You might get ~ 1Mbps drop in peak Wi-Fi throughput, only when Bluetooth also sends at same time. Anyway, it is also tested stable Wi-Fi and Bluetooth.

This case will also need lesser lines (no need of uart lines)..

FG & NG both should be equally stable. Both are under active development and active support. We suggest to use NG for Linux (considering seamless way of co-working with wpa_supplicant).

In case you want to control the Wi-Fi connection through user space application directly (using control part APIs e.g `connect_ap(ssid, password)`), FG would be in better position to integrate.

tech9492, 2022-10-29:
Hi @mantriyogesh 

> Sorry, I am on vacation till Monday, so there are delays in the response.

No worries, I will wait for your response once you are back.

Thank you for the clarification regarding SPI vs SPI+UART and FG vs NG.

> We suggest to use NG for Linux (considering seamless way of co-working with wpa_supplicant).

I am running NG with SPI only setup. I can confirm that it is plug & play for wpa_supplicant and hciconfig(bluetooth).

Just to update, I changed the structure of my build. I am now using RPi OS Lite (32bit Bullseye) with ESP-Hosted NG (SPI only). So I can't confirm if `Failed to resolve overlay 'disable-bt'` and `dterror: can't find symbol 'uart0_pins'` are still present due to the following:

1. I do not have a desktop (RPi OS Lite)
2. I did not perform the UART configuration, since I am using SPI only

I went through `dmesg` and syslog, I can't find the 2 errors. I guess it should be related to the UART configuration.

> In SPI only setup, both Wi-Fi and Bluetooth share the same SPI bandwidth.
You might get ~ 1Mbps drop in peak Wi-Fi throughput, only when Bluetooth also sends at same time. Anyway, it is also tested stable Wi-Fi and Bluetooth.

The way I intend to use it, I do not think ~ 1Mbps drop in peak Wi-Fi throughput will be noticeable.

mantriyogesh, 2022-10-31:
Okay. Thanks for the update.
Good to know that you are using SPI only with NG with success.

Do you have config.txt and cmdline.txt when you tried SPI+UART?
Also, can you please explain or point me why `uart0_pins` was needed?

We have never tested the solution on Raspberry Pi 1B+.
So this is very important for us to understand the issues you faced, if you at all wish to go back to SPI + UART setup.

tech9492, 2022-10-31:
Hi @mantriyogesh 

> Do you have config.txt and cmdline.txt when you tried SPI+UART?

No, I do not have the config.txt and cmdline.txt for that configuration. The only modifications I made, were the ones mentioned in the SPI+UART guide. For config.txt - enable and set clock frequency for SPI, enable UART. For cmdline.txt - removed `console=serial0,115200`.

> Also, can you please explain or point me why uart0_pins was needed?

I have no idea regarding the two errors. They appeared as pop-up messages at top right corner, just when the desktop was ready. Please find below a picture showing the errors.
 
[image]

I am assuming it hints at why bluetooth is not working since bluetooth is over UART. Have you tested the solution on Raspberry Pi Zero? It is similar to the Raspberry Pi 1B+. 

Additionally, I would like to load the ESP Hosted driver at/after boot. How do you suggest I should go about? I came across a bash script you proposed in [issue#34](https://github.com/espressif/esp-hosted/issues/34#issuecomment-879670025), is it straightforward to implement? 

Thank you.

mantriyogesh, 2022-11-01:
Actually, Bluetooth (HCI) over UART is pass through mode. Meaning, The Linux driver for Raspberry Pi is directly used and ESP-Hosted code is bypassed as such.

If I check https://pinout.xyz/pinout/uart the uart pins are not differentiated for Raspberry Pi 1B+ as such. 
So I assume, the uart pins on RPi (Tx, Rx, CTS, RTS) should work for uart. The pins used for UART0 are 10 rx, 8 tx, 36 cts, 11 rts.

When I check https://www.raspberrypi.com/documentation/computers/configuration.html?jr=on#configuring-uarts
[image]
I see the RPi 3 & 4 are using secondary as uart0 & they have `bluetooth` mention. This is **not** the case with RPi 1
Please also note the difference of primary and secondary uarts for RPi 1 & 3/4.

Coming ahead, 
https://www.raspberrypi.com/documentation/computers/configuration.html?jr=on#uarts-and-device-tree
explains the `disable-bt` and `miniuart-bt`.
Using these configs still worth a try. But what puzzles me more is why the 'Bluetooth' is not mentioned in RPi 1

Anyway, These are just data points I could gather but cannot test at my side (Don't have RPi 1 with me).

Coming to next question, Yes you can create the script and point it in startup/init & Kill scripts (rc scripts if I remember correct name). This script is minimal version of [rpi_init.sh](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/rpi_init.sh) from `FG` repo. Better to port from the latest code. Although **considering you are using NG solution**, you might not need character driver (lines containing /dev/esps0). You might only want to remove & insert driver with correct args.

tech9492, 2022-11-01:
Hi @mantriyogesh 

Thank you very much for the precise explanation regarding UART.

>I see the RPi 3 & 4 are using secondary as uart0 & they have bluetooth mention. This is not the case with RPi 1
Please also note the difference of primary and secondary uarts for RPi 1 & 3/4. 

>But what puzzles me more is why the 'Bluetooth' is not mentioned in RPi 1

Based on my understanding from the [RPi UART documentation](https://www.raspberrypi.com/documentation/computers/configuration.html?jr=on#configuring-uarts), (Bluetooth) is used to designate the UART configuration for boards having built-in Wi-Fi/BT chipset. Hence, why UART0 is secondary and connected to bluetooth. Mini UART is set to primary. Below is an extract concerning secondary UART taken from the above link.

> Secondary UART
The secondary UART is not normally present on the GPIO connector. By default, the secondary UART is connected to the Bluetooth side of the combined wireless LAN/Bluetooth controller, on models which contain this controller.  

> Anyway, These are just data points I could gather but cannot test at my side (Don't have RPi 1 with me).

I know RPi Zero is similar to RPi 1B+, same MPU just lacking a few I/O connectors.

I went through the uarts and device tree [link](https://www.raspberrypi.com/documentation/computers/configuration.html?jr=on#uarts-and-device-tree), I am confused as to what changes I should make in `config.txt` to test SPI+UART. Could you please point them out so I can test and provide you feedback.

Thank you

tech9492, 2022-11-03:
Hi @mantriyogesh 

> Coming to next question, Yes you can create the script and point it in startup/init & Kill scripts (rc scripts if I remember correct name). This script is minimal version of [rpi_init.sh](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/rpi_init.sh) from FG repo. Better to port from the latest code. Although considering you are using NG solution, you might not need character driver (lines containing /dev/esps0). You might only want to disable spidev (if not already done in device tree itself) and remove (if inserted) & insert driver with correct args.

By rc scripts, did you mean rc.local? 

Additionally, can I run it as a systemd service with `Type=idle` ?

To be honest, I do not really master coding/programming. I would be very grateful if you could help me out/provide a sample of a script that I can use for my scenario. 

I actually created a script with the following commands, that I am currently using to manually load the driver:
> #!/bin/bash
sudo killall wpa_supplicant
cd esp-hosted/esp_hosted_ng/host/
bash rpi_init.sh spi
exit

If I could just run this script automatically at some point after boot up, I would be satisfied.

Thank you.

mantriyogesh, 2022-11-03:
We have to disable serial which by default gets used for Linux console logs. For that you can use `sudo raspi-config` shown similar to https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable/enabling-serial-console

:warning: Note, Above link 'enables' the serial port. we have to select `No` in `Serial` menu for disabling the Serial console over uart.

But anyway, Needs to manually test and verify by using options like, `disable-bt` and/or `miniuart-bt`. This I cannot confirm as I do not have RPi 1B+. But may work in one or other trial & test.

For boot up script, rc.local (old method) also is fine. Any one method is fine from:
https://www.baeldung.com/linux/run-script-on-startup
These methods highly depend upon the distribution (softwares like systemd is supported or not etc).
Choose anyone and only one you from these methods.

tech9492, 2022-11-03:
Hi @mantriyogesh 

Thank you for the information regarding the configuration options for SPI+UART. I will try them out and update you regarding any issues faced.

For the boot up script, I managed to create a systemd service. Please find below the service file for loading ESP32 driver.

`ESP32Hosted.service`
> [Unit]
Description=ESP32 Hosted
After=multi-user.target
[Service]
ExecStart=/bin/bash  /home/logger/Start_ESP_Hosted.sh
[Install]
WantedBy=multi-user.target

`$ sudo systemctl status ESP32Hosted.service`
> ● ESP32Hosted.service - ESP32 Hosted
     Loaded: loaded (/etc/systemd/system/ESP32Hosted.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Fri 2022-11-04 00:50:57 +04; 3min 27s ago
    Process: 435 ExecStart=/bin/bash /home/logger/Start_ESP_Hosted.sh (code=exited, status=0/SUCCESS)
   Main PID: 435 (code=exited, status=0/SUCCESS)
        CPU: 2min 38.275s
>
>Nov 04 00:50:52 filamentpi sudo[910]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Nov 04 00:50:52 filamentpi sudo[910]: pam_unix(sudo:session): session closed for user root
Nov 04 00:50:52 filamentpi bash[448]: esp32 module inserted
Nov 04 00:50:57 filamentpi sudo[979]:     root : PWD=/home/logger/esp-hosted/esp_hosted_ng/host ; USER=root ; COMMAND=/usr/sbin/ifconfig espsta0 up
Nov 04 00:50:57 filamentpi sudo[979]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Nov 04 00:50:57 filamentpi sudo[979]: pam_unix(sudo:session): session closed for user root
Nov 04 00:50:57 filamentpi bash[448]: ESP32 host init successfully completed
Nov 04 00:50:57 filamentpi sudo[447]: pam_unix(sudo:session): session closed for user root
Nov 04 00:50:57 filamentpi systemd[1]: ESP32Hosted.service: Succeeded.
Nov 04 00:50:57 filamentpi systemd[1]: ESP32Hosted.service: Consumed 2min 38.275s CPU time.

I would appreciate any suggestions to improve the process. It is working but I am not too sure about the structuring of the service. Do you find any weakness or reliability issues that might appear in the future?

Do you know how I could create a target file i.e. `ESP32Hosted.target` ? Once `ESP32Hosted.service: Succeeded`, which I could then use to start a MQTT gateway service.

Thank you.

mantriyogesh, 2022-11-04:
I think this should be good.
I hope you already have 'enable'd the service.

Some reference: https://www.shubhamdipt.com/blog/how-to-create-a-systemd-service-in-linux/

You can test if the service loaded correctly or not after every reboot by checking status & doing `sudo lsmod | grep -i esp32`

tech9492, 2022-11-04:
Hi @mantriyogesh 

> I hope you already have 'enable'd the service.

Yes I did enable it. But I am not sure why it changes to `inactive(dead)` once it is done.
1. I have an `exit` at the end of my bash script `Start_ESP_Hosted.sh`. Should I remove it?
2. I did not put `Restart=always` in the unit section of `ESP32Hosted.service`. Should I add it?  

I ran `sudo lsmod | grep -i esp32`, please find the output below. Is there any module missing?
```
esp32_spi              40960  0
bluetooth             425984  25 esp32_spi,bnep
cfg80211              782336  1 esp32_spi
```

Additionally, do you have any idea about the fault below? It is intermittent when I run `dmesg`
```
[ 1380.181129] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[ 1380.181195] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[ 1380.181318] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[ 1380.181442] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[ 1380.181566] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[ 1380.181692] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[ 1380.181816] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[ 1380.181942] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[ 1380.182067] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[ 1380.182192] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
```

Thank you.

mantriyogesh, 2022-11-04:
> But I am not sure why it changes to inactive(dead) once it is done.

Script exits. But thats okay as lsmod shows module loaded.

> I have an exit at the end of my bash script Start_ESP_Hosted.sh. Should I remove it?

Doesn't matter both are same

> I did not put Restart=always in the unit section

I'm not to sure if this is mandatory. Anyway as your service is working, we can disregard this.

> Is there any module missing?

Output looks fine

> NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!

You can ignore this warning.

tech9492, 2022-11-12:
Hi @mantriyogesh 

I had a go at SPI+UART setup again, still facing the timeout issues.

1.  Using `raspi-config`, serial login shell is disabled but serial interface is enabled. 
2.  I disabled and mask `serial-getty@ttyAMA0.service`
3. I ran `ls -l /dev/ser*`, below is the output
```
lrwxrwxrwx 1 root root 7 Nov 12 14:01 /dev/serial0 -> ttyAMA0
```
ttyS0 is not present, in my case primary UART (ttyAMA0) is already running through GPIO 14 & 15. I tried `disable-bt`, `miniuart-bt` and also no dtoverlay in the config.txt. The timeout issue was still present under all 3 conditions.

4. Using `gpio readall`, GPIO 14 & 15 are both in ALT0 mode and GPIO 16 & 17 in ALT3.

Please let me know if i can perform any additional test.

Thank you

closed 2023-03-19 by tech9492