espressif/esp-hosted #165: Esp32 with other linux platform

meera1310, 2022-10-07:
(1) whenever I **modprobe esp32_spi** I am getting response **ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]**
(2) **ls /dev/spi***  response : **ls: cannot access /dev/spi*: No such file or directory**
(3) I am using **0.4** release both(ESP and Host) side
(4) Errors getting while connecting host with linux device: 
esp_hosted_driver_read, Unable to read data before 30 seconds timeout 
transport_pserial_data_handler:154 Failed to read RX data
wifi_get_mac:144 Failed to process rx data
Failed to get softAP mac address
Failure in setting softAP config

meera1310, 2022-10-07:
Here is my yocto recipe

SUMMARY = "ESP-Hosted solution provides a way to use ESP board as a communication processor i.e. host for Wi-Fi and Bluetooth/BLE connectivity"
HOMEPAGE = "https://github.com/espressif/esp-hosted/"
LICENSE = "Apache-2.0"

#LIC_FILES_CHKSUM = "file://LICENSE;md5=8264535c0c4e9c6c335635c4026a8022"

#SRC_URI = "gitsm://github.com/espressif/esp-hosted.git;branch=ESP-Hosted_MCU_Host"

LIC_FILES_CHKSUM = "file://${WORKDIR}/git/LICENSE;md5=7f43e699e0a26fae98c2938092f008d2"

SRC_URI = "gitsm://github.com/espressif/esp-hosted.git;protocol=https;branch=ESP-Hosted_MCU_Host"

SRCREV = "release/v0.4"

S = "${WORKDIR}/git/host/linux/host_driver/esp32"

do_configure_prepend() {
        sed -i 's/#define SPI_DATA_READY_PIN.*/#define SPI_DATA_READY_PIN 304/g' ${S}/spi/esp_spi.h
        sed -i 's/#define HANDSHAKE_PIN.*/#define HANDSHAKE_PIN 297/g' ${S}/spi/esp_spi.h
        sed -i 's/esp_board.bus_num = .*/esp_board.bus_num = 1;/g' ${S}/spi/esp_spi.c
        sed -i 's/esp_board.mode = SPI_MODE_2;/esp_board.mode = SPI_MODE_1;/g' ${S}/spi/esp_spi.c
        sed -i 's/ARCH := .*/ARCH := arm64/g' ${S}/Makefile
        sed -i 's/#define GPIO_MOSI.*/#define GPIO_MOSI 473/g' ${WORKDIR}/git/esp/esp_driver/network_adapter/main/spi_slave_api.c
        sed -i 's/#define GPIO_MISO.*/#define GPIO_MISO 474/g' ${WORKDIR}/git/esp/esp_driver/network_adapter/main/spi_slave_api.c
        sed -i 's/#define GPIO_SCLK.*/#define GPIO_SCLK 472/g' ${WORKDIR}/git/esp/esp_driver/network_adapter/main/spi_slave_api.c
        sed -i 's/#define GPIO_CS.*/#define GPIO_CS 475/g' ${WORKDIR}/git/esp/esp_driver/network_adapter/main/spi_slave_api.c
        sed -i 's/.mode=SPI_MODE_2,/.mode=SPI_MODE_1,/g' ${WORKDIR}/git/esp/esp_driver/network_adapter/main/spi_slave_api.c
}

inherit module

EXTRA_OEMAKE_append_task-install = " -C ${STAGING_KERNEL_DIR} M=${S}"
EXTRA_OEMAKE += "KERNEL=${STAGING_KERNEL_DIR}"
EXTRA_OEMAKE += "target=spi"

RPROVIDES_${PN} += "kernel-module-esp-hosted"

mantriyogesh, 2022-10-07:
A little bit long but detailed reply ahead:

1. Please do not use ESP-Hosted_MCU_Host branch. It is discontinued.
2. Please use master branch, but make sure you use the same git commit at ESP and Host.
For example, If ESP is using release/v0.4 release binaries, please use mapping commit which is mentioned on release.
[image]
3. Have you checked the SPI is working with spidev? What I mean, the underlying SPI device is working fine without even going for ESP-Hosted project. If underlying SPI is not working fine, ESP-Hosted simply would fail.
4. In your configuration, you have changed the SPI mode. I hope this change you have done after underlying testing of basic SPI in step (3) above. We suggest keeping the SPI mode exactly the same at both communicating entities. If you feel there is timing problem, then only you can try for mode mismatch. I know must not have done this, but warning that we should not copy/paste config or GPIOs working for someone else or some Linux platform to your setup. This will simply fail as every Linux is a different platform.
5. It is really worth to go through porting guide, which explains to you the configurations you might have to take care. https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md
6. Have you tested the GPIOs working ? We would be needing to use GPIOs which are unused in your system, otherwise the solution would fail. Some of the GPIOs (already covered in porting guide above) are resetpin, Data ready & handshake ping. All are mandatory and expected to be tested to be working.
7. Please use minicom or similar software after flashing release binaries to collect the ESP side logs.
For host, dmesg or /var/log/kern.log should capture the logs.
8. GPIO expectation from configured GPIOs:
- 8.1 SPI pins : tested in step (3) or real SPI example like [ESP-IDF spi slave example](https://github.com/espressif/esp-idf/tree/release/v4.0/examples/peripherals/spi_slave), where TX and RX both could be tested. In case you are using receiver, sender should be used as your Linux SPI and vice versa.
- SPI config like bus number and chip select could be changed in https://github.com/espressif/esp-hosted/blob/da792ce76501831a007202d719f4bd95a95a3e0d/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L356-L359
- 8.2 Resetpin: on every loading of driver, ESP should be seen to be resetting (minicom or idf.py monitor log will show loading of firmware from start) Also have a look at https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L17, where you can specify resetpin.
- 8.3 Data read and handshake pins: The host side code should be matching with your GPIOs. Handshake & Data Ready pins configured at https://github.com/espressif/esp-hosted/blob/da792ce76501831a007202d719f4bd95a95a3e0d/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.h#L22-L26 & could be tested when you add manual printk logs into these two functions, https://github.com/espressif/esp-hosted/blob/da792ce76501831a007202d719f4bd95a95a3e0d/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L75-L91
- 8.4 rpi_init.sh script needs to be ported as per your platform. This script was written keeping in mind Raspberry Pi as platform. It should be easy task, but you have to assess each line, if it applied to your Linux platform. If differ, change accordingly.
9. If you follow these steps, you should get working ESP-Hosted. The problem is ESP-Hosted expects that underlying platform (Like SPI, GPIOs, their modes, configs etc)working as we would have no control of the platform and will greatly vary for each Linux variant.

Please keep in mind that the hyperlinks mentioned here are wrt current master (which is ever changing). The code part highlighted to be changed as per your corresponding git commit.

mantriyogesh, 2022-10-07:
Once you get first event from ESP to Host (please check reference logs in documentation),
At Host side, Please -
1. Please note, the AP (to which you want to connect to ) is radiating & in vicinity of ESP.
2. C demo app: 
```sh
$ cd esp_hosted_fg/host/linux/host_control/c_support/
```
Change the SSID, password in `ctrl_config.h` with `STATION_MODE_SSID` & `STATION_MODE_PWD`.
3. `make clean && make` & make sure that AP is broadcasting within range.
4. sudo ./test.out for all options, select expected as argument and run.

meera1310, 2022-10-07:
sorry @mantriyogesh I am confused from your answer. let's go step by step
(1) after modprobe esp32_spi , ls /dev/spidev1.0 should be exposed or not?
if i enable spidev1.0 from dts there were so many errors in modprobe

mantriyogesh, 2022-10-07:
Always a log would be much helpful, unfortunately, just the observation that it is not working doesn't give much clue to us.

What it could mean that your base SPI is not working (prerequisite for ESP-Hosted) or it is not correctly configured in Device Tree (or something else?)

You are trying to jump directly to the solution, which will fail unless the underlying SPI bus is not working.
Please follow the steps till step (3) first from earlier post. Test your solution till underlying SPI is working. Till this point, it doesn't make sense to even disable the spidev. As it would mean, you are assuming that SPI is just working.

Why it may not work? One of following:
a. Your SPI pins are not correctly configured in device tree
b. SPI bus is already been used by something or somebody else
c. GPIO pins used for SPI are being used for some other peripherals.

Please read how to test spidev. spidev is just usespace tool which acquires the spi bus and performs some tests on it.
If spidev or for that matter, any user space application/library, which will confirm underlying spi is working, then we will go disabling spidev (through device tree generally).

Why we need to disable spidev? 
At a time only one can entity can grab the control of SPI bus. Generally, in Linux, spidev is always present. Unless we disable, no one can claim the bus.

How to confirm if spidev is disabled?
By default, /dev/spidevX.Y is created. X is SPI bus number and Y being chip select. Befre disbling spidev for X.Y, ls -lrt /dev/spi* will list /dev/spidevX.Y . When you disable spidev for X SPI bus instance and Y chip select, /dev/spidevX.Y only will be removed. Please note, there might be many /dev/spidev*.* But we will need only one X bus instance and Y chip select out of it. You would be knowing which bus you are using and chip select you are using, as you would be doing device tree config to activate that bus.

Long story short.
1. Device tree should be configured to get the /dev/spidevX.Y listed 
2. Test with spidev if the SPI bus is working or not
3. Disable the spidev with expected X bus and Y chip select. Refer example disabler in https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/spidev_disabler.dts. Please note, This kind of Device tree config highly depends upon the Linux Soc you use.
4. ls -lrt /dev/spidevX.Y should not be listed now.

Please refer to earlier post and try to follow it. Unfortunately, getting your Linux flavour's SPI up is something you only will have to do. And there are steps defined, but no shortcuts for that. It is a prerequisite for ESP-Hosted.

Similar to issue, recently, https://github.com/espressif/esp-hosted/issues/162 has got the spidev working.

mantriyogesh, 2022-10-07:
Please try above & do connect for any issues. We will definitely look into the problems.

meera1310, 2022-10-07:
Hiii i have enabled spidev from dts and spi tested from user space utility and it works perfectly so my spidev1.0 is working fine. With that I have tested all the gpios Handshake,Data ready, reset pins seperately from user space and that are working absolutely fine. But when I disable spidev from dts then i m getting response ESP bus[1] cs[0] and spi clock[10]. But I can not establish connection between ESP32 and renesas.

mantriyogesh, 2022-10-07:
1. Have you changed code 8.1 to 8.4?
3. Are you flashing ESP manually or using binaries?
4. ESP side code change in your earlier post should be reverted back. In fact, revert All code change and use the same git commit at both sides, without any changes (except porting changes stated above in point 8 spi bus, chip select numbers, handshake & data ready pins, reset pin etc) specially, Revert the  spi mode to 2 in host?
6. Also can you please send complete logs in esp & host?

Easiest to understand these issues if you have logic analyzer to test the signals on lines, if you have one.

meera1310, 2022-10-08:
(1) i have loaded code in esp32 with this command  used these binaries.
 python3 esptool.py<http://esptool.py> --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash --flash_mode qio --flash_size detect --flash_freq 40m 0x1000 esp_hosted_bootloader_esp32_spi_v0.4.bin 0x8000 esp_hosted_partition-table_esp32_spi_v0.4.bin 0xd000 esp_hosted_ota_data_initial_esp32_spi_v0.4.bin 0x10000 esp_hosted_firmware_esp32_spi_v0.4.bin
(2) I don't have any idea about 8.1 and 8.4
(3) Host side i have made yocto recipe which I have described earlier
(4) Do you think so it is version mismatch between host and esp???

  2.  Are you flashing ESP manually or using binaries?
  3.  ESP side code change in your earlier post should be reverted back. In fact, revert All code change and use the same git commit at both sides, without any changes (except porting changes stated above in point 8 spi bus, chip select numbers, handshake & data ready pins, reset pin etc) specially, Revert the spi mode to 2 in host?
  4.  Also can you please send complete logs in esp & host?

mantriyogesh, 2022-10-08:
In fact this comment is formed for your scenarios.
1. Still I demand the logs. Please check previous comment(s) carefully.
2. Please refer https://github.com/espressif/esp-hosted/issues/165#issuecomment-1271319848 for 8.1-8.4. Please check the earlier comments.
3. Yocto / non yocto scenario would not differ wrt ESP Hosted. Repeating again, getting your platform up is not in the scope of ESP-Hosted. We can only guide just to help you out. Getting Linux with peripherals ready, is prerequisite for ESP-Hosted. We can't verify your dts. Whatever/wherever incorrect, we havecoveredin prior comments.
4. You are in better position to make sure the commits match. We cannot help unless clear logs at both side are captured, at the least required.

I suggest you to go through earlier comments and get us answer for the open questions to help in better way.

meera1310, 2022-10-10:
Host side logs and ESP32 logs attached with this
[esp32 logs.txt](https://github.com/espressif/esp-hosted/files/9743363/esp32.logs.txt)
[host_logs.txt](https://github.com/espressif/esp-hosted/files/9743366/host_logs.txt)

meera1310, 2022-10-10:
Although I have changed my recipe
SUMMARY = "ESP-Hosted solution provides a way to use ESP board as a communication processor i.e. host for Wi-Fi and Bluetooth/BLE connectivity"
HOMEPAGE = "https://github.com/espressif/esp-hosted/"
LICENSE = "Apache-2.0"

LIC_FILES_CHKSUM = "file://${WORKDIR}/git/LICENSE;md5=7f43e699e0a26fae98c2938092f008d2"

SRC_URI = "gitsm://github.com/espressif/esp-hosted.git;protocol=https;branch=master"

SRCREV = "release/v0.4"

S = "${WORKDIR}/git/host/linux/host_driver/esp32"

do_configure_prepend() {
        sed -i 's/#define SPI_DATA_READY_PIN.*/#define SPI_DATA_READY_PIN 304/g' ${S}/spi/esp_spi.h
        sed -i 's/#define HANDSHAKE_PIN.*/#define HANDSHAKE_PIN 297/g' ${S}/spi/esp_spi.h
        sed -i 's/esp_board.bus_num = .*/esp_board.bus_num = 1;/g' ${S}/spi/esp_spi.c
        sed -i 's/ARCH := .*/ARCH := arm64/g' ${S}/Makefile
}

inherit module

EXTRA_OEMAKE_append_task-install = " -C ${STAGING_KERNEL_DIR} M=${S}"
EXTRA_OEMAKE += "KERNEL=${STAGING_KERNEL_DIR}"
EXTRA_OEMAKE += "target=spi"

RPROVIDES_${PN} += "kernel-module-esp-hosted"

mantriyogesh, 2022-10-10:
Okay, now the first step is to get the first event from ESP to Host, which is not observed on host log.
The possible reason could be :
1. SPI bus is not working (config issue / timing issue)
2. Data ready or Handshake is not working at GPIO level
3. If you are using jumper wires, Wires are long or not correctly connected. We suggest shortening the wires, of around (6 cm) and of equal length.

Is it possible for you to test ESP-Hosted-FG with master?
The reason being, If it doesn't work on master, I will try to give you some debug patch at both sides, which should be easy to create & test on master.

mantriyogesh, 2022-10-10:
Also, If you could wait for a couple of days, we are also going to release next release for ESP-Hosted-FG

meera1310, 2022-10-10:
Can you please tell me how to add **dtoverlay**  in yocto. I have already added dtc in my yocto

mantriyogesh, 2022-10-10:
dtoverlay may not be applicable for yocto. You might have to handle it with your device tree.

mantriyogesh, 2022-10-10:
This is Raspberry Pi specific script. You might need to port it as per your Linux distro.
Major shell specific things should be similar. But dts/overlay for spidev disabling, you might have to add in the device tree itself as per your SoC (I believe you have done it already?)

mantriyogesh, 2022-10-11:
As I see, you are getting trouble getting the first event.

Can you please try the ESP-IDF example of SPI sender & receiver with your Linux (Please look that configs of SPI match. For example, SPI mode)?
If TX & RX works fine in this IDF example, do change the config at both sides, to match the SPI config that of ESP-Hosted.

Once you get this working, we will have confidence that the underlying SPI (4 lines of SPI+GND) are working fine.
You might have to use some user-space application/lib just test through /dev/spidev1.0 for this. (You might have to comment spidev disabler changes in DT for this IDF example testing)

meera1310, 2022-10-11:
can you please send me link of that example?

mantriyogesh, 2022-10-11:
https://github.com/espressif/esp-idf/tree/release/v4.0/examples/peripherals/spi_slave

meera1310, 2022-10-11:
sender code should be flashed in renesas and receiver code should be flashed in esp-32?

mantriyogesh, 2022-10-11:
Both codes might be needed to be tried (one after the other working).
These are ESP codes only.
You might have to find a user space application which would work on your Linux to close match these ESP side codes.
These code directly would not work on Linux (they are for ESP)

meera1310, 2022-10-11:
Hii now dtoverlay successfully compiled in my image so it is successfully disable spidev1.0 from ls /dev/. But there are some errors attached in log file. Please help me with this 
[spi_logs.txt](https://github.com/espressif/esp-hosted/files/9754968/spi_logs.txt)

mantriyogesh, 2022-10-11:
`Failed to obtain SPI master handle` means that the kernel driver is not able to claim the SPI Bus.
Are the code changes in 8.1 in prior post done?

meera1310, 2022-10-11:
Do i need to add in recipe according to renesas gpio?
sed -i 's/#define GPIO_MOSI./#define GPIO_MOSI 473/g' ${WORKDIR}/git/esp/esp_driver/network_adapter/main/spi_slave_api.c
sed -i 's/#define GPIO_MISO./#define GPIO_MISO 474/g' ${WORKDIR}/git/esp/esp_driver/network_adapter/main/spi_slave_api.c
sed -i 's/#define GPIO_SCLK./#define GPIO_SCLK 472/g' ${WORKDIR}/git/esp/esp_driver/network_adapter/main/spi_slave_api.c
sed -i 's/#define GPIO_CS./#define GPIO_CS 475/g' ${WORKDIR}/git/esp/esp_driver/network_adapter/main/spi_slave_api.c

meera1310, 2022-10-11:
Here is the log of my spidev1.0 working. Do i need to set spi mode 0 both the side?
root@smarc-rzg2lc:~/spidev-test# ./spidev-aarch64 
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 KHz)
RX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....�..................�.

mantriyogesh, 2022-10-11:
> Do i need to add in recipe according to renesas gpio? sed -i 's/#define GPIO_MOSI./#define GPIO_MOSI 473/g' ${WORKDIR}/git/esp/esp_driver/network_adapter/main/spi_slave_api.c sed -i 's/#define [...]

I don't understand this. What is motive of change?

mantriyogesh, 2022-10-11:
Please try to understand esp code will affect spi peripheral at esp side.
Please help me to understand why this change is expected for you?

mantriyogesh, 2022-10-11:
> yes strlcpy(esp_board.modalias, "esp_spi", sizeof(esp_board.modalias)); esp_board.mode = SPI_MODE_2; esp_board.max_speed_hz = spi_clk_mhz * NUMBER_1M; esp_board.bus_num = 1; esp_board.chip_select = 0;

What is output of  `ls -lrt /dev/spidev1.0` before modprobe?

mantriyogesh, 2022-10-11:
> As I see, you are getting trouble getting the first event.
>
> Can you please try the ESP-IDF example of SPI sender & receiver with your Linux (Please look that configs of SPI match. For example, SPI [...]

Have you tested this?

meera1310, 2022-10-12:
Here are my complete logs. And **Failed to obtain SPI master handle** means there is some problem with bus number.
[soc_logs.txt](https://github.com/espressif/esp-hosted/files/9761690/soc_logs.txt)

mantriyogesh, 2022-10-12:
That looks something not right.
Is it possible to post your Device Tree changes?

meera1310, 2022-10-12:
(1)**In rzg2lc-smarc.dtsi patch applied for enabling SPI** 

iff --git a/rzg2lc-smarc.dtsi b/rzg2lc-smarc.dtsi.org
index c06cabb7a734..e8ad7ce2395f 100644
--- a/arch/arm64/boot/dts/renesas/rzg2lc-smarc.dtsi
+++ b/arch/arm64/boot/dts/renesas/rzg2lc-smarc.dtsi
@@ -21,7 +21,7 @@
 #define SW_SD0_DEV_SEL 1
 #define SW_I2S0_I2S1   1
 
-#define SW_SCIF_CAN    0
+#define SW_SCIF_CAN    1
 #if (SW_SCIF_CAN)
 /* Due to HW routing, SW_RSPI_CAN is always 0 when SW_SCIF_CAN is set to 1 */
 #define SW_RSPI_CAN    0

(2) **in r9a07g044.dtsi spidev enabled patch**
--- a/arch/arm64/boot/dts/renesas/r9a07g044.dtsi
+++ b/arch/arm64/boot/dts/renesas/r9a07g044.dtsi
@@ -366,7 +366,14 @@ spi1: spi@1004b000 {
                        num-cs = <1>;
                        #address-cells = <1>;
                        #size-cells = <0>;
-                       status = "disabled";
+                       status = "okay";
+                        spidev@1 {
+                               compatible = "spidev";
+                               spi-max-frequency = <10000000>;
+                               reg = <0>;
+                        };
+
+
                };
(3) my spidev_disabler.dts attached
[r9a07g044.dtsi.txt](https://github.com/espressif/esp-hosted/files/9762197/r9a07g044.dtsi.txt)
[rzg2lc-smarc.dtsi.txt](https://github.com/espressif/esp-hosted/files/9762198/rzg2lc-smarc.dtsi.txt)
[spidev_disabler.dts.txt](https://github.com/espressif/esp-hosted/files/9762209/spidev_disabler.dts.txt)

mantriyogesh, 2022-10-12:
1.  I suspect you are disabling spi controller itself instead of disabling spidev.
One reference DT I can show here,
[image]
Here spi bus instance is recorded and only spidev is disabled.

You can remove 
```
        target-path = "/soc/spi@1004b000";
```
from spidev_disabler.dts

and make the overall DT changes such that the spidev reference is used & used overlay status as disabled.
Check in [RaspberryPi-SPIDisabler](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/spidev_disabler.dts) where,
```
        target = <&spidev0>;
```
2. In one of previous communication, 
> Here is the log of my spidev1.0 working. Do i need to set spi mode 0 both the side?
root@smarc-rzg2lc:~/spidev-test# ./spidev-aarch64
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 KHz)
RX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D | ......@....�..................�.

There seems something off. Please check https://wiki.st.com/stm32mpu/wiki/How_to_use_SPI_from_Linux_userland_with_spidev as reference for spidev testing.

meera1310, 2022-10-12:
Hii now I can successfully modprobe my device

OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@1004b000/spidev@1/status
[   75.356436] esp_reset, ESP32: Resetpin of Host is 464
[   75.356578] esp_reset, ESP32: Triggering ESP reset.
[   75.359295] ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]

But I am not getting any logs from esp32. Whenever I connect esp32 with renesas I am getting logs from esp32 in continuous mode
**rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57**

mantriyogesh, 2022-10-12:
Check https://github.com/espressif/esp-hosted/issues/162#issuecomment-1271331954

meera1310, 2022-10-12:
when removing EN pin from esp32 I am getting logs like this whenever I **press EN button**.
rst:0x1 (**POWERON_RESET**),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

mantriyogesh, 2022-10-12:
Might be your Linux has some stale signals.
Can you flash ESP first and then switch_on/connect to the host?

mantriyogesh, 2022-10-12:
Please check https://www.esp32.com/viewtopic.php?t=3452
easiest suggestion to flash the software and then connect / start the host.

meera1310, 2022-10-13:
whenever I don't connect esp32 with renesas I get reset signal from renesas (**MicrosoftTeams-image.png**) on CH4. As soon as, I connect esp32 with renesas I don't get reset signal measured on renesas
[image]

mantriyogesh, 2022-10-14:
This looks strange. Can you measure on ESP chipset resetpin (default signal)?
ESP side resetpin is expected to be input, actually.
Irrespective of ESP connected or not, Linux should get the signal out on Linux side resetpin.

mantriyogesh, 2022-10-17:
Did you get it working @meera1310 ?

mantriyogesh, 2022-10-18:
@meera1310 ?

meera1310, 2022-10-18:
No.

meera1310, 2022-10-19:
I am not getting this logs, except this all other things are working now. My device can reset after modprobe
 **Received INIT event from esp32
 ESP32 capabilities: 0x78**

mantriyogesh, 2022-10-19:
Okay. This is an important step.
I will try to give you a debug patch & ESP binaries. Would that be fine for you to test those?

meera1310, 2022-10-19:
sure

mantriyogesh, 2022-10-19:
Okay, Will give you the debug package possibly by tomorrow or max by Friday.

meera1310, 2022-10-19:
Now I can receive these 2 logs also but I am not getting IP of the device. You can check from my screenshots
(1) **renesas side firmware version** : [image]
(2)**ESP side firmware version**:
[image]
(3)**INIT logs getting from esp host default frequency 1 mhz**
[image]
(4) **[image]**:

mantriyogesh, 2022-10-19:
This is Great. The first event is received.
Can you please send esp complete log?

Any change you had done? This would help me understand the environment.

meera1310, 2022-10-19:
Hiii, Now it works everything thank you so much for your constant support. I changed frequency 1 mhz and commented **adjust_spi_clock(*(pos + 2));** from esp_spi.c and it works

mantriyogesh, 2023-01-12:
@meera1310 Can you please help close the issue?

closed 2023-01-17 by meera1310