espressif/esp-hosted #166: Hosted-NG: How do I upgrade the firmware when I only use the SDIO interface?

dukelec, 2022-10-09:
I am designing a pcb for a product and I have the esp32 serial port on the pcb for the first time to burn Hosted-NG firmware.
Since only the sdio port is connected to the cpu running linux, how can I upgrade the esp32 firmware via sdio?

The cpu connected to esp32 via sdio has only 32 MBytes or 64 MBytes of memory, and the storage space is too small to use python scripts.
For the first burn I use a linux pc and there is no limitation to use python.

Also, is it possible to add a "wifi single carrier tx out" feature to the Hosted-NG firmware to facilitate factory and maintenance testing of the product.
For example, poor soldering of the capacitor to the esp crystal can cause the frequency to deviate from the allowed range (between triangle 1 and 2). The picture shows the ESP8089 linux driver enabling "wifi single carrier tx out":
[image]

For complete RF testing I can use a separate firmware, but the "wifi single carrier tx out" tests for frequency deviation and whether the device RF works, I hope you add this to the Hosted-NG firmware, otherwise the product will be a lot of trouble to produce and repair.
Thanks.

mantriyogesh, 2022-10-09:
Hello @dukelec ,

Thank you for all your detailed questions & all the information.
As I understand, there are two questions.
A) OTA support in NG
B) RF beacon test

<ins>[Part A]</ins>
Actually, the OTA (over the air) firmware update is supported in ESP-Hosted-**FG**. We have plan to propagate these changes into **NG** ,but it might take some time. By the time, if it is feasible you can use OTA supported in C Language (as you said,python would be bulky for you)using **FG** solution.

There are two parts in OTA. (1) download in part or full and (2) flash in parts
Python code is written in such away that small parts of binaries sequentially 'downloaded' on the fly and immediately 'flashed' (basically, you do not need to buffer the binary on the host). Basically (1)&(2) supported.

Right now, C demo expects the binary on the host but flashing is supported in parts. Basically, (1) is left for user's for their own HTTP client lib. (2) is supported. Please do try this OTA firmware update on FG.
The similar logic (like python), to download part can be easily implemented, using your http client library (you might already be using in your code) and mimicking the python code.

C demo should have very minimal footprint.

<ins>[Part B]</ins>
We had done such RF beacon test in past for some customer. I hope I have understood your question correctly.
If yes, Please spare me some time to check the implementation for RF beacon test and get back to you.

dukelec, 2022-10-09:
Hi Mantriyogesh,
Thanks for your answer.

[Part A]
I hope you can prioritize adding IAP functionality to NG, otherwise NG and SDIO single interface mode will not suitable for any product.
I would recommend firmware upgrade directly through SDIO instead of over-the-air.
If the bootloader firmware supports IAP, even if the main firmware is not perfect in a month or two, my product can still be shipped first.

My current chip of choice is ESP32-D0WD-V3, which has no FLASH and it only has 448 KB ROM. I guess it doesn't have enough space for OTA, but making bootloader.bin support IAP should be possible, and much easier than OTA.

[Part B]
I'm not sure what kind of RF beacon you mentioned, I feel like it should be possible, just make sure esp32 can emit a narrow enough spectrum signal and can be displayed by the spectrum analyzer.

The "wifi single carrier tx out" in the esp8089 documentation corresponds to the "TX tone: Single carrier TX signals." in the esp32 documentation:
[image]

[Part C]
This is a new part.
I would like to suggest you to add WiFi + BT function to the UART single interface in NG mode.
Because in many scenarios, the speed of UART is sufficient and using UART single interface is simple and convenient enough.

mantriyogesh, 2022-10-14:
Hello @dukelec,
Thank you for exact information and question.
 
[Part A]

> My current chip of choice is ESP32-D0WD-V3, which has no FLASH and it only has 448 KB ROM. I guess it doesn't have enough space for OTA, but making bootloader.bin support IAP should be possible, and much easier than OTA.

Yes, ESP32-D0WD-V3 doesn't come with flash. But you would need to provide one. 448KB ROM is reserved for first stage bootloader and cannot be updated. 
Please see 'example' block diagram for ESP32-Wroom-32E
[image]
from 
https://datasheet.lcsc.com/lcsc/2007301108_Espressif-Systems-ESP32-WROOM-32E-4MB_C701341.pdf

Here you can see, the module has flash provided outside the `ESP32-D0WD-V3`. There will be two options:
1. You can design your own way to use `ESP32-D0WD-V3` similar to this by manually providing the expected things.
  - ESP-Hosted runs fine on typical 4MB flash (which includes two OTA partitions).
  - We have even packed ESP-Hosted (including OTA) in 2MB (with some memory optimizations). See partition table for ESP32-C2:

[image]
- Please be warned, this partition table is example only & has very less (almost no) headroom. Ideally, 4MB gives us (& users) freedom for futuristic & practical flash size.
2. You can use the ESP module similar to ESP32-WROOM-32 directly, in which case FCC certifications & design efforts could be minimized.

**Some points in Firmware update**
- Outside ESP-Hosted: In case, you need something urgent, ESP is also supporting other different ways to do firmware update through extra lines directly from your Linux similar to [this](https://github.com/espressif/esp-serial-flasher/compare/master...mantriyogesh:esp-serial-flasher:master). Please note, there are numerous other ways to do this. 
- When we say, term OTA, In **FG**, we do it through the transport only. I mean, the firmware update would happen through SDIO (or SPI, whichever is the current transport). 
- For **NG**, We considered OTA as a high priority feature to be implemented, but cannot comment over the exact timelines.
- OTA will be supported similar to ESP-Hosted-FG, where there would be two OTA partitions (similar to above image). At implementation level, it may or may not exactly follow ESP-Hosted-FG (still under discussion, for example through character driver or IOCTL or something else). 

[Part B]
This is already done for another customer for **FG**. Again, in **NG**, we can do this once we have an interface (like character driver or IOCTL or something else) to communicate from application to Kernel driver. This we can easily support in FG, as we have character driver (serial interface we call it) there already present. In general, once this user-space to kernel-space interface is finalized, this should be an easy task to do (I see zero risks there)

[Part C]
Wi-Fi & Bluetooth over UART is something we discarded, considering UART speed can limit Wi-Fi speed. But considering simplicity & convenience point, we will check on this and get back.

dukelec, 2022-10-14:
Sorry, I didn't know much about esp32 before and just realized that I have to use an external flash or use a chip with internal flash, like ESP32-U4WDH.

However, I read that the espressif engineer mentioned in 2018:
"in theory the ESP32 can work without an external SPI flash chip, by getting a RAM-only image uploaded over another port (serial, SDIO)"
(From: https://www.esp32.com/viewtopic.php?t=5591 )

It would be nice if this could be implemented for Hosted-NG to reduce costs.

closed 2022-10-14 by dukelec

mantriyogesh, 2022-10-15:
Yeah, it definitely can be useful. It will further reduce the cost for the end users because they won't require the flash part.
But the problem with this is the entire firmware should fit in the RAM
If you can do that then it an interesting feature I would say
We do have an experimental feature which you can try out CONFIG_APP_BUILD_TYPE_ELF_RAM, in this we can directly load the application in the RAM and start executing from there