espressif/esp-hosted #167: Unable to connect to certain APs - ESP32

doutordre, 2022-10-14:
Hello,

I'm having trouble connecting my RPi CM3+, hosting an ESP32 through SDIO, to any AP at my office or to mobile hotspots. 
I can connect to my hope AP without troubeThere are no problems when I'm at home and connect .

I also had some trouble earlier with signal integrity which was fixed by following #26 

To demonstrate, below are the dmesg and ESP32 outputs when I attempt to connect to my mobile phone hotspot "test".

<details>
  <summary>*dmesg*</summary>

 ```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.61-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (2
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 3 Plus Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 28 pages/cpu s77272 r8192 d29224 u114688
[    0.000000] pcpu-alloc: s77272 r8192 d29224 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Compositet
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 665492K/970752K available (11520K kernel code, 1956K rwdata, 4140K rodata, 3840K init, 974K bss, 43116K )
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 38040 entries in 149 pages
[    0.000000] ftrace: allocated 149 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000356] Console: colour dummy device 80x25
[    0.001285] printk: console [tty1] enabled
[    0.001372] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.001436] pid_max: default: 32768 minimum: 301
[    0.001644] LSM: Security Framework initializing
[    0.001929] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001999] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.003715] cgroup: Disabling memory control group subsystem
[    0.007599] rcu: Hierarchical SRCU implementation.
[    0.009061] EFI services will not be available.
[... 546 lines cut ...]
I (40748) FW_MAIN: STA connect request

I (40748) FW_CMD: Attempting secured connection
I (40748) FW_CMD: test, channel: 6
I (42010) FW_CMD: STA Disconnect event: 201

I (42010) FW_CMD: cleanup_ap_bssid
I (42024) FW_MAIN: Delete key request

I (42042) FW_MAIN: Delete key request

I (42052) FW_MAIN: Delete key request

I (42072) FW_MAIN: Delete key request

I (42082) FW_MAIN: Delete key request

I (42100) FW_MAIN: Delete key request

I (42110) FW_MAIN: Scan request

  ```
</details>

Note that at my office there are many hidden APs when I scan with iwlist. When I'm back I'll post the logs here.

mantriyogesh, 2022-10-17:
Hello @doutordre 

Thanks a lot, for checking the detailed documentation and the issues before raising the issue. Much appreciated.
I understand your issue, that some APs you are not able to connect and some you are not able to connect.
The question pops to me, are you able to connect the same AP in these two environments?
- [ ] Can you please confirm if you can connect to hotspot in home environment but same hotspot you cannot connect to in office environment?

- [ ] Actually, I suspect your office has some noise in the channels you are using. Typically, running microwave oven, High bluetooth traffic near or in between ESP and AP might cause very high interference.

- [ ] Anyway, Can you please re-configure your office AP (altogether some different AP in same office environment), to radiate in 40MHz and 'automatic' channel selection?

- [ ] Error, '201' you get is basically, ESP-IDF is getting `REASON_NO_AP_FOUND` error. See https://bbs.espressif.com/viewtopic.php?t=4604 & https://github.com/espressif/esp-idf/issues/7954 some of the similar issues faced by ESP users.

- [ ] Is your AP is having good signal strength at ESP (Station mode: ESP is connecting to AP)? They should not be very close (as it might result in signal saturation) and should also not be very far (might result in weak signal)

doutordre, 2022-10-18:
Hi @mantriyogesh,
Thank you for your prompt reply.

I looked further at this and found out why I couldn't connect to my office (corporate) network: they use EAP as security so obviously the wpa.conf file had to reflect this. I'm not very experienced in Linux or networks, so I wasn't aware of this fact. I'll have to work this out with our IT department.
I found this out because I was able to connect to my office's Guest network!

In any case, I haven't been able to connect to mobile hotspots. It doesn't work either in my office or my home. 
As I was suspicious my mobile phone (Xiaomi A2 lite) was at fault I generated a mobile hotspot on my laptop called "pc_test". The problem persists, but with a different error code.
The logs are as follows

<details>
  <summary>dmesg</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.61-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU 2
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 3 Plus Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 28 pages/cpu s77272 r8192 d29224 u114688
[    0.000000] pcpu-alloc: s77272 r8192 d29224 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Compot
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 665492K/970752K available (11520K kernel code, 1956K rwdata, 4140K rodata, 3840K init, 974K bss, 431)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 38040 entries in 149 pages
[    0.000000] ftrace: allocated 149 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000353] Console: colour dummy device 80x25
[    0.001282] printk: console [tty1] enabled
[    0.001368] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.001432] pid_max: default: 32768 minimum: 301
[    0.001637] LSM: Security Framework initializing
[    0.001920] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001988] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.003712] cgroup: Disabling memory control group subsystem
[    0.007611] rcu: Hierarchical SRCU implementation.
[... 761 lines cut ...]
I (21606) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (21610) FW_CMD: STA disconnected

I (21614) FW_CMD: STA Disconnect event: 2

I (21618) FW_CMD: cleanup_ap_bssid
I (21640) FW_MAIN: Delete key request

I (21660) FW_MAIN: Delete key request

I (21670) FW_MAIN: Delete key request

I (21678) FW_MAIN: Delete key request

I (21698) FW_MAIN: Delete key request

I (21708) FW_MAIN: Delete key request

I (21726) FW_MAIN: Scan request

  ```
</details>

I'll have a look at the thread you linked and search the espressif foruns in the meantime.
Many thanks

mantriyogesh, 2022-10-18:
Can you please take Wi-Fi sniffer log? (it should be easy on laptops. Check Wi-Fi monitor mode if Linux)
Possibly in open security mode?
If this issue is not experienced on open mode, please make sure M1-M4 are captured in the capture for your security mode.

doutordre, 2022-10-21:
Hi @mantriyogesh,

I have attempted connecting the ESP32 to my mobile hotspot as a private network with WPA2-Personal security and as an open network. It failed and gave a very similar output as the logs above detail.
 
My laptop runs Windows so I have made a log with Wireshark of a connection to my mobile hotspot as an open network.
I don't know what we're looking for so I have run it without any filters applied. The logs are in attachment

Let me know if this is of any help.
[open_network_test.txt](https://github.com/espressif/esp-hosted/files/9838418/open_network_test.txt)
[private_network_test.txt](https://github.com/espressif/esp-hosted/files/9838419/private_network_test.txt)