espressif/esp-hosted #168: kernel 5 and NG

micheledarold, 2022-10-18:
Hi, what about kernel 5.x and NG version?
I have tried to compile but I got errors. It's supported?

Thank you

MIPMHannes, 2022-10-18:
Yeah kernel 5.x is supported. I got it up and running on kernel version 5.15.61-v7l+. The issue might be something else.

mantriyogesh, 2022-10-18:
Thank you. Are you still stuck or got it working?

micheledarold, 2022-10-18:
Hi, is still not working, I can't send the trace of the error because I need to complete some other urgent tasks and I switch to old version of kernel and yocto.
By the way I'm using yocto Rocko (kernel 4.x) and the problem are with yocto Dunfell kernel 5.10

mantriyogesh, 2022-10-18:
Any build logs would be much appreciated.

mantriyogesh, 2022-10-20:
@micheledarold please do connect when you are able to work on 5.10 kernel again.
Kernel 5.10 tested with Raspberry Pi & it is working as expected..

micheledarold, 2022-10-21:
Hi @mantriyogesh the problem was related to the Yocto recipe, the bluetooth was missing and so the kernel not work, now with the right setting everything is compiling
Referred also to #137 .
Thank you

closed 2022-10-21 by micheledarold