espressif/esp-hosted #169: ESP32-C3

ricoheller, 2022-10-21:
Hi @mantriyogesh, 

I've recently ported the ESP32-NG SPI  (esp32_spi) driver to my custom board.
SPI communication has been tested with spidev_test + spi slave example from Espressif. It's working fine.

I am using the evalutation firmware with driver patches provided [here](https://github.com/espressif/esp-hosted/issues/150#issuecomment-1285377376).
The problem is that hci interface does not turn up.

**dmesg** logs:

```vim
[    4.911039] esp32_spi: loading out-of-tree module taints kernel.
[    4.927364] esp_reset, ESP32: Triggering ESP reset.
[    5.145543] ESP32 peripheral is registered to SPI bus [3],chip select [0], SPI Clock [10]
               Received ESP bootup event
[    6.633188] EVENT: 3
[    6.635424] EVENT: 2
[    6.637617] ESP Reconfigure SPI CLK to 30 MHz
[    6.855385] ESP32 peripheral is registered to SPI bus [3],chip select [0], SPI Clock [30]
[    7.074954] EVENT: 0
[    7.077163] EVENT: 1
[    7.079390] esp32: process_fw_data ESP chipset's last reset cause:
[    7.088315] esp32: ESP Firmware version: 1.0.1
[    7.092761] ESP chipset detected [esp32-c3]
[   12.154866] esp32: Command[1] timed out
[   12.158725] esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[   12.165964] esp32: add_network_iface() failure, ret: -1
[   12.175228] ESP peripheral capabilities: 0x68
[   12.394862] ESP Bluetooth init
[   12.403605]   * WLAN on SPI
[   12.408855]     - HCI over SPI
```

**hciconfig** output:

```bash
root@hardknott:~# hciconfig
hci0:   Type: Primary  Bus: SPI
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN
        RX bytes:0 acl:0 sco:0 events:0 errors:0
        TX bytes:3 acl:0 sco:0 commands:0 errors:0

root@hardknott:~# hciconfig hci0 up
Can't init device hci0: Connection timed out (110)
```

What am I missing here?

Thanks for your support.

mantriyogesh, 2022-10-21:
Thank you @ricoheller for quick verification.
1. Can you please get the ESP side logs (minicom can be used)
2. Does it help if you comment below?
[image]

mantriyogesh, 2022-10-21:
In **addition** to above change, can you just decrease SPI freq from 10 to 1 MHz?
https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_ng/host/spi/esp_spi.c#L28

My intention is to understand any timing issues at SPI.

Also if possible, can you also mismatch the SPI mode at https://github.com/espressif/esp-hosted/blob/3769eea28bbbc59d8ea2aff425d62564b7fb3e68/esp_hosted_ng/host/spi/esp_spi.c#L458 from 2 to other modes?

ricoheller, 2022-10-21:
> Thank you @ricoheller for quick verification.
> 
>     1. Can you please get the ESP side logs (minicom can be used)
> 
>     2. Does it help if you comment below?
> 
> 
> [image]

Hi @mantriyogesh,

Commenting the code above did the trick! It is now working!

**dmesg** logs:

```vim
[    4.879915] esp32_spi: loading out-of-tree module taints kernel.
[    4.899046] esp_reset, ESP32: Triggering ESP reset.
[    5.120784] ESP32 peripheral is registered to SPI bus [3],chip select [0], SPI Clock [10]
               Received ESP bootup event
[    6.544476] EVENT: 3
[    6.546784] EVENT: 2
[    6.548981] ESP Reconfigure SPI CLK to 30 MHz
[    6.553345] EVENT: 0
[    6.555647] EVENT: 1
[    6.557850] esp32: process_fw_data ESP chipset's last reset cause:
[    6.566948] esp32: ESP Firmware version: 1.0.1
[    6.571406] ESP chipset detected [esp32-c3]
[    6.699189] ESP peripheral capabilities: 0x68
[    6.924934] ESP Bluetooth init
[    6.933642]   * WLAN on SPI
[    6.936714]   * BT/BLE
[    6.939088]     - HCI over SPI
[    6.942147]     - BLE only
```

**esp32** logs:

```vim
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16cc
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
I (30) boot: ESP-IDF v4.4.1-593-g6d85d53cee-dirty 2nd stage bootloader
I (30) boot: compile time 20:45:06
I (30) boot: chip revision: 3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 factory          factory app      00 00 00010000 00100000
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (102) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (109) boot: End of partition table
I (113) boot: Defaulting to factory image
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=18310h ( 99088) map
I (142) esp_image: segment 1: paddr=00028338 vaddr=3fc90a00 size=032d0h ( 13008) load
I (145) esp_image: segment 2: paddr=0002b610 vaddr=40380000 size=04a08h ( 18952) load
I (151) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=8afe8h (569320) map
I (245) esp_image: segment 4: paddr=000bb010 vaddr=40384a08 size=0bdf8h ( 48632) load
[... 62 lines cut ...]
I (585) FW_MAIN: ESP Bluetooth MAC addr: 68-67-25-54-24-3a

I (585) FW_SPI: Using SPI interface
I (595) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (595) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1615) FW_MAIN: Initial set up done
I (1715) FW_MAIN: INIT Interface command

I (1715) wifi:mode : sta (68:67:25:54:24:38)
I (1715) wifi:enable tsf
I (1715) FW_CMD: Wifi Sta mode set

I (1775) FW_MAIN: Get MAC command
```

**hciconfig** output:

```vim
root@hardknott:~# hciconfig
hci0:   Type: Primary  Bus: SPI
        BD Address: 68:67:25:54:24:3A  ACL MTU: 251:12  SCO MTU: 0:0
        UP RUNNING
        RX bytes:279 acl:0 sco:0 events:51 errors:0
        TX bytes:159 acl:0 sco:0 commands:0 errors:0
```vim

closed 2022-10-25 by ricoheller

mantriyogesh, 2022-11-15:
Hello @ricoheller 

We tried a different way to re-init spi clock.
You have already commented spi_reinit_spidev(). Further,

Please add this code:
[image]
[image]

Actually, your issue is not getting reproduced at our end, so we are trying to get if this fix works on your Linux.
If it works, we will use this code. This would be great help to us.

ricoheller, 2022-11-15:
> Hello @ricoheller
>
> We tried a different way to re-init spi clock. You have already commented spi_reinit_spidev(). Further,
>
> Please add this code: [image] [image]
>
> Actually, your issue is not getting [...]

Hi @mantriyogesh,

Adding the above lines didn't work for me:

**dmesg** log:

```vim
[  142.412546] esp_reset, ESP32: Triggering ESP reset.
[  142.625428] ESP32 peripheral is registered to SPI bus [3],chip select [0], SPI Clock [10]
[  144.028306]
               Received ESP bootup event
[  144.033482] EVENT: 3
[  144.035744] EVENT: 2
[  144.037938] ESP Reconfigure SPI CLK to 30 MHz
[  144.255403] ESP32 peripheral is registered to SPI bus [3],chip select [0], SPI Clock [30]
[  144.474871] EVENT: 0
[  144.477076] EVENT: 1
[  144.479267] esp32: process_fw_data ESP chipset's last reset cause:
[  144.479272] POWERON_RESET
[  144.488218] esp32: ESP Firmware version: 1.0.1
[  144.492668] ESP chipset detected [esp32-c3]
[  149.514877] esp32: Command[1] timed out
[  149.518742] esp32: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
[  149.525974] esp32: add_network_iface() failure, ret: -1
[  149.531206] network iterface init failed
[  149.535211] ESP peripheral capabilities: 0x68
[  149.754845] ESP Bluetooth init
[  149.758353] Capabilities: 0x68. Features supported are:
[  149.763713]   * WLAN on SPI
[  149.766617]   * BT/BLE
[  149.769030]     - HCI over SPI
[  149.772093]     - BLE only
```

**hciconfig** output:

```bash
$ hciconfig
hci0:   Type: Primary  Bus: SPI
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN
        RX bytes:0 acl:0 sco:0 events:0 errors:0
        TX bytes:6 acl:0 sco:0 commands:0 errors:0      
$ hciconfig hci0 up
Can't init device hci0: Connection timed out (110)
```

mantriyogesh, 2022-11-15:
After

> ESP32 peripheral is registered to SPI bus [3],chip select [0], SPI Clock [10]

Line,

> ESP32 peripheral is registered to SPI bus [3],chip select [0], SPI Clock [30]

is popped. Are you sure, you have commented `spi_reinit_spidev()`?

mantriyogesh, 2022-11-21:
Hello @ricoheller 

Thank you for all trials. I appreciate the help.
One question, is the input power supplied as per Linux SoC requirement?