espressif/esp-hosted #170: esp32 hosted - NG on imx8m : works

StephaneDFR, 2022-11-02:
First of all, **thanks a lot** for this nice project. 

We were able to run esp32 hosted (NG version 1.0.1) on an I.MX8 processor running linux kernel 5.15, with an on board esp32-Wroom32E module. The tough part was the hardware with the pull up requirements (needed to fuse flash voltage), as the driver just compiled without any modification.
We did not a lot of test yet, but we were successful to connect to our internet box and stream a usb camera plugged on our board. 

**Especially looking forward the 'AP' mode of esp32 with this NG driver**, as 'AP' mode is a must have for our application.

By the way, if this can help for debug, there are a lot of "FW_CMD: STA mic fail" message on esp32 uart output. dunno if it's important or not.  Dmesg shows also some messages such as (but very few compared to STA mic fail messages) : 
[ 9128.112832] mac: 00000000cdf1369e: 30 7c b2 dc 56 a1                                0|..V.
[ 9128.127125] mac: 000000008a57adf2: ff ff ff ff ff ff                                ......
[10470.562927] mac: 000000008a57adf2: ff ff ff ff ff ff                                ......

See attached logs if it can helps. 
best regards

[esp32_output.log.txt](https://github.com/espressif/esp-hosted/files/9917875/esp32_output.log.txt)
[dmesg.big.log.txt](https://github.com/espressif/esp-hosted/files/9917876/dmesg.big.log.txt)
[dmesg.log.txt](https://github.com/espressif/esp-hosted/files/9917877/dmesg.log.txt)

StephaneDFR, 2022-11-02:
> Just want to point out that 'AP' mode is currently **not** supported with NG driver as you can see [here](https://github.com/espressif/esp-hosted#22-key-differences)

I know ;) 
Just waiting for the next release with (finger crossed), AP mode

mantriyogesh, 2022-11-02:
Hello @StephaneDFR  @MIPMHannes ,

Thank you for the kind words and using NG software.

Yes, SoftAP mode is **not** yet supported in **Next Generation**. It would definitely come, but can't promise on timeline as if now.

SoftAP mode however is **supported & stable** with **First Generation**, if in case you wish to use by the time.

I'll notify here once the SoftAP is integrated in NG.

StephaneDFR, 2022-11-02:
thanks @mantriyogesh. i may give a try to FG driver but, of course, using standards linux tools such as hostapd would be a must.
I turned notifications ON to follow any update on this projet :)

StephaneDFR, 2022-12-20:
No problem. i closed the issue because it was 'it works on imx8m' 
Still waiting for AP mode for sure with NG release, but that's an other thing ;) and as you said, i still can use FG mode
Thanks for your work.